Ysurac/openmptcprouter #1391: Port forwading

tekmedm, 2020-11-17:
I tried port forwarding to a server on lan for my webserver on local lan as described in existing documentation about that.
 title is Redirect port via VPN on OpenMPTCProuter
Example to redirect port 80 from the VPS to the IP 192.168.100.3 on the network

I can't seen any 192.168.100.3 ip address configuration only  192.168.100.16.. reproduce setup nothing routed.

All is up to date router and VPS.
Please help

Ysurac, 2020-11-17:
I fixed the ip in the doc.

tekmedm, 2020-11-17:
> Répondre à : Ysurac/openmptcprouter <reply@reply.github.com>

Please can you confirm by default zone source is always VPN ?

Ysurac, 2020-11-17:
yes, all forwarded traffic use VPN.

If you have issues, check https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding#debug

tekmedm, 2020-11-17:
Thanks a lot Yannick

We have a huge disruption I’m trying to restore service using your amazing stuff  through 4G, we have 300Mbps link. 

Do you think I can also transfert OVH Toip Voice calls ad have a minimum quality  ? 

 

Regards,

Mehdi

Date : mardi 17 novembre 2020 à 19:49

tekmedm, 2020-11-17:
Thanks Yannick

closed 2020-11-17 by tekmedm

Adorfer, 2020-11-17:
for the moment i can not get port-forwarding to work for ipv6. (for OMR55.3 it was working as far as i remember)