Ysurac/openmptcprouter #1392: 3CX SIP ALG

HeLLsSs, 2020-11-17:
Hello everyone, I have a problem installing a local 3CX

testing 3CX SIP Server... done
    stopping service... done
    detecting SIP ALG... detected (received c8c82d3f ≠ fa580842) (How to resolve?)
    testing port 5060... done
    starting service... done

3CX detects that the SIP ALG is active? Except that I can not find any how to disable it on OpenMPTCProuter

If anyone can help me ... thanks

HeLLsSs, 2020-11-18:
Pour information, j'ai supprimé le package kmod-nf-nathelper-extra qui ajoute les régles ci-dessous

opkg remove kmod-nf-nathelper-extra

Chaîne zone_lan_helper (1 Références)
Chaîne PREROUTING, Règle #1
Pqts.	Trafic	Cible	Prot.	Entrée	Sortie	Source	Destination	Options	Commentaire
0 	0 B	CT	udp	*	*	0.0.0.0/0	0.0.0.0/0	udp dpt:10080 CT helper amanda	Amanda backup and archiving proto
0 	0 B	CT	tcp	*	*	0.0.0.0/0	0.0.0.0/0	tcp dpt:21 CT helper ftp	FTP passive connection tracking
0 	0 B	CT	udp	*	*	0.0.0.0/0	0.0.0.0/0	udp dpt:1719 CT helper RAS	RAS proto tracking
0 	0 B	CT	tcp	*	*	0.0.0.0/0	0.0.0.0/0	tcp dpt:1720 CT helper Q.931	Q.931 proto tracking
0 	0 B	CT	tcp	*	*	0.0.0.0/0	0.0.0.0/0	tcp dpt:6667 CT helper irc	IRC DCC connection tracking
0 	0 B	CT	tcp	*	*	0.0.0.0/0	0.0.0.0/0	tcp dpt:1723 CT helper pptp	PPTP VPN connection tracking
0 	0 B	CT	tcp	*	*	0.0.0.0/0	0.0.0.0/0	tcp dpt:5060 CT helper sip	SIP VoIP connection tracking
144 	88.11 KB	CT	udp	*	*	0.0.0.0/0	0.0.0.0/0	udp dpt:5060 CT helper sip	SIP VoIP connection tracking
0 	0 B	CT	udp	*	*	0.0.0.0/0	0.0.0.0/0	udp dpt:161 CT helper snmp	SNMP monitoring connection tracking

Je n'ai plus de probléme avec le 3CX du coup

Ysurac, 2020-11-18:
Et surtout la réception et émission d'appels fonctionne ?

flypc-fr, 2021-01-14:
Merci pour vos explications, comment avez vous mis en place le Full Cone Nat ? 
D'avance merci.

closed 2021-04-19 by github-actions[bot]