Ysurac/openmptcprouter #1393: Reseaux different ?

Esysteme, 2020-11-18:
**Chaque box doit avoir un adressage réseaux différent.**

ma question est pourquoi, il faut que les reseaux soient different ?

j'ai l'impression que ça marche mieux dans la même plage.

question : avec RPI4, je suis bridé à 500MB en download (ça parait normal), par contre en upload 300MB et je ne comprend pas pourquoi.

(j'ai 3 fibres aggréger, bouygue / free / orange), le but est d'avoir une connexion résiliante sans payer de la fibre pro.

Ysurac, 2020-11-18:
Car lors de l'utilisation de macvlan sur le RPI tout le trafic provient d'une seule interface et apparaît donc sur tous les WANs définis ce qui pause certains soucis. Il n'est pas nécessaire d'avoir un adressage réseau différent sur les architectures avec plusieurs ports ethernet ou lors de l'utilisation de vlan.

Pour l'upload il faudrait voir la charge CPU sur le routeur et sur le VPS. Mais il est effectivement souvent plus faible que le download.

Esysteme, 2020-11-18:
Le VPS n'a pas de problème de CPU ou de bande passante 

test bande passante VPS : 
```
wget http://speedtest.tele2.net/10GB.zip
--2020-11-18 10:53:42--  http://speedtest.tele2.net/10GB.zip
Resolving speedtest.tele2.net (speedtest.tele2.net)... 90.130.70.73, 2a00:800:1010::1
Connecting to speedtest.tele2.net (speedtest.tele2.net)|90.130.70.73|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 10737418240 (10G) [application/zip]
Saving to: ‘10GB.zip’

10GB.zip                                             100%[=====================================================================================================================>]  10.00G  97.7MB/s    in 91s     

2020-11-18 10:55:13 (112 MB/s) - ‘10GB.zip’ saved [10737418240/10737418240]

Depuis OVH mais comme c'est un VPS OVH 1G/s le test passe l'interne et est peut-être biaisé :  

wget http://ovh.net/files/10Gio.dat
--2020-11-18 10:49:47--  http://ovh.net/files/10Gio.dat
Resolving ovh.net (ovh.net)... 213.186.33.6
Connecting to ovh.net (ovh.net)|213.186.33.6|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 10737418240 (10G) [application/octet-stream]
Saving to: ‘10Gio.dat’

10Gio.dat                                            100%[=====================================================================================================================>]  10.00G   111MB/s    in 95s     

2020-11-18 10:51:22 (108 MB/s) - ‘10Gio.dat’ saved [10737418240/10737418240]

```

Sachant que Orange fournit 400/400Mb très stable. et Free moins stable avec 900Mb en download et 150 en up. l'upload sur l'agrégation est inférieur à Orange qui tourne 300~350 Mb /sec.

à moins que le RPi4 pose problème à l'upload. J'ai essayé de passé sur une VM, mais mon niveau est ras des paquerettes en virtu, il y aurait pas un script d'installation ?

Quels sont les meilleurs débits que vous avez pu observer ?

kevin39, 2020-11-18:
Je dis pas que tu n'as pas un problème mais n'oublie pas que la bande passante n'est pas garantie en fibre grand publique et peut varier beaucoup au long de la journée (à la grande différence des fibres pro qui ne sont pas mutualisés avec autant de client et généralement avec un débit garanti).

Plusieurs remarques : 
- En testant WAN par WAN, tu as combien en upload sur chaque fibre ?
- Ne test par ton VPS OVH sur le speedtest OVH, c'est pas le bon plan en effet, il va être biaisé. Tu peux utiliser speedtest-cli par exemple.

Pour tester l'upload seulement tu peux utiliser ce genre d'outil : https://testmy.net/upload (mais ça impliquera de se connecter en direct à chaque box ou de jouer avec omr-bypass).

Ysurac, 2020-11-18:
Pour tester sur une VM, avec virtualbox c'est très rapide en mettant une interface en "accès par pont" puis https://github.com/Ysurac/openmptcprouter/wiki/Router-install#for-virtualbox
Pour les tests depuis OpenMPTCProuter, via SSH ou via Services->Terminal, il y a ```omr-test-speed```

Esysteme, 2020-11-19:
WAN par WAN 

en téléversant un fichier de 10 Gb sur le vps depuis mon PC, à l'instant t.

Free
```
root@68k-lecentre:/home/aurelien/Download# scp -P 65222 10Gb.dat root@51.210.x.x:/root
10Gb.dat                                                                                                                                                                         100% 1192MB  54.9MB/s   00:21    
```

(J'ai changé ma config réseau entre 2)

Orange
```
root@68k-lecentre:/home/aurelien/Download# scp -P 65222 10Gb.dat root@51.210.x.x:/root
10Gb.dat                                                                                                                                                                         100% 1192MB  47.5MB/s   00:25 
```

Esysteme, 2020-11-19:
Après mon premier test sur le VPS concernait tele2 en suède (112 MB/s) soit meilleur que le speed test interne OVH curieusement, où je sais qu' OVH n'a pas infrastructure ici.

closed 2020-11-25 by Esysteme