Ysurac/openmptcprouter #1394: Unable to connect to OMR inside virtualbox

ndaptardar, 2020-11-18:
## Expected Behavior
After installing OMR ext4 image (openmptcprouter-v0.56.5-r0+14653-a439f1bb47-x86-64-generic-ext4-combined.vdi.gz) in Virtualbox (version 5.2) as guest, and setting up 1 network as below, Virtualbox host (windows 10 machine) should be able to connect to Luci on 192.168.100.1

[image]

## Current Behavior
Not able to find/ping 192.168.100.1 from Virtualbox host. But I can see OMR has eth0 with 192.168.100.1. Windows machine IP is 192.168.0.10 and Virtualbox host only network adapter on windows has 192.168.56.1.

## Specifications

  - OpenMPTCProuter version: openmptcprouter-v0.56.5-r0+14653-a439f1bb47-x86-64-generic-ext4-combined.vdi.gz
  - OpenMPTCProuter VPS version: Not using it
  - OpenMPTCProuter platform: Virtualbox x86_64

Any idea how to fix this? I would like to try and use OMR as Virtualbox guest OS.

Ysurac, 2020-11-18:
You need to have add an IP on 192.168.100.x network on your host.

Adorfer, 2020-11-18:
i do not see any reason to use emulated NICs instead of virtio-net.

ndaptardar, 2020-11-19:
I added 192.168.100.2 on virtualbox host only network adapter, but no luck. 

Then I checked "route -n" on OMR, and here is that I see. Is this correct?
[image]

Ysurac, 2020-11-19:
You should be able to ping OMR IP from host. Set also an IP on 192.168.100.x network on your windows client computer.

ndaptardar, 2020-11-19:
> You should be able to ping OMR IP from host. Set also an IP on 192.168.100.x network on your windows client computer.

Tried that, but ping not working. Also, I dont know how to set IP on windows host  but I tried this ... windows has 3 adapters. Ethernet (not used), Wireless (not used) and Virtualbox host only adapter, which has IP 192.168.56.1. I set another IP on 
 Virtualbox host only adapter as 192.168.100.2. But 100.2 is not pinging 100.1. and vice versa. Its as if virtual box doesnt know that there is eth0 with 192.168.100.1 in it. I tried adding route in OMR to 192.168.56.1 as gateway. Its added, but ping doesnt work. Am I missing something? Are there any other steps I need to carry out?

Adorfer, 2020-11-19:
this sounds more like a networking issue on Virtualbox in winodows-host/bridge.

please post OMR output of 

     ip l

ndaptardar, 2020-11-19:
> this sounds more like a networking issue on Virtualbox in winodows-host/bridge.
> 
> please post OMR output of
> 
> ```
>  ip l
> ```
[image]

[image]

Adorfer, 2020-11-19:
it looks like you have only a single (virtual) network adaptor in the vm, in plus the mac is different from what you set up. 
you should add a second adapter. 
In plus i do not see an argument why you have to go with the emulated hardware NIC instead of virtio-net, ince it's known to be a) cpu consuming and b) instable on the long run (if you look for transfering TB and not only GB)

ndaptardar, 2020-11-20:
> it looks like you have only a single (virtual) network adaptor in the vm, in plus the mac is different from what you set up.
> you should add a second adapter.
> In plus i do not see an argument why you [...]

Yes, I agree on speed and stability. I just wanted to run 2 OMR in parallel. I had RPI 4B, and wanted another to setup. Anyways, I burned USB and tried it on my laptop, it worked, so no need to go with virtualbox.

Thanks anyways all for the response!!

closed 2020-11-20 by ndaptardar