Ysurac/openmptcprouter #1396: Network adapter not recognized

jpossetti, 2020-11-19:
## Expected Behavior

Installed openmptcp router on a new i3 9000f series, GIGABYTE H310M motherboard. Installed my network adapter (TP-LINK pci-e TG-3468 but it is NOT recognized by the wizzard. This network adapter was recognized on my old PC but not now in this new deploy.

## Current Behavior
The adapter is linking but not recognized by the os. Only onboard adapter (and 4g/LTE usb dongles) are recognized OK. 

## Specifications

  - OpenMPTCProuter version: 0.56.5
  - OpenMPTCProuter VPS version: 0.1020 0.5.65-mptcp
  - OpenMPTCProuter platform: x86_64

Ysurac, 2020-11-19:
A module can be needed. You need to find the chipset used by this card and install the required driver.

Adorfer, 2020-11-19:
a) try a differend slot in order to mix up the IRQ assignment
b) there might be some interference on some Mainboards, see https://linuxreviews.org/TP-LINK_TG-3468_PCIe_Gigabit_Adapter

jpossetti, 2020-11-20:
> a) try a differend slot in order to mix up the IRQ assignment
> b) there might be some interference on some Mainboards, see https://linuxreviews.org/TP-LINK_TG-3468_PCIe_Gigabit_Adapter

Thanks, NOT DRIVER ISSUE. I have same both onboard and pci-e network chipset (r8169) read some linux dist can cause some problems with this. Solved to install in different PCI-e slot (need to reinstall openmptcp)

closed 2020-11-20 by jpossetti