Ysurac/openmptcprouter #1397: Chromecast et dns Google

Merrick28, 2020-11-19:
Bonjour à tous,
je suis l'heureux possesseur d'un nouveau Chromecast.
J'utilise OpenMPTCProuter version 0.56.3-bce430e2 et VPS OVH 0.1022 5.4.74-mptcp

Youtube et consorts semblent bloqués depuis l'ip de mon vps, et j'ai du mettre un OMR Bypass sur l'ip source du chromecast pour que ces applis marchent. Je ne peux pas le faire par protocole car le chromecast utilise en dur les dns de google (8.8.8.8 et 8.8.4.4). Du coup, le chromecast ne bénéficie plus de l'aggrégation, y compris pour les autres applis.
Est-il possible de rediriger tout le trafic interne vers les ip des dns de google vers le dns du routeur ? Cela permettrait d'avoir un bypass par protocole/appli et non plus par ip source.

Ysurac, 2020-11-19:
Pour bloquer les DNS de google depuis un chromecast ayant l'IP 192.168.100.6, il faut ajouter dans Réseau->Parefeu puis onglet règles spécifiques :
```
iptables -I FORWARD -s 192.168.100.6 --destination 8.8.8.8 -j REJECT
iptables -I FORWARD -s 192.168.100.6 --destination 8.8.4.4 -j REJECT
```
Normalement il utilise les DNS distribués par DHCP. Sinon il doit être possible aussi de rediriger vers l'IP du routeur au lieu de celles de google via iptables.

Merrick28, 2020-11-19:
C'est parfait !
J'ai mis en OMR Bypass les 'protocoles et services' google et youtube (youtube seul ne suffisait pas), les règles indiquées, et tout marche très bien. Merci beaucoup.

closed 2020-11-19 by Merrick28