Ysurac/openmptcprouter #1398: Mise à jour 0.56.5 et 0.1022 problèmes

JeedHome44, 2020-11-19:
Bonjour,

Hier, j'ai mis à jour d'abord mon VPS en 22, j'étais en 18. J'ai ensuite mis le Pi 4 en 56.5 j'étais en 56.1

Tout marchait bien avant ça. Depuis la mise à jour, le VPS passe régulièrement de OK à HS. et sur le routeur, c'est marqué "Le VPN ne répond pas".
J'ai aussi régulièrement le wan 4G qui passe de OK à "pas de réponse du serveur après 1 seconde". Le wan Freebox quand à lui est ok. LA connexion 4G marche bien pourtant.

J'ai mis la connexion 4G en maître et ca marchait très bien avant la mise à jour.
J'ai fait la mise à jour en téléchargeant le fichier sysupgrade et ça c'est bien passé pourtant.

J'ai l'impression que la mise à jour ne veut plus de la connexion 4G en maître. Quelque chose a été changé à ce niveau là?
J'ai redémarré pourtant le VPS après installation et ce matin le Pi 4B pour être sur que ca ne venait pas d'un reboot pas fait mais ca ne change rien.

Je suis sous Pi 4B 2Go
VPS OVH sous Debian10

JeedHome44, 2020-11-19:
[image]
[image]

JeedHome44, 2020-11-19:
Est-ce un problème de Shadowsocks?

JeedHome44, 2020-11-19:
J'ai mis l'ADSL est maître pour tester et toujours même problème.

JeedHome44, 2020-11-19:
Et je ne pige pas car tout revient au vert de temps en temps sur "Le VPN ne répond pas" sur le routeur puis ça refait en boucle le problème.

Ysurac, 2020-11-19:
Pourquoi ne pas utiliser le template ? ça aide pas pour avoir les infos rapidement...
Comme toujours, il faut voir les logs dans Etat->Journal système.

JeedHome44, 2020-11-19:
On voit mon ip externe sur les logs donc pas trop envi de les partager ici...
Je ne comprends pas vraiment le template en fait. Je n'ai pas bien expliqué le problème?

Dans les logs, je vois beaucoup de ligne comme celles-ci:

Thu Nov 19 11:17:01 2020 daemon.info glorytun[28591]: STOPPED tun0
Thu Nov 19 11:17:02 2020 daemon.err glorytun[28591]: XXXXXXXXX.65001: connected
Thu Nov 19 11:17:02 2020 daemon.info glorytun[28591]: STARTED tun0

Thu Nov 19 11:16:13 2020 user.notice OMR-VPS: Can't get token, try later (can't ping server, server API answer)

JeedHome44, 2020-11-19:
Ceci aussi:

Thu Nov 19 11:11:58 2020 user.notice post-tracking-post-tracking: wan2 (wan2) switched off because ping from 192.168.8.2 error (114.114.115.115,4.2.2.1)
Thu Nov 19 11:11:58 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.8.1 dev wan2
Thu Nov 19 11:11:58 2020 daemon.err glorytun[28591]: XXXXXXXXX.65001: connected

JeedHome44, 2020-11-19:
Globalement, il y a beaucoup de lignes comme ça:

Thu Nov 19 10:50:01 2020 daemon.info glorytun[28591]: STOPPED tun0
Thu Nov 19 10:50:01 2020 daemon.err glorytun[28591]: XXXXXXXXXXX.65001: connected
Thu Nov 19 10:50:01 2020 daemon.info glorytun[28591]: STARTED tun0
Thu Nov 19 10:50:06 2020 user.notice post-tracking-post-tracking: wan2 (wan2) switched up
Thu Nov 19 10:50:06 2020 user.notice post-tracking-post-tracking: Set ip route replace default scope global  nexthop via 192.168.0.1 dev wan1 weight 1 nexthop via 192.168.8.1 dev wan2 weight 2

Ysurac, 2020-11-19:
Le template permet d'avoir une présentation standardisé et d'avoir rapidement accès aux informations importantes. Ce qui est beaucoup plus pratique.
Je peux pas aider avec juste des petits extrait de logs...

JeedHome44, 2020-11-19:
je viens de reboot le routeur 4G et ça a l'air de ne plus faire de problème.
ça arrive ce genre de soucis quand on met à jour OMR sans reboot les clients internet?

Ysurac, 2020-11-19:
non, jamais eu ce cas avec des routeurs ethernet.

JeedHome44, 2020-11-19:
Bon, fausse joie.
Ça recommence... 😞

JeedHome44, 2020-11-19:
Possibilité de vous envoyer les logs en MP?

Ysurac, 2020-11-19:
Le log peut être mis ici ou via un pastebin en enlevant l'IP du VPS et les IPs des WANs.

JeedHome44, 2020-11-20:
Bonjour,

En regardant les configurations, j'ai constaté que mes 2 connexions sur sur MacVLAN. Avant c'était sûr normal avec adresse statique. J'ai donc remis comme avant.
Pourquoi ça a changé tout seul?

C'est peut être ça le problème?
Qu'est-ce qui a changé également en configuration entre la 56.1 et la 56.5?

JeedHome44, 2020-11-20:
J'ai refait la mise à jour du VPS ce matin et ça ne fait plus de problème.
Il y a un ordre particulier à faire? D'abord le Pi puis le VPS?

JeedHome44, 2020-11-20:
Encore parle trop vite... des fois ça marche et des fois non...
Je ne comprends pas ce qui ne va pas.

Pour mettre les logs ici, ils sont très long. Je ne peux pas cacher chaque ligne d'ip. Ça me prendrait énormément de temps.

Pas possible de vous envoyer ça par mail?

Ysurac, 2020-11-20:
Il me faut que la fin des logs, maximum la dernière heure. Pour remplacer l'IP tout éditeur de texte permet un recherche et remplace, sinon un sed en ligne de commande sous Linux permet de remplacer ça rapidement.

JeedHome44, 2020-11-20:
voici les logs de la dernière heure avec ip du VPS caché. Merci

Fri Nov 20 16:16:39 2020 daemon.err glorytun[28613]: XXX.XX.XXX.XXX.65001: connected
Fri Nov 20 16:16:39 2020 daemon.info glorytun[28613]: STARTED tun0
Fri Nov 20 16:16:47 2020 daemon.info dnsmasq-dhcp[29885]: DHCPREQUEST(eth0) 192.168.100.57 70:4f:57:62:03:9c
Fri Nov 20 16:16:47 2020 daemon.info dnsmasq-dhcp[29885]: DHCPACK(eth0) 192.168.100.57 70:4f:57:62:03:9c Access-Point-Garage
Fri Nov 20 16:17:08 2020 daemon.info dnsmasq-dhcp[29885]: DHCPREQUEST(eth0) 192.168.100.57 70:4f:57:62:03:9c
Fri Nov 20 16:17:08 2020 daemon.info dnsmasq-dhcp[29885]: DHCPACK(eth0) 192.168.100.57 70:4f:57:62:03:9c Access-Point-Garage
Fri Nov 20 16:17:23 2020 daemon.info glorytun[28613]: STOPPED tun0
Fri Nov 20 16:17:24 2020 daemon.err glorytun[28613]: XXX.XX.XXX.XXX.65001: connected
Fri Nov 20 16:17:24 2020 daemon.info glorytun[28613]: STARTED tun0
Fri Nov 20 16:17:28 2020 daemon.info dnsmasq-dhcp[29885]: DHCPREQUEST(eth0) 192.168.100.57 70:4f:57:62:03:9c
Fri Nov 20 16:17:28 2020 daemon.info dnsmasq-dhcp[29885]: DHCPACK(eth0) 192.168.100.57 70:4f:57:62:03:9c Access-Point-Garage
Fri Nov 20 16:17:48 2020 daemon.info dnsmasq-dhcp[29885]: DHCPREQUEST(eth0) 192.168.100.57 70:4f:57:62:03:9c
Fri Nov 20 16:17:48 2020 daemon.info dnsmasq-dhcp[29885]: DHCPACK(eth0) 192.168.100.57 70:4f:57:62:03:9c Access-Point-Garage
Fri Nov 20 16:17:52 2020 daemon.info dnsmasq-dhcp[29885]: DHCPDISCOVER(eth0) 70:ee:50:00:14:86
Fri Nov 20 16:17:52 2020 daemon.info dnsmasq-dhcp[29885]: DHCPOFFER(eth0) 192.168.100.18 70:ee:50:00:14:86
Fri Nov 20 16:17:52 2020 daemon.info dnsmasq-dhcp[29885]: DHCPREQUEST(eth0) 192.168.100.18 70:ee:50:00:14:86
Fri Nov 20 16:17:52 2020 daemon.info dnsmasq-dhcp[29885]: DHCPACK(eth0) 192.168.100.18 70:ee:50:00:14:86 Station-Netatmo
Fri Nov 20 16:18:08 2020 daemon.info dnsmasq-dhcp[29885]: DHCPREQUEST(eth0) 192.168.100.57 70:4f:57:62:03:9c
Fri Nov 20 16:18:08 2020 daemon.info dnsmasq-dhcp[29885]: DHCPACK(eth0) 192.168.100.57 70:4f:57:62:03:9c Access-Point-Garage
Fri Nov 20 16:18:09 2020 daemon.info glorytun[28613]: STOPPED tun0
Fri Nov 20 16:18:09 2020 daemon.err glorytun[28613]: XXX.XX.XXX.XXX.65001: connected
Fri Nov 20 16:18:09 2020 daemon.info glorytun[28613]: STARTED tun0
Fri Nov 20 16:18:28 2020 daemon.info dnsmasq-dhcp[29885]: DHCPREQUEST(eth0) 192.168.100.57 70:4f:57:62:03:9c
Fri Nov 20 16:18:28 2020 daemon.info dnsmasq-dhcp[29885]: DHCPACK(eth0) 192.168.100.57 70:4f:57:62:03:9c Access-Point-Garage
Fri Nov 20 16:18:41 2020 daemon.info dnsmasq-dhcp[29885]: DHCPRELEASE(eth0) 192.168.100.18 70:ee:50:00:14:86
Fri Nov 20 16:18:49 2020 daemon.info dnsmasq-dhcp[29885]: DHCPREQUEST(eth0) 192.168.100.57 70:4f:57:62:03:9c
Fri Nov 20 16:18:49 2020 daemon.info dnsmasq-dhcp[29885]: DHCPACK(eth0) 192.168.100.57 70:4f:57:62:03:9c Access-Point-Garage
Fri Nov 20 16:18:54 2020 daemon.info glorytun[28613]: STOPPED tun0
Fri Nov 20 16:18:55 2020 daemon.err glorytun[28613]: XXX.XX.XXX.XXX.65001: connected
Fri Nov 20 16:18:55 2020 daemon.info glorytun[28613]: STARTED tun0
Fri Nov 20 16:19:09 2020 daemon.info dnsmasq-dhcp[29885]: DHCPREQUEST(eth0) 192.168.100.57 70:4f:57:62:03:9c
Fri Nov 20 16:19:09 2020 daemon.info dnsmasq-dhcp[29885]: DHCPACK(eth0) 192.168.100.57 70:4f:57:62:03:9c Access-Point-Garage
Fri Nov 20 16:19:29 2020 daemon.info dnsmasq-dhcp[29885]: DHCPREQUEST(eth0) 192.168.100.57 70:4f:57:62:03:9c
Fri Nov 20 16:19:29 2020 daemon.info dnsmasq-dhcp[29885]: DHCPACK(eth0) 192.168.100.57 70:4f:57:62:03:9c Access-Point-Garage
Fri Nov 20 16:19:39 2020 daemon.info glorytun[28613]: STOPPED tun0
Fri Nov 20 16:19:49 2020 daemon.info dnsmasq-dhcp[29885]: DHCPREQUEST(eth0) 192.168.100.57 70:4f:57:62:03:9c
Fri Nov 20 16:19:49 2020 daemon.info dnsmasq-dhcp[29885]: DHCPACK(eth0) 192.168.100.57 70:4f:57:62:03:9c Access-Point-Garage
Fri Nov 20 16:20:04 2020 daemon.err glorytun[28613]: XXX.XX.XXX.XXX.65001: connected
Fri Nov 20 16:20:04 2020 daemon.info glorytun[28613]: STARTED tun0
Fri Nov 20 16:20:10 2020 daemon.info dnsmasq-dhcp[29885]: DHCPREQUEST(eth0) 192.168.100.57 70:4f:57:62:03:9c
Fri Nov 20 16:20:10 2020 daemon.info dnsmasq-dhcp[29885]: DHCPACK(eth0) 192.168.100.57 70:4f:57:62:03:9c Access-Point-Garage
Fri Nov 20 16:20:25 2020 daemon.info glorytun[28613]: STOPPED tun0
Fri Nov 20 16:20:25 2020 daemon.err glorytun[28613]: XXX.XX.XXX.XXX.65001: connected
Fri Nov 20 16:20:25 2020 daemon.info glorytun[28613]: STARTED tun0
Fri Nov 20 16:20:30 2020 daemon.info dnsmasq-dhcp[29885]: DHCPREQUEST(eth0) 192.168.100.57 70:4f:57:62:03:9c
Fri Nov 20 16:20:30 2020 daemon.info dnsmasq-dhcp[29885]: DHCPACK(eth0) 192.168.100.57 70:4f:57:62:03:9c Access-Point-Garage
Fri Nov 20 16:20:50 2020 daemon.info dnsmasq-dhcp[29885]: DHCPREQUEST(eth0) 192.168.100.57 70:4f:57:62:03:9c
Fri Nov 20 16:20:50 2020 daemon.info dnsmasq-dhcp[29885]: DHCPACK(eth0) 192.168.100.57 70:4f:57:62:03:9c Access-Point-Garage
Fri Nov 20 16:20:55 2020 user.notice post-tracking-post-tracking: wan2 (wan2) switched off because ping from 192.168.8.2 error (114.114.114.114,114.114.115.115)
Fri Nov 20 16:20:55 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.8.1 dev wan2
Fri Nov 20 16:21:10 2020 daemon.info glorytun[28613]: STOPPED tun0
Fri Nov 20 16:21:10 2020 daemon.info dnsmasq-dhcp[29885]: DHCPREQUEST(eth0) 192.168.100.57 70:4f:57:62:03:9c
Fri Nov 20 16:21:10 2020 daemon.info dnsmasq-dhcp[29885]: DHCPACK(eth0) 192.168.100.57 70:4f:57:62:03:9c Access-Point-Garage
[... cut ...]
Fri Nov 20 17:08:11 2020 daemon.err uhttpd[3392]: luci: accepted login on / for root from 192.168.100.30
Fri Nov 20 17:08:14 2020 daemon.info dnsmasq-dhcp[32388]: DHCPDISCOVER(eth0) 00:0e:c6:b6:65:53
Fri Nov 20 17:08:14 2020 daemon.info dnsmasq-dhcp[32388]: DHCPOFFER(eth0) 192.168.100.13 00:0e:c6:b6:65:53
Fri Nov 20 17:08:14 2020 daemon.info dnsmasq-dhcp[32388]: DHCPREQUEST(eth0) 192.168.100.13 00:0e:c6:b6:65:53
Fri Nov 20 17:08:14 2020 daemon.info dnsmasq-dhcp[32388]: DHCPACK(eth0) 192.168.100.13 00:0e:c6:b6:65:53 Mi-Box-Bureau

Ysurac, 2020-11-20:
Je vois que wan2 a tendance à se déconnecter/reconnecter, l'interface master est laquelle ?
Et "Sélection de l'interface maître" est réglé comment ? (dans Système->OpenMPTCProuter, onglet "Paramètres avancés")

JeedHome44, 2020-11-20:
J'ai mis en master la 4G et en activé l'ADSL car je veux avoir certains appareils qui passent en 4G prio et si elle lache, passe en ADSL. Exemple: netflix...
Il n'est pas possible de mettre l'Adsl en master et les by-pass, les mettre sur la 4G. Si la 4G saute, ça ne passe pas sur l'ADSL? Les By-pass je n'ai pas renseigné les WAN pour que ce soit le master mis en priorité.
C'est pour ça que je veux la 4G en priorité.

Les 2 WAN étaient réglés sur adresse statique avant la mise à jour et après c'était sur MacVLAN. J'ai donc remis sur Normal les 2.

Je précise qu'il n'y avait aucune déconnection comme ça avant la mise à jour.
Dés que je l'ai faite, ça a fait ces déconnections en boucle.

Comme s'il y avait un problème de communication avec le serveur VPS.

Régulièrement, le WAN 4G master marque "aucune communication avec le serveur après 1 seconde".

Ça ne serait pas la mise à jour du VPS qui a un soucis?

JeedHome44, 2020-11-20:
J'ai oublie aussi de préciser que j'ai testé de changer le master en le mettant sur l'ADSL et quand je fais ça, c'est l'ADSL qui fait des erreur de serveur après 1 seconde.
Ça ne serait donc pas un problème avec la mise à jour du VPS?

JeedHome44, 2020-11-20:
Est il possible de revenir a la mise à jour 18 du VPS?
Si non, est-ce possible de faire un BackUp du VPS comme celui qui est possible du Pi?

Merci

Ysurac, 2020-11-20:
Non pas possible de revenir en arrière, et pas de backup du VPS car il n'y a pas grand chose à backuper dans une utilisation normale car tout est envoyé par le routeur.
L'erreur est bien lié à un soucis de communication entre le routeur et le VPS mais pas lié à la version du VPS car il s'agit d'un simple PING ICMP.

JeedHome44, 2020-11-20:
Donc c'est la version de OMR qui a un soucis? J'ai l'impression que c'est le démon Glorytun non?

Ysurac, 2020-11-20:
Je n'arrive pas à reproduire le soucis, donc je ne peux pas dire. Tu peux tenter un autre VPN.

JeedHome44, 2020-11-20:
Je ne sais pas comment ça fonctionne vraiment mais question bête: faut il obligatoirement passer par un VPN?

JeedHome44, 2020-11-20:
Avec Pi-hole, il n'y avait pas une question d'un VPN obligatoire?

Ysurac, 2020-11-20:
Le VPN sert pour tout ce qui ne passe pas par le proxy et pour la redirection de ports.
Donc pour UDP et ICMP notamment, dans le cas du proxy shadowsocks, et juste ICMP pour le proxy V2Ray.

JeedHome44, 2020-11-21:
Y a-t-il possibilité de vérifier l'état du démon glorytun sur le VPS? J'ai l'impression que ça vient de la.

JeedHome44, 2020-11-21:
Et qu'entendra vous de tenter de changer de VPN? Ou faire ça dans les configurations?
Merci

Ysurac, 2020-11-21:
Il me faudrait le fichier ```/etc/shorewall/rules``` sur le VPS.
(Pour changer de VPN, c'est dans Système->OpenMPTCProuter, la case "paramètres avancés")

JeedHome44, 2020-11-21:
Dans "Plugin de brouillage", j'ai le choix uniquement entre v2ray et simple-obfs.

C'est ça?

JeedHome44, 2020-11-21:
Je vois qu'il y a une option "Désactiver le test Ping vers le serveur". Ça n'est pas ce test qui pose problème?

JeedHome44, 2020-11-21:
Voici le fichier du VPS demandé. Ca ne serait pas la modification faites pour Pi-hole qui pose problème?

#
# Shorewall version 4.0 - Sample Rules File for two-interface configuration.
# Copyright (C) 2006-2014,2007 by the Shorewall Team
#
# This library is free software; you can redistribute it and/or
# modify it under the terms of the GNU Lesser General Public
# License as published by the Free Software Foundation; either
# version 2.1 of the License, or (at your option) any later version.
#
# See the file README.txt for further details.
#------------------------------------------------------------------------------
# For information about entries in this file, type "man shorewall-rules"
######################################################################################################################################################################################################
#ACTION         SOURCE          DEST            PROTO   DEST    SOURCE          ORIGINAL        RATE            USER/   MARK    CONNLIMIT       TIME            HEADERS         SWITCH          HELPER
#                                                       PORT    PORT(S)         DEST            LIMIT           GROUP
?SECTION ALL
?SECTION ESTABLISHED
?SECTION RELATED
?SECTION INVALID
?SECTION UNTRACKED
?SECTION NEW

#       Don't allow connection pickup from the net
#
Invalid(DROP)   net             all             tcp
#
#       Accept DNS connections from the firewall to the network
#
DNS(ACCEPT)     $FW             net
#
#       Allow Ping from/to the VPN
#
Ping(ACCEPT)    vpn             $FW
Ping(ACCEPT)    $FW             vpn
#
#       Allow Ping from the firewall to the network
#
Ping(ACCEPT)    $FW             net
#
# Drop Ping from the "bad" net zone.. and prevent your log from being flooded..
#
#Ping(DROP)     net             $FW
Ping(ACCEPT)    net             $FW
#
#       Accept connection from port > 65000 for shadowsocks and glorytun on the firewall
#
ACCEPT          net             $FW             tcp     65000-65535
ACCEPT          net             $FW             udp     65000-65535
#
#       Accept connection from SSH to the firewall
#
ACCEPT          net             $FW             tcp     65222
#
#       DHCP forward to the VPN from the firewall
#
DHCPfwd(ACCEPT) $FW             vpn
#
#       Redirect all port from 1 to 64999 to the VPN client from the network
#
#DNAT           net             vpn:$OMR_ADDR   tcp     1-64999
#DNAT           net             vpn:$OMR_ADDR   udp     1-64999
ACCEPT          net             $FW             tcp     80      # OMR openmptcprouter open shadowsocks port tcp

ACCEPT          net             $FW             tcp     65101   # OMR openmptcprouter open shadowsocks port tcp
ACCEPT          net             $FW             udp     65101   # OMR openmptcprouter open shadowsocks port udp
DNAT            net             vpn:$OMR_ADDR   tcp     35500   # OMR openmptcprouter redirect router 35500 port tcp
DNAT            net             vpn:$OMR_ADDR   udp     35500   # OMR openmptcprouter redirect router 35500 port udp
DNAT            net             vpn:$OMR_ADDR   tcp     8888    # OMR openmptcprouter redirect router 8888 port tcp
DNAT            net             vpn:$OMR_ADDR   tcp     9000    # OMR openmptcprouter redirect router 9000 port tcp
DNAT            net             vpn:$OMR_ADDR   tcp     10000   # OMR openmptcprouter redirect router 10000 port tcp
DNAT            net             vpn:$OMR_ADDR   udp     10000   # OMR openmptcprouter redirect router 10000 port udp
DNAT            net             vpn:$OMR_ADDR   tcp     49001   # OMR openmptcprouter redirect router 49001 port tcp
DNAT            net             vpn:$OMR_ADDR   udp     49001   # OMR openmptcprouter redirect router 49001 port udp
DNAT            net             vpn:$OMR_ADDR   tcp     35000   # OMR openmptcprouter redirect router 35000 port tcp
DNAT            net             vpn:$OMR_ADDR   udp     35000   # OMR openmptcprouter redirect router 35000 port udp
DNAT            net             vpn:$OMR_ADDR   tcp     35100   # OMR openmptcprouter redirect router 35100 port tcp
DNAT            net             vpn:$OMR_ADDR   udp     35100   # OMR openmptcprouter redirect router 35100 port udp
DNAT            net             vpn:$OMR_ADDR   tcp     36000   # OMR openmptcprouter redirect router 36000 port tcp
DNAT            net             vpn:$OMR_ADDR   udp     36000   # OMR openmptcprouter redirect router 36000 port udp
DNAT            net             vpn:$OMR_ADDR   tcp     32000   # OMR openmptcprouter redirect router 32000 port tcp
DNAT            net             vpn:$OMR_ADDR   udp     32000   # OMR openmptcprouter redirect router 32000 port udp
DNAT            net             vpn:$OMR_ADDR   tcp     49002   # OMR openmptcprouter redirect router 49002 port tcp
DNAT            net             vpn:$OMR_ADDR   udp     49002   # OMR openmptcprouter redirect router 49002 port udp
DNAT            net             vpn:$OMR_ADDR   tcp     7       # OMR openmptcprouter redirect router 7 port tcp
DNAT            net             vpn:$OMR_ADDR   udp     7       # OMR openmptcprouter redirect router 7 port udp
DNAT            net             vpn:$OMR_ADDR   tcp     10100   # OMR openmptcprouter redirect router 10100 port tcp
DNAT            net             vpn:$OMR_ADDR   udp     10100   # OMR openmptcprouter redirect router 10100 port udp
DNAT            net             vpn:$OMR_ADDR   udp     8888    # OMR openmptcprouter redirect router 8888 port udp
DNAT            net             vpn:$OMR_ADDR   udp     9000    # OMR openmptcprouter redirect router 9000 port udp

Ysurac, 2020-11-21:
Je parlais de l'onglet "assistant de configuration" et de la case "paramètres avancés"
Beaucoup de ports redirigés mais pas de soucis à ce niveau, même si il est étrange de rediriger le port 7.
Pour Pi-Hole il n'y a rien dans le firewall.

JeedHome44, 2020-11-21:
Pour que je puisse consulter Pi-hole par l'ip du VPS, tu m'avais fait ajouter une ligne sur le VPS pour le pour 80.
Ce n'est pas ça qui pose problème?
Et pour les ports de ce fichier, je n'ai rien ajouté manuellement. Donc je ne sais pas pourquoi il y en a autant.
C'est peut être la que j'ai ajouté une ligne pour Pi-hole comme tu m'avais conseillé

Ysurac, 2020-11-21:
L'ouverture des ports est faites depuis la configuration du routeur.
Pour ta régle pour voir le port 80, elle est bien présente.

JeedHome44, 2020-11-21:
En allant sur Configuration avancée du serveur, je constate que "ID utilisateur V2Ray est vide.
Il ne me semblait pas que c'était vide avant la mise à jour.
Je ne l'ai pas touché.
Il est là le problème?

JeedHome44, 2020-11-21:
Je ne comprends pas. J'essaie de changer le VPN par défaut pour voir si ça change quelque chose mais quand je clique sur sauvegarder et appliquer, la page tourne mais ne charge pas j'ai l'impression. Ça ne s'applique pas. Il y a donc un problème sur OMR?

JeedHome44, 2020-11-21:
Encore mieux. Ça ne chargeait plus et la, impossible d'accéder à OMR via 192.168.100.1
Ça sent un gros bug de mise à jour ça non?

JeedHome44, 2020-11-21:
J'ai de nouveau l'accès à l'interface OMR. J'ai clique sur forcer la récupération de toutes les clés du serveur et j'ai de nouveau l'ID utilisateur V2Ray.

Est-ce que ça pouvait être ça le problème?

Ysurac, 2020-11-21:
Lors de l'application des changements, selon les changements, ça relance le réseau d'où la coupure/lenteur parfois.
Pour l'utilisateur v2ray ça a de l'importance que si on utilise le proxy V2Ray, par défaut shadowsocks est utilisé.

JeedHome44, 2020-11-21:
Bon, je ne comprends plus rien. sur l'aperçu réseau, tout est au vert maintenant. mais là c'est Netflix qui ne marche plus à la maison alors que les By-Pass sont toujours actifs. je n'ai plus accès par Putty au VPS et plus d'accès à Pi-hole ni interne, ni externe. l'ip externe du serveur ne marche plus non plus pour accèder à mon home assistant octoprint ou tout autre appareil sur lequel j'ai mis une redirection de port.... je regrette vraiment d'avoir mis à jour là... une vraie prise de tête alors que tout roulait parfaitement avant...

JeedHome44, 2020-11-21:
J'ai reboot OMR et la j'ai tout au rouge maintenant dans Aperçu réseau.

Freebox (192.168.0.1)
Aucune IP définie

Huawei (192.168.8.1)
Aucune IP définie

OpenMPTCProuteur
(192.168.100.1)
Le VPN ne répond pas

VPS
Pas de sortie
Pas d'adresse IP
Pas de réponse serveur...

Donc la je pige pas après un redémarrage c'est la cata.

Elle a vraiment un soucis cette mise à jour la... je fais comment maintenant?

JeedHome44, 2020-11-21:
[image]

JeedHome44, 2020-11-21:
Ca me fait ça dans les logs:

`Sat Nov 21 14:31:33 2020 daemon.notice netifd: omrip1gre (25746): Command failed: Unknown error
Sat Nov 21 14:31:33 2020 daemon.notice netifd: Interface 'omrip1gre' is now down
Sat Nov 21 14:31:33 2020 daemon.notice netifd: Interface 'omrip1gre' is setting up now
Sat Nov 21 14:31:33 2020 daemon.notice netifd: omrip1gre (25789): Command failed: Unknown error
Sat Nov 21 14:31:33 2020 daemon.notice netifd: Interface 'omrip1gre' is now down
Sat Nov 21 14:31:33 2020 daemon.notice netifd: Interface 'omrip1gre' is setting up now
Sat Nov 21 14:31:33 2020 daemon.notice netifd: omrip1gre (25822): Command failed: Unknown error
Sat Nov 21 14:31:33 2020 daemon.notice netifd: Interface 'omrip1gre' is now down
Sat Nov 21 14:31:33 2020 daemon.notice netifd: Interface 'omrip1gre' is setting up now
Sat Nov 21 14:31:33 2020 daemon.notice netifd: omrip1gre (25851): Command failed: Unknown error
Sat Nov 21 14:31:33 2020 daemon.notice netifd: Interface 'omrip1gre' is now down
Sat Nov 21 14:31:33 2020 daemon.notice netifd: Interface 'omrip1gre' is setting up now
Sat Nov 21 14:31:33 2020 daemon.notice netifd: omrip1gre (25884): Command failed: Unknown error
Sat Nov 21 14:31:33 2020 daemon.notice netifd: Interface 'omrip1gre' is now down
Sat Nov 21 14:31:33 2020 daemon.notice netifd: Interface 'omrip1gre' is setting up now
Sat Nov 21 14:31:33 2020 daemon.notice netifd: omrip1gre (25927): Command failed: Unknown error
Sat Nov 21 14:31:33 2020 daemon.notice netifd: Interface 'omrip1gre' is now down
Sat Nov 21 14:31:33 2020 daemon.notice netifd: Interface 'omrip1gre' is setting up now
Sat Nov 21 14:31:33 2020 daemon.notice netifd: omrip1gre (25960): Command failed: Unknown error
Sat Nov 21 14:31:33 2020 daemon.notice netifd: Interface 'omrip1gre' is now down
Sat Nov 21 14:31:33 2020 daemon.notice netifd: Interface 'omrip1gre' is setting up now
Sat Nov 21 14:31:33 2020 daemon.notice netifd: omrip1gre (25991): Command failed: Unknown error
Sat Nov 21 14:31:33 2020 daemon.err /usr/bin/ss-local[15254]: getpeername: Socket not connected
Sat Nov 21 14:31:33 2020 daemon.notice netifd: Interface 'omrip1gre' is now down
Sat Nov 21 14:31:33 2020 daemon.notice netifd: Interface 'omrip1gre' is setting up now
Sat Nov 21 14:31:33 2020 daemon.notice netifd: omrip1gre (26033): Command failed: Unknown error
Sat Nov 21 14:31:33 2020 daemon.notice netifd: Interface 'omrip1gre' is now down
Sat Nov 21 14:31:33 2020 daemon.notice netifd: Interface 'omrip1gre' is setting up now
Sat Nov 21 14:31:33 2020 daemon.notice netifd: omrip1gre (26085): Command failed: Unknown error
Sat Nov 21 14:31:33 2020 daemon.notice netifd: Interface 'omrip1gre' is now down
Sat Nov 21 14:31:33 2020 daemon.notice netifd: Interface 'omrip1gre' is setting up now
Sat Nov 21 14:31:33 2020 daemon.notice netifd: omrip1gre (26120): Command failed: Unknown error
Sat Nov 21 14:31:33 2020 daemon.notice netifd: Interface 'omrip1gre' is now down
Sat Nov 21 14:31:33 2020 daemon.notice netifd: Interface 'omrip1gre' is setting up now`

Kalimeiro, 2020-11-21:
Bonjour,

Dans "Système" puis "OpenMPTCProuter" va dans l'onglet "Assistant de configuration", coche la case "Configuration avancée", et configure comme suit :

- Paramètre du Proxy : utilise Shadowsocks.
- Paramètre du VPN : Glorytun TCP

Applique et sauvegarde, redémarre le OMR, reconnecte toi à l'interface WEB.

Retourne dans Dans "Système" puis "OpenMPTCProuter" et va dans l'onglet "Paramètres avancés", coche la case "Activer le brouillage pour Shadowsocks" et choisi "V2RAY" et le type de brouillage "HTTP" puis dans "Autres paramètres" plus bas assure toi que "Sélection de l'interface maitre" et en équilibrage, si tu as toujours des problèmes de non réponse de tes routeurs alors choisi "En cas de changements dans l'assistant" et assure toi d'avoir en maître un routeur stable tant du point de vue LAN que de l'accès internet fourni par cet équipement.

Ensuite : "Services" puis "Shadowsocks-libev" onglet "Serveurs distants" efface tout ce qui concerne OMRxGRE, puis la même chose dans "Règles de redirection", pour terminer va dans "Réseau" puis "Interfaces", et supprime tout ce qui concerne OMRxGRE (il y en a 4), applique et sauvegarde, redémarre ton OMR.

De rien.

JeedHome44, 2020-11-21:
Salut.

J'ai bien fait de faire un BackUp avant la mise à jour 56.5

Vu que rien ne marchait, j'ai remis le BackUp 56.1 et c'est reparti.

Pour l'instant, ça tient j'ai l'impression.

C'est quoi ce OMRxGre?

Ysurac, 2020-11-21:
Les interfaces OMRxGRE sont créées quand le serveur possède plusieurs adresses IPs.

JeedHome44, 2020-11-21:
Merci pour tes explications.
En effet, quand j'avais les problèmes, j'avais vu ce OMRxGre

La, vu que ça ne marchait plus, en mettant le BackUp fait en 56.1, je ne vois plus ces OMRxGre.

Et bizarrement, tout est redevenu très stable.

C'était donc ça qui fait un problème dans OMR.

Par curiosité, j'aimerai comprendre ce que c'est! Ah je pense que tu m'as répondu entre temps! 😉

JeedHome44, 2020-11-21:
Le serveur possède plusieurs IP?

Le miens ne possède qu'une adresse IP. Pourquoi ça c'est donc créé?

C'est donc ça le problème que j'avais depuis le début?

Et je n'ai pas activé le brouillage ShadowSocks. Est-ce important de l'activer?

Merci pour votre aide! On a enfin mis le doigt sur ce qui n'allait pas on dirait

Ysurac, 2020-11-21:
Quand plusieurs IPs publiques sont détectées les interfaces GRE sont créées. Tu peux voir dans ```/etc/openmptcprouter-vps-admin/omr-admin-config.json``` à quoi ça correspond. Si il y a rien tu peux voir dans services->shadowsocks-libev côté routeur à quoi ça peut correspondre.

Pour le brouillage ça sert quand on est dans des pays où le trafic est surveillé, par exemple en Chine. Sinon il y a peu d'intérêt. Il y a encore moins d'intérêt depuis que le proxy v2ray est disponible directement.

JeedHome44, 2020-11-21:
Merci.
Ça pouvait donc expliquer mes problèmes?
En remettant la sauvegarde faites en 56.1 avant de mettre à jour, tout marche bien sans problème là.

JeedHome44, 2020-11-22:
Bon, ce matin, ça recommence encore et ça déconnecte du serveur après 1 seconde sur la 4G.
Mais il y a un truc que je ne comprends pas.
Avant, avec la 4G, j'avais toujours la même IP externe. La je vois qu'elle change de temps en temps.
Ma 4G vient de Free. Ne seraient ils pas en train de modifier leurs antennes pour la 5G?
Surtout que l'antenne proche de chez moi est récente donc je pense qu'elle doit sûrement être pré-équipée.

kevin39, 2020-11-22:
Si ton IP de sortie en 4G change régulièrement c'est que ta 4G a effectivement des coupures mais ce n'est pas la faute d'OMR. Après avec OMR c'est censé être transparent, tu n'es pas censée ressentir les déconnexions vu que tu as plusieurs lignes.

Quant à la 5G, je pense qu'il est peu probable que se soit des tests sur ton antenne même si techniquement c'est sans doute possible.

JeedHome44, 2020-11-22:
J'ai mis le master sur l'ADSL et ça fait pareil. Ce n'est donc pas la connexion 4G qui est la cause. Il y a donc encore quelque chose qui ne va pas avec cette mise à jour... je commence à désespérer là...

JeedHome44, 2020-11-22:
Et j'ai vérifié, l'ip ne change pas continuellement non plus et les erreur de communication serveur ont lieu avec la même ip 4g. Ce n'est donc pas le problème. Tout marchait parfaitement en 56.1 et VPS 18...

kevin39, 2020-11-22:
Si tu penses que la MAj a un causé des problèmes, fais une sauvegarde (du routeur) puis un reset. Reconfigure ensuite tes interfaces comme une 1ère installation. Tu verras bien si cela change quelque chose. Tu pourras restaurer ta sauvegarde si tu tu te trompe ou n'y arrive pas.

JeedHome44, 2020-11-22:
Là ça veut dire tout refaire et en ce moment pas trop le temps. ça me prend déjà la tête depuis plusieurs jours et toute la famille commence à en avoir un peu marre que tout bug depuis que j'ai fait cette maj... là ça marche à moitié donc pour l'instant je vais laisser comme ca... je ne pensais pas galérer autant avec une mise à jour qui doit apporter plus de solutions que de problèmes normalement... :(

[24 more replies]

closed 2021-02-26 by github-actions[bot]