Ysurac/openmptcprouter #1399: problem after upgrade

fdaforno, 2020-11-20:
Hi all
After upgrade to the latest release, I get this error 

I tried many times to run again the script on the VPS but nothing changed

OMR-VPS: Can't get token, try later (can ping server, server API answer)
OMR-VPS: Can't get token, try later (can ping server, server API answer)
OMR-VPS: Can't get token, try later (can ping server, server API answer)
OMR-VPS: Can't get token, try later (can ping server, server API answer)

  - OpenMPTCProuter version: v0.56.5 r0+14653-a439f1bb47
  - OpenMPTCProuter VPS version: v0.56.5
  - OpenMPTCProuter platform: x86

fdaforno, 2020-11-20:
ov 20 01:49:54 mail omr-admin.py[34708]: INFO:     93.43.200.142:54508 - "POST /token HTTP/1.1" 400 Bad Request
Nov 20 01:49:56 mail omr-admin.py[34708]: INFO:     93.43.200.142:54510 - "GET / HTTP/1.1" 200 OK
Nov 20 01:49:56 mail omr-admin.py[34708]: DEBUG:    Calling on_field_start with no data
Nov 20 01:49:56 mail omr-admin.py[34708]: DEBUG:    Calling on_field_name with data[0:8]
Nov 20 01:49:56 mail omr-admin.py[34708]: DEBUG:    Calling on_field_data with data[9:24]
Nov 20 01:49:56 mail omr-admin.py[34708]: DEBUG:    Calling on_field_end with no data
Nov 20 01:49:56 mail omr-admin.py[34708]: DEBUG:    Calling on_field_start with no data
Nov 20 01:49:56 mail omr-admin.py[34708]: DEBUG:    Calling on_field_name with data[25:33]
Nov 20 01:49:56 mail omr-admin.py[34708]: DEBUG:    Calling on_field_data with data[34:98]
Nov 20 01:49:56 mail omr-admin.py[34708]: DEBUG:    Calling on_field_end with no data
Nov 20 01:49:56 mail omr-admin.py[34708]: DEBUG:    Calling on_end with no data
Nov 20 01:49:56 mail omr-admin.py[34708]: DEBUG:    wrong password
Nov 20 01:49:56 mail omr-admin.py[34708]: DEBUG:    Incorrect username or password

fdaforno, 2020-11-20:
i found the problem

closed 2020-11-20 by fdaforno

xobtoor, 2021-06-09:
Do you remember what the issue was, because we're facing on our login service the exact same issue.