Ysurac/openmptcprouter #1400: VPN NE REPOND PAS

matthieu71000, 2020-11-20:
## Expected Behavior
Hi, I need help please

After Update Router and VPS

Message  : Le VPN ne répond pas

## Current Behavior

external connection don't work after update

## Specifications

  - OpenMPTCProuter version: 0.56.5
  - OpenMPTCProuter VPS version: Version 0.1022 5.4.74
  - OpenMPTCProuter platform: Raspberry Pi 3 Model B Rev 1.2

Ysurac, 2020-11-20:
What is the VPN used ? What do you have in log, in State->System log ?

matthieu71000, 2020-11-20:
Glorytun TCP

Fri Nov 20 13:15:20 2020 user.notice ModemManager: hotplug: remove network interface tun0: event processed
Fri Nov 20 13:15:20 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=tun0, subsystem=net
Fri Nov 20 13:15:20 2020 daemon.err glorytun[10118]: AES-256-GCM is not available on your platform
Fri Nov 20 13:15:20 2020 daemon.info glorytun[10118]: INITIALIZED tun0
Fri Nov 20 13:15:20 2020 daemon.notice netifd: Interface 'omrvpn' is disabled
Fri Nov 20 13:15:20 2020 daemon.notice netifd: Interface 'omrvpn' is enabled
Fri Nov 20 13:15:20 2020 daemon.notice netifd: Interface 'omrvpn' is disabled
Fri Nov 20 13:15:20 2020 daemon.warn dnsmasq[26028]: failed to create listening socket for fe80::55ed:9523:2612:d9c%tun0: Address not available
Fri Nov 20 13:15:20 2020 daemon.warn dnsmasq[26028]: failed to create listening socket for fe80::55ed:9523:2612:d9c%tun0: Address not available
Fri Nov 20 13:15:20 2020 daemon.notice netifd: Interface 'omrvpn' is enabled
Fri Nov 20 13:15:20 2020 daemon.notice netifd: Network device 'tun0' link is up
Fri Nov 20 13:15:20 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Fri Nov 20 13:15:20 2020 daemon.notice netifd: Interface 'omrvpn' is setting up now
Fri Nov 20 13:15:20 2020 daemon.notice netifd: Interface 'omrvpn' is now up
Fri Nov 20 13:15:20 2020 user.notice ModemManager: hotplug: add network interface tun0: event processed
Fri Nov 20 13:15:20 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tun0, subsystem=net
Fri Nov 20 13:15:20 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Nov 20 13:15:20 2020 user.notice omr-tracker: Launching...
Fri Nov 20 13:15:21 2020 user.notice omr-tracker: Launched
Fri Nov 20 13:15:21 2020 daemon.err glorytun[10118]: 51.75.27.35.65001: connected
Fri Nov 20 13:15:21 2020 user.notice post-tracking-post-tracking: omrvpn down
Fri Nov 20 13:15:21 2020 daemon.info glorytun[10118]: STARTED tun0
Fri Nov 20 13:15:21 2020 daemon.info vnstatd[29209]: Info: SIGTERM received, exiting.
Fri Nov 20 13:15:21 2020 daemon.info vnstatd[10730]: Info: vnStat daemon 2.6 started. (pid:10730 uid:0 gid:0 64-bit)
Fri Nov 20 13:15:21 2020 daemon.info vnstatd[10730]: Info: Monitoring (4): wan4 (100 Mbit) wan3 (100 Mbit) wan2 (100 Mbit) wan1 (100 Mbit)
Fri Nov 20 13:15:21 2020 daemon.info glorytun[10118]: STOPPED tun0
Fri Nov 20 13:15:21 2020 daemon.notice netifd: Network device 'tun0' link is down
Fri Nov 20 13:15:21 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Fri Nov 20 13:15:21 2020 daemon.notice netifd: Interface 'omrvpn' is now down
Fri Nov 20 13:15:21 2020 daemon.info glorytun: starting glorytun vpn instance vpn
Fri Nov 20 13:15:21 2020 user.notice ModemManager: hotplug: remove network interface tun0: event processed
Fri Nov 20 13:15:21 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=tun0, subsystem=net
Fri Nov 20 13:15:21 2020 daemon.err glorytun[10809]: AES-256-GCM is not available on your platform
Fri Nov 20 13:15:21 2020 daemon.info glorytun[10809]: INITIALIZED tun0
Fri Nov 20 13:15:21 2020 daemon.notice netifd: Interface 'omrvpn' is disabled
Fri Nov 20 13:15:21 2020 daemon.notice netifd: Interface 'omrvpn' is enabled
Fri Nov 20 13:15:21 2020 daemon.debug dnsmasq[26028]: listening on tun0(#36): fe80::f56c:ab72:8916:9a55%tun0 port 53
Fri Nov 20 13:15:21 2020 daemon.notice netifd: Interface 'omrvpn' is disabled
Fri Nov 20 13:15:21 2020 daemon.debug dnsmasq[26028]: stopped listening on tun0(#36): fe80::f56c:ab72:8916:9a55%tun0 port 53
Fri Nov 20 13:15:21 2020 daemon.notice netifd: Interface 'omrvpn' is enabled
Fri Nov 20 13:15:21 2020 daemon.notice netifd: Network device 'tun0' link is up
Fri Nov 20 13:15:21 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Fri Nov 20 13:15:21 2020 daemon.notice netifd: Interface 'omrvpn' is setting up now
Fri Nov 20 13:15:21 2020 daemon.notice netifd: Interface 'omrvpn' is now up
Fri Nov 20 13:15:21 2020 user.notice ModemManager: hotplug: add network interface tun0: event processed
Fri Nov 20 13:15:21 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tun0, subsystem=net
Fri Nov 20 13:15:21 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Nov 20 13:15:22 2020 daemon.err glorytun[10809]: 51.75.27.35.65001: connected
Fri Nov 20 13:15:22 2020 daemon.info glorytun[10809]: STARTED tun0
Fri Nov 20 13:15:23 2020 user.notice post-tracking-post-tracking: Master up : Replace default route by 192.168.102.1 dev wan2
Fri Nov 20 13:15:26 2020 daemon.info glorytun[10809]: STOPPED tun0
Fri Nov 20 13:15:29 2020 daemon.err glorytun[10809]: 51.75.27.35.65001: connected
Fri Nov 20 13:15:29 2020 daemon.info glorytun[10809]: STARTED tun0
Fri Nov 20 13:15:36 2020 daemon.err /usr/bin/ss-local[10060]: getpeername: Socket not connected
Fri Nov 20 13:15:37 2020 daemon.err /usr/bin/ss-redir[10061]: remote recv: Operation timed out
Fri Nov 20 13:15:39 2020 daemon.err /usr/bin/ss-local[10060]: getpeername: Socket not connected
Fri Nov 20 13:15:47 2020 daemon.err uhttpd[5995]: error: couldn't find the ModemManager process in the bus
Fri Nov 20 13:15:48 2020 daemon.err uhttpd[5995]: error: couldn't find the ModemManager process in the bus
Fri Nov 20 13:15:54 2020 daemon.err /usr/bin/ss-local[10060]: getpeername: Socket not connected
[... cut ...]
Fri Nov 20 13:25:48 2020 daemon.err /usr/bin/ss-redir[29859]: remote recv: Operation timed out
Fri Nov 20 13:26:05 2020 daemon.err uhttpd[5995]: luci: accepted login on / for root from 192.168.1.61
Fri Nov 20 13:26:10 2020 daemon.err /usr/bin/ss-redir[29859]: remote recv: Operation timed out
Fri Nov 20 13:26:20 2020 daemon.info glorytun[30614]: STOPPED tun0
Fri Nov 20 13:26:20 2020 daemon.err glorytun[30614]: 51.75.27.35.65001: connected
Fri Nov 20 13:26:20 2020 daemon.info glorytun[30614]: STARTED tun0

Ysurac, 2020-11-20:
The server was rebooted ?
You seems to use AES but AES doesn't seems to be supported hardware.
Did you try to validate the wizard again ?

matthieu71000, 2020-11-20:
Yes I restart the server
AES ?? qu'est ce que c'est ?
Yes I try to validate wizard again and again.

matthieu71000, 2020-11-20:
I just do a clean install before backup
Start factory and restore my Backup
I don't change anything for AES

Ysurac, 2020-11-20:
It's in System->OpenMPTCProuter, in the wizard the "advanced settings" checkbox, set the encryption to chacha20.

matthieu71000, 2020-11-20:
Nov 20 15:10:34 vps690679 glorytun-tcp-run[1609]: 193.252.XX.XX.18518: connected
Nov 20 15:10:34 vps690679 glorytun-tcp-run[1609]: peer wants chacha20
Nov 20 15:10:45 vps690679 ss-server[481]: failed to handshake with 193.252.XX: invalid address type
Nov 20 15:10:54 vps690679 ss-server[481]: failed to handshake with 193.252.XX: invalid address type
Nov 20 15:10:59 vps690679 ss-server[481]: failed to handshake with 193.252.XX: invalid address type
Nov 20 15:11:03 vps690679 omr-service[475]: perl: warning: Setting locale failed.
Nov 20 15:11:03 vps690679 omr-service[475]: perl: warning: Please check that your locale settings:
Nov 20 15:11:03 vps690679 omr-service[475]: #011LANGUAGE = (unset),
Nov 20 15:11:03 vps690679 omr-service[475]: #011LC_ALL = (unset),
Nov 20 15:11:03 vps690679 omr-service[475]: #011LANG = "fr_FR.UTF-8"
Nov 20 15:11:03 vps690679 omr-service[475]:     are supported and installed on your system.
Nov 20 15:11:03 vps690679 omr-service[475]: perl: warning: Falling back to the standard locale ("C").
Nov 20 15:11:06 vps690679 ss-server[481]: getpeername: Transport endpoint is not connected
Nov 20 15:11:10 vps690679 ss-server[481]: getpeername: Transport endpoint is not connected
Nov 20 15:11:14 vps690679 ss-server[481]: failed to handshake with 193.252.XX: invalid host name
Nov 20 15:11:14 vps690679 ss-server[481]: failed to handshake with 193.252.XX: invalid address type
Nov 20 15:11:18 vps690679 OMR-Service: No answer from VPN client end, restart Glorytun-TCP
Nov 20 15:11:18 vps690679 glorytun-tcp-run[1609]: INITIALIZED gt-tun0
Nov 20 15:11:18 vps690679 glorytun-tcp-run[1609]: STARTED gt-tun0
Nov 20 15:11:18 vps690679 glorytun-tcp-run[1609]: STOPPED gt-tun0
Nov 20 15:11:18 vps690679 systemd-networkd[218]: gt-tun0: Lost carrier
Nov 20 15:11:18 vps690679 systemd[1]: Stopping Glorytun TCP on tun0...
Nov 20 15:11:18 vps690679 systemd[1]: glorytun-tcp@tun0.service: Succeeded.
Nov 20 15:11:18 vps690679 systemd[1]: Stopped Glorytun TCP on tun0.
Nov 20 15:11:18 vps690679 systemd[1]: Starting Glorytun TCP on tun0...
Nov 20 15:11:18 vps690679 systemd-udevd[1761]: link_config: autonegotiation is unset or enabled, the speed and duplex are not writable.
Nov 20 15:11:18 vps690679 systemd-networkd[218]: gt-tun0: Gained carrier
Nov 20 15:11:18 vps690679 systemd-networkd[218]: gt-tun0: Gained IPv6LL
Nov 20 15:11:18 vps690679 systemd[1]: Started Glorytun TCP on tun0.
Nov 20 15:11:19 vps690679 glorytun-tcp-run[1754]: 193.252.XX.18659: connected
Nov 20 15:11:19 vps690679 glorytun-tcp-run[1754]: peer wants chacha20

matthieu71000, 2020-11-20:
ov 20 15:15:04 vps690679 OMR-Service: No answer from VPN client end, restart Glorytun-TCP
Nov 20 15:15:04 vps690679 systemd[1]: Stopping Glorytun TCP on tun0...
Nov 20 15:15:04 vps690679 glorytun-tcp-run[16013]: INITIALIZED gt-tun0
Nov 20 15:15:04 vps690679 glorytun-tcp-run[16013]: STARTED gt-tun0
Nov 20 15:15:04 vps690679 glorytun-tcp-run[16013]: STOPPED gt-tun0
Nov 20 15:15:04 vps690679 systemd-networkd[218]: gt-tun0: Lost carrier
Nov 20 15:15:04 vps690679 systemd[1]: glorytun-tcp@tun0.service: Succeeded.
Nov 20 15:15:04 vps690679 systemd[1]: Stopped Glorytun TCP on tun0.
Nov 20 15:15:04 vps690679 systemd[1]: Starting Glorytun TCP on tun0...
Nov 20 15:15:04 vps690679 systemd-udevd[16229]: link_config: autonegotiation is unset or enabled, the speed and duplex are not writable.
Nov 20 15:15:04 vps690679 systemd-networkd[218]: gt-tun0: Gained carrier
Nov 20 15:15:04 vps690679 systemd-networkd[218]: gt-tun0: Gained IPv6LL
Nov 20 15:15:04 vps690679 systemd[1]: Started Glorytun TCP on tun0.
Nov 20 15:15:05 vps690679 glorytun-tcp-run[16218]: 193.252.XX.XX.21012: connected
Nov 20 15:15:17 vps690679 v2ray[408]: 2020/11/20 15:15:17 127.0.0.1:37725 accepted tcp:127.0.0.1:0 [api]
Nov 20 15:15:17 vps690679 v2ray[408]: 2020/11/20 15:15:17 127.0.0.1:37727 accepted tcp:127.0.0.1:0 [api]
Nov 20 15:15:34 vps690679 omr-service[475]: perl: warning: Setting locale failed.
Nov 20 15:15:34 vps690679 omr-service[475]: perl: warning: Please check that your locale settings:
Nov 20 15:15:34 vps690679 omr-service[475]: #011LANGUAGE = (unset),
Nov 20 15:15:34 vps690679 omr-service[475]: #011LC_ALL = (unset),
Nov 20 15:15:34 vps690679 omr-service[475]: #011LANG = "fr_FR.UTF-8"
Nov 20 15:15:34 vps690679 omr-service[475]:     are supported and installed on your system.
Nov 20 15:15:34 vps690679 omr-service[475]: perl: warning: Falling back to the standard locale ("C").
Nov 20 15:15:37 vps690679 v2ray[408]: 2020/11/20 15:15:37 127.0.0.1:37755 accepted tcp:127.0.0.1:0 [api]
Nov 20 15:15:37 vps690679 v2ray[408]: 2020/11/20 15:15:37 127.0.0.1:37757 accepted tcp:127.0.0.1:0 [api]
Nov 20 15:15:50 vps690679 OMR-Service: No answer from VPN client end, restart Glorytun-TCP
Nov 20 15:15:50 vps690679 glorytun-tcp-run[16218]: INITIALIZED gt-tun0
Nov 20 15:15:50 vps690679 glorytun-tcp-run[16218]: STARTED gt-tun0
Nov 20 15:15:50 vps690679 glorytun-tcp-run[16218]: STOPPED gt-tun0
Nov 20 15:15:50 vps690679 systemd-networkd[218]: gt-tun0: Lost carrier
Nov 20 15:15:50 vps690679 systemd[1]: Stopping Glorytun TCP on tun0...
Nov 20 15:15:50 vps690679 systemd[1]: glorytun-tcp@tun0.service: Succeeded.
Nov 20 15:15:50 vps690679 systemd[1]: Stopped Glorytun TCP on tun0.
Nov 20 15:15:50 vps690679 systemd[1]: Starting Glorytun TCP on tun0...
Nov 20 15:15:50 vps690679 systemd-networkd[218]: gt-tun0: Gained carrier
Nov 20 15:15:50 vps690679 systemd-udevd[16419]: link_config: autonegotiation is unset or enabled, the speed and duplex are not writable.
Nov 20 15:15:50 vps690679 systemd-networkd[218]: gt-tun0: Gained IPv6LL
Nov 20 15:15:50 vps690679 systemd[1]: Started Glorytun TCP on tun0.
Nov 20 15:15:50 vps690679 ss-server[1981]: getpeername: Transport endpoint is not connected
Nov 20 15:15:57 vps690679 v2ray[408]: 2020/11/20 15:15:57 127.0.0.1:37775 accepted tcp:127.0.0.1:0 [api]
Nov 20 15:15:57 vps690679 v2ray[408]: 2020/11/20 15:15:57 127.0.0.1:37777 accepted tcp:127.0.0.1:0 [api]
Nov 20 15:16:04 vps690679 glorytun-tcp-run[16421]: 193.252.XX.XX.21303: connected
Nov 20 15:16:20 vps690679 omr-service[475]: perl: warning: Setting locale failed.
Nov 20 15:16:20 vps690679 omr-service[475]: perl: warning: Please check that your locale settings:
Nov 20 15:16:20 vps690679 omr-service[475]: #011LANGUAGE = (unset),
Nov 20 15:16:20 vps690679 omr-service[475]: #011LC_ALL = (unset),
Nov 20 15:16:20 vps690679 omr-service[475]: #011LANG = "fr_FR.UTF-8"
Nov 20 15:16:20 vps690679 omr-service[475]:     are supported and installed on your system.
Nov 20 15:16:20 vps690679 omr-service[475]: perl: warning: Falling back to the standard locale ("C").
Nov 20 15:16:35 vps690679 OMR-Service: No answer from VPN client end, restart Glorytun-TCP
Nov 20 15:16:35 vps690679 glorytun-tcp-run[16421]: INITIALIZED gt-tun0
Nov 20 15:16:35 vps690679 glorytun-tcp-run[16421]: STARTED gt-tun0
Nov 20 15:16:35 vps690679 glorytun-tcp-run[16421]: STOPPED gt-tun0
Nov 20 15:16:35 vps690679 systemd-networkd[218]: gt-tun0: Lost carrier
Nov 20 15:16:35 vps690679 systemd[1]: Stopping Glorytun TCP on tun0...
Nov 20 15:16:35 vps690679 systemd[1]: glorytun-tcp@tun0.service: Succeeded.
Nov 20 15:16:35 vps690679 systemd[1]: Stopped Glorytun TCP on tun0.
Nov 20 15:16:35 vps690679 systemd[1]: Starting Glorytun TCP on tun0...
Nov 20 15:16:35 vps690679 systemd-udevd[16604]: link_config: autonegotiation is unset or enabled, the speed and duplex are not writable.
Nov 20 15:16:35 vps690679 systemd-networkd[218]: gt-tun0: Gained carrier
Nov 20 15:16:35 vps690679 systemd-networkd[218]: gt-tun0: Gained IPv6LL
Nov 20 15:16:35 vps690679 systemd[1]: Started Glorytun TCP on tun0.
Nov 20 15:16:36 vps690679 v2ray[408]: 2020/11/20 15:16:36 127.0.0.1:37831 accepted tcp:127.0.0.1:0 [api]
Nov 20 15:16:36 vps690679 v2ray[408]: 2020/11/20 15:16:36 127.0.0.1:37833 accepted tcp:127.0.0.1:0 [api]
Nov 20 15:16:43 vps690679 glorytun-tcp-run[16597]: 193.252.XX.XX.21461: connected
Nov 20 15:16:48 vps690679 ss-server[1981]: getpeername: Transport endpoint is not connected
[... cut ...]
Nov 20 15:17:05 vps690679 omr-service[475]: perl: warning: Setting locale failed.
Nov 20 15:17:05 vps690679 omr-service[475]: perl: warning: Please check that your locale settings:
Nov 20 15:17:05 vps690679 omr-service[475]: #011LANGUAGE = (unset),
Nov 20 15:17:05 vps690679 omr-service[475]: #011LC_ALL = (unset),
Nov 20 15:17:05 vps690679 omr-service[475]: #011LANG = "fr_FR.UTF-8"
Nov 20 15:17:05 vps690679 omr-service[475]:     are supported and installed on your system.
Nov 20 15:17:05 vps690679 omr-service[475]: perl: warning: Falling back to the standard locale ("C").

matthieu71000, 2020-11-20:
> It's in System->OpenMPTCProuter, in the wizard the "advanced settings" checkbox, set the encryption to chacha20.

Le VPN ne répond pas

Ysurac, 2020-11-20:
On the VPS: ```systemctl stop omr```.

matthieu71000, 2020-11-20:
> On the VPS: `systemctl stop omr`.

done

matthieu71000, 2020-11-20:
Nov 20 15:41:00 vps690699 shorewall[2517]: Creating iptables-restore input...
Nov 20 15:41:00 vps690699 shorewall[2517]: Compiling /etc/shorewall/stoppedrules...
Nov 20 15:41:00 vps690699 shorewall[2517]: Shorewall configuration compiled to /var/lib/shorewall/.reload
Nov 20 15:41:00 vps690699 shorewall[2517]: Reloading Shorewall....
Nov 20 15:41:00 vps690699 shorewall[2517]: Initializing...
Nov 20 15:41:00 vps690699 shorewall[2517]: Setting up Route Filtering...
Nov 20 15:41:00 vps690699 shorewall[2517]: Setting up Martian Logging...
Nov 20 15:41:00 vps690699 shorewall[2517]: Setting up Accept Source Routing...
Nov 20 15:41:00 vps690699 shorewall[2517]: Setting up Traffic Control...
Nov 20 15:41:00 vps690699 shorewall[2517]: Preparing iptables-restore input...
Nov 20 15:41:00 vps690699 shorewall[2517]: Running /sbin/iptables-restore --wait 60...
Nov 20 15:41:00 vps690699 shorewall[2517]: IPv4 Forwarding Enabled
Nov 20 15:41:00 vps690699 root: Shorewall reloaded
Nov 20 15:41:00 vps690699 shorewall[2517]: done.
Nov 20 15:41:01 vps690699 systemd[1]: Reloaded Shorewall IPv4 firewall.
Nov 20 15:41:01 vps690699 systemd[1]: Reloading Shorewall IPv4 firewall.
Nov 20 15:41:01 vps690699 shorewall[2810]: Compiling using Shorewall 5.2.3.2...
Nov 20 15:41:01 vps690699 shorewall[2810]: perl: warning: Setting locale failed.
Nov 20 15:41:01 vps690699 shorewall[2810]: perl: warning: Please check that your locale settings:
Nov 20 15:41:01 vps690699 shorewall[2810]: #011LANGUAGE = (unset),
Nov 20 15:41:01 vps690699 shorewall[2810]: #011LC_ALL = (unset),
Nov 20 15:41:01 vps690699 shorewall[2810]: #011LANG = "fr_FR.UTF-8"
Nov 20 15:41:01 vps690699 shorewall[2810]:     are supported and installed on your system.
Nov 20 15:41:01 vps690699 shorewall[2810]: perl: warning: Falling back to the standard locale ("C").
Nov 20 15:41:01 vps690699 shorewall[2810]: Processing /etc/shorewall/params ...
Nov 20 15:41:01 vps690699 ss-server[1981]: server recv: Connection reset by peer
Nov 20 15:41:01 vps690699 shorewall[2810]: Processing /etc/shorewall/shorewall.conf...
Nov 20 15:41:01 vps690699 shorewall[2810]: Loading Modules...
Nov 20 15:41:01 vps690699 shorewall[2810]: Compiling /etc/shorewall/zones...
Nov 20 15:41:01 vps690699 shorewall[2810]: Compiling /etc/shorewall/interfaces...
Nov 20 15:41:01 vps690699 shorewall[2810]: Determining Hosts in Zones...
Nov 20 15:41:01 vps690699 shorewall[2810]: Locating Action Files...
Nov 20 15:41:01 vps690699 shorewall[2810]: Compiling /etc/shorewall/policy...
Nov 20 15:41:01 vps690699 shorewall[2810]: Adding Anti-smurf Rules
Nov 20 15:41:01 vps690699 shorewall[2810]: Adding rules for DHCP
Nov 20 15:41:01 vps690699 shorewall[2810]: Compiling TCP Flags filtering...
Nov 20 15:41:01 vps690699 shorewall[2810]: Compiling Kernel Route Filtering...
Nov 20 15:41:01 vps690699 shorewall[2810]: Compiling Martian Logging...
Nov 20 15:41:01 vps690699 shorewall[2810]: Compiling Accept Source Routing...
Nov 20 15:41:01 vps690699 shorewall[2810]: Compiling /etc/shorewall/tcinterfaces...
Nov 20 15:41:01 vps690699 shorewall[2810]: Compiling /etc/shorewall/snat...
Nov 20 15:41:01 vps690699 shorewall[2810]: Compiling MAC Filtration -- Phase 1...
Nov 20 15:41:01 vps690699 shorewall[2810]: Compiling /etc/shorewall/rules...
Nov 20 15:41:02 vps690699 shorewall[2810]: Compiling /etc/shorewall/conntrack...
Nov 20 15:41:02 vps690699 shorewall[2810]: Compiling MAC Filtration -- Phase 2...
Nov 20 15:41:02 vps690699 shorewall[2810]: Applying Policies...
Nov 20 15:41:02 vps690699 shorewall[2810]: Generating Rule Matrix...
Nov 20 15:41:02 vps690699 shorewall[2810]: Creating iptables-restore input...
Nov 20 15:41:02 vps690699 shorewall[2810]: Compiling /etc/shorewall/stoppedrules...
Nov 20 15:41:02 vps690699 shorewall[2810]: Shorewall configuration compiled to /var/lib/shorewall/.reload
Nov 20 15:41:02 vps690699 shorewall[2810]: Reloading Shorewall....
Nov 20 15:41:02 vps690699 shorewall[2810]: Initializing...
Nov 20 15:41:02 vps690699 shorewall[2810]: Setting up Route Filtering...
Nov 20 15:41:02 vps690699 shorewall[2810]: Setting up Martian Logging...
Nov 20 15:41:02 vps690699 shorewall[2810]: Setting up Accept Source Routing...
Nov 20 15:41:02 vps690699 shorewall[2810]: Setting up Traffic Control...
Nov 20 15:41:02 vps690699 shorewall[2810]: Preparing iptables-restore input...
Nov 20 15:41:02 vps690699 shorewall[2810]: Running /sbin/iptables-restore --wait 60...
Nov 20 15:41:03 vps690699 shorewall[2810]: IPv4 Forwarding Enabled
Nov 20 15:41:03 vps690699 root: Shorewall reloaded
Nov 20 15:41:03 vps690699 shorewall[2810]: done.
Nov 20 15:41:03 vps690699 systemd[1]: Reloaded Shorewall IPv4 firewall.
Nov 20 15:41:03 vps690699 systemd[1]: Reloading Shorewall IPv4 firewall.
Nov 20 15:41:03 vps690699 shorewall[3101]: Compiling using Shorewall 5.2.3.2...
Nov 20 15:41:03 vps690699 shorewall[3101]: perl: warning: Setting locale failed.
Nov 20 15:41:03 vps690699 shorewall[3101]: perl: warning: Please check that your locale settings:
Nov 20 15:41:03 vps690699 shorewall[3101]: #011LANGUAGE = (unset),
Nov 20 15:41:03 vps690699 shorewall[3101]: #011LC_ALL = (unset),
Nov 20 15:41:03 vps690699 shorewall[3101]: #011LANG = "fr_FR.UTF-8"
Nov 20 15:41:03 vps690699 shorewall[3101]:     are supported and installed on your system.
[... 200 lines cut ...]
Nov 20 15:41:11 vps690699 shorewall[4267]: Compiling /etc/shorewall/tcinterfaces...
Nov 20 15:41:11 vps690699 shorewall[4267]: Compiling /etc/shorewall/snat...
[... cut ...]
Nov 20 15:41:12 vps690699 shorewall[4267]: Running /sbin/iptables-restore --wait 60...
Nov 20 15:41:12 vps690699 shorewall[4267]: IPv4 Forwarding Enabled
Nov 20 15:41:12 vps690699 root: Shorewall reloaded
Nov 20 15:41:12 vps690699 shorewall[4267]: done.
Nov 20 15:41:12 vps690699 systemd[1]: Reloaded Shorewall IPv4 firewall.
Nov 20 15:41:14 vps690699 v2ray[408]: 2020/11/20 15:41:14 127.0.0.1:42313 accepted tcp:127.0.0.1:0 [api]
Nov 20 15:41:14 vps690699 v2ray[408]: 2020/11/20 15:41:14 127.0.0.1:42315 accepted tcp:127.0.0.1:0 [api]

matthieu71000, 2020-11-20:
var/log/messages

Nov 20 15:37:37 vps6906 root: Shorewall reloaded
Nov 20 15:37:39 vps6906 root: Shorewall reloaded
Nov 20 15:37:41 vps6906 root: Shorewall reloaded
Nov 20 15:37:44 vps6906 OMR-Service: No answer from VPN client end, restart Glorytun-TCP
Nov 20 15:38:29 vps6906 OMR-Service: No answer from VPN client end, restart Glorytun-TCP

matthieu71000, 2020-11-20:
J'ai relancé la mise à du VPS sans succès.

I renew vps update same problem

Ysurac, 2020-11-20:
Many shorewall reload, I would need full configuration, without server/keys, available in System->OpenMPTCProuter, "Show all settings" tab.

matthieu71000, 2020-11-20:
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/use-application-dns.net/'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.cot/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1' '/www.easyjet.com/omr_dst_bypass_wan4,omr6_dst_bypass_wan4' '/mafreebox.free.fr/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/ldlc-pro.com/omr_dst_bypass_wan4,omr6_dst_bypass_wan4' '/indemnisation.allianz-assistance.fr/omr_dst_bypass_wan4,omr6_dst_bypass_wan4' '/brusselsairlines.com/omr_dst_bypass_wan4,omr6_dst_bypass_wan4' '/arlo.netgear.com/omr_dst_bypass_wan4,omr6_dst_bypass_wan4' '/canal.fr/omr_dst_bypass_wan4,omr6_dst_bypass_wan4' '/canalplus.com/omr_dst_bypass_wan4,omr6_dst_bypass_wan4' '/canal-plus.com/omr_dst_bypass_wan4,omr6_dst_bypass_wan4' '/compte-canal.canal-plus.com/omr_dst_bypass_wan4,omr6_dst_bypass_wan4' '/aircanada.com/omr_dst_bypass_wan4,omr6_dst_bypass_wan4' '/particulier.edf.fr/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/www.aeroplan.com/omr_dst_bypass_wan4,omr6_dst_bypass_wan4' '/aeroplan.com/omr_dst_bypass_wan4,omr6_dst_bypass_wan4' '/dailymotion.com/omr_dst_bypass_wan4,omr6_dst_bypass_wan4' '/intel.fr/omr_dst_bypass_wan4,omr6_dst_bypass_wan4'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='10'
dhcp.lan.limit='60'
dhcp.lan.leasetime='12h'
dhcp.lan.force='1'
dhcp.lan.ra_default='1'
dhcp.lan.dhcpv6='server'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dhcp.@host[0]=host
dhcp.@host[0].dns='1'
dhcp.@host[0].mac='B0:61:C7:08:53:A4'
dhcp.@host[0].ip='192.168.1.172'
dhcp.@host[0].name='VOIP_DL'
dhcp.@host[1]=host
dhcp.@host[1].name='VOIP_FC'
dhcp.@host[1].dns='1'
dhcp.@host[1].mac='B0:61:C7:08:53:A3'
dhcp.@host[1].ip='192.168.1.173'
dhcp.@host[2]=host
dhcp.@host[2].name='VOIP_SM'
dhcp.@host[2].dns='1'
dhcp.@host[2].mac='B0:61:C7:07:73:88'
dhcp.@host[2].ip='192.168.1.171'
dhcp.@host[3]=host
dhcp.@host[3].name='VOIP_MB'
dhcp.@host[3].dns='1'
dhcp.@host[3].mac='B0:61:C7:08:53:3B'
dhcp.@host[3].ip='192.168.1.170'
dhcp.@host[4]=host
dhcp.@host[4].name='VOIP_CB'
dhcp.@host[4].dns='1'
dhcp.@host[4].mac='B0:61:C7:08:53:3D'
dhcp.@host[4].ip='192.168.1.174'
dhcp.@host[5]=host
dhcp.@host[5].dns='1'
dhcp.@host[5].ip='192.168.1.26'
dhcp.@host[5].name='CB.pc'
dhcp.@host[5].mac='1C:69:7A:67:83:25'
dhcp.@host[6]=host
dhcp.@host[6].dns='1'
dhcp.@host[6].mac='3c:07:54:10:94:95'
dhcp.@host[6].ip='192.168.1.80'
dhcp.@host[6].name='MP.mini'
dhcp.@host[7]=host
dhcp.@host[7].dns='1'
[... 1911 lines cut ...]
v2ray.main_routing.rules='omrrouting' 'omrroutingo'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1' 'wan2' 'wan3' 'wan4'
vnstat.@vnstat[0].backup='0'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='36'
wireless.radio0.hwmode='11a'
wireless.radio0.path='platform/soc/3f300000.mmc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.htmode='VHT80'
wireless.radio0.disabled='1'
wireless.radio0.country='00'
wireless.radio1=wifi-device
wireless.radio1.type='mac80211'
wireless.radio1.channel='36'
wireless.radio1.hwmode='11a'
wireless.radio1.path='platform/soc/3f300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio1.htmode='VHT80'
wireless.radio1.disabled='1'
wireless.default_radio1=wifi-iface
wireless.default_radio1.device='radio1'
wireless.default_radio1.network='lan'
wireless.default_radio1.mode='ap'
wireless.default_radio1.ssid='OpenWrt'
wireless.default_radio1.encryption='none'

Ysurac, 2020-11-20:
I don't find issue. I compile a new snapshot image and maybe this will solve your issue...

matthieu71000, 2020-11-20:
for router ?

Ysurac, 2020-11-20:
For both, but router image should be enough.
RPI3 image is compiled: https://github.com/Ysurac/openmptcprouter/wiki/Snapshots

matthieu71000, 2020-11-20:
Thanks I test

matthieu71000, 2020-11-20:
Routeur Crash after restore Backup

Ysurac, 2020-11-20:
This is strange... I will test the image on my RPI3

matthieu71000, 2020-11-20:
I restore a 55.3 with an old backup

matthieu71000, 2020-11-20:
I already restart VPS update

matthieu71000, 2020-11-20:
No outputNo outputNo server IP address, No WAN IP addressCan't get public IP address from ShadowSocksCan't contact Server Admin Script (No result - No API script answer)Version 0.1022 5.4.74-mptcp(Latest available version 0.1017)
--

tboome33, 2020-11-20:
same problems here with the same version :-(

Great regards

Ysurac, 2020-11-20:
Also an update ?

matthieu71000, 2020-11-20:
> Also an update ?

I restart vps script, reinstall 55.3 router and restore old backup. 
Already VPN TUNNEL DOWN

In log it repeat 
Fri Nov 20 20:40:43 2020 daemon.notice procd: /etc/rc.d/S99openmptcprouter-vps: {"result":"done","reason":"changes applied","route":"shorewallclose"}

Ysurac, 2020-11-20:
What do you have in ```/etc/shorewall/rules``` on the VPS ?

matthieu71000, 2020-11-20:
Really a lot of lines like this (I replace port and ip)

 For information about entries in this file, type "man shorewall-rules"
######################################################################################################################################################################################################
#ACTION         SOURCE          DEST            PROTO   DEST    SOURCE          ORIGINAL        RATE            USER/   MARK    CONNLIMIT       TIME            HEADERS         SWITCH          HELPER
#                                                       PORT    PORT(S)         DEST            LIMIT           GROUP
?SECTION ALL
?SECTION ESTABLISHED
?SECTION RELATED
?SECTION INVALID
?SECTION UNTRACKED
?SECTION NEW

#       Don't allow connection pickup from the net
#
Invalid(DROP)   net             all             tcp
#
#       Accept DNS connections from the firewall to the network
#
DNS(ACCEPT)     $FW             net
#
#       Allow Ping from/to the VPN
#
Ping(ACCEPT)    vpn             $FW
Ping(ACCEPT)    $FW             vpn
#
#       Allow Ping from the firewall to the network
#
Ping(ACCEPT)    $FW             net
#
# Drop Ping from the "bad" net zone.. and prevent your log from being flooded..
#
#Ping(DROP)     net             $FW
Ping(ACCEPT)    net             $FW
#
#	Accept connection from port > 65000 for shadowsocks and glorytun on the firewall
#
ACCEPT          net             $FW             tcp     65000-65535
ACCEPT          net             $FW             udp     65000-65535
#
#	Accept connection from SSH to the firewall
#
ACCEPT          net             $FW             tcp     65222
#
#	DHCP forward to the VPN from the firewall
#
DHCPfwd(ACCEPT)	$FW		vpn
#
#	Redirect all port from 1 to 64999 to the VPN client from the network
#
DNAT		net		vpn:$OMR_ADDR	tcp	1-64999
DNAT		net		vpn:$OMR_ADDR	udp	1-64999
DNAT		net:54.3x.xx.xx		vpn:$OMR_ADDR	tcp	8080	# OMR openmptcprouter redirect router 8080 port tcp from 54.3x.xx.xx
DNAT		net:54.3x.xx.xx		vpn:$OMR_ADDR	udp	8080	# OMR openmptcprouter redirect router 8080 port udp from 54.3x.xx.xx
DNAT		net:54.3x.xx.xx		vpn:$OMR_ADDR	tcp	8080	# OMR openmptcprouter redirect router 8080 port tcp from 54.3x.xx.xx
DNAT		net:54.3x.xx.xx		vpn:$OMR_ADDR	udp	8080	# OMR openmptcprouter redirect router 8080 port udp from 54.3x.xx.xx
DNAT		net:54.3x.xx.xx		vpn:$OMR_ADDR	tcp	8080	# OMR openmptcprouter redirect router 8080 port tcp from 54.3x.xx.xx
DNAT		net:54.3x.xx.xx		vpn:$OMR_ADDR	udp	8080	# OMR openmptcprouter redirect router 8080 port udp from 54.3x.xx.xx
DNAT		net:54.3x.xx.xx		vpn:$OMR_ADDR	tcp	8080	# OMR openmptcprouter redirect router 8080 port tcp from 54.3x.xx.xx
DNAT		net:54.3x.xx.xx		vpn:$OMR_ADDR	udp	8080	# OMR openmptcprouter redirect router 8080 port udp from 54.3x.xx.xx
DNAT		net:54.3x.xx.xx		vpn:$OMR_ADDR	tcp	8080	# OMR openmptcprouter redirect router 8080 port tcp from 54.3x.xx.xx
DNAT		net:54.3x.xx.xx		vpn:$OMR_ADDR	udp	8080	# OMR openmptcprouter redirect router 8080 port udp from 54.3x.xx.xx
DNAT		net:54.3x.xx.xx		vpn:$OMR_ADDR	tcp	8080	# OMR openmptcprouter redirect router 8080 port tcp from 54.3x.xx.xx
DNAT		net:54.3x.xx.xx		vpn:$OMR_ADDR	tcp	8080	# OMR openmptcprouter redirect router 8080 port tcp from 54.3x.xx.xx

matthieu71000, 2020-11-20:
> What do you have in `/etc/shorewall/rules` on the VPS ?

is it possible to purge this file ?

Ysurac, 2020-11-20:
yes, delete duplicated lines

matthieu71000, 2020-11-20:
Perfect it works, MERCI BCP
My routeur is now with  0.55.3 version
I test the update ?

Ysurac, 2020-11-21:
You can try.
I found the problem with firewall rules and I will release a new VPS script soon.

matthieu71000, 2020-11-21:
This night I reinstall last routeur version, restore my last backup. Remove all my firewall table on VPS and routeur and after reconfigure it. It works

matthieu71000, 2020-11-21:
Encore merci pour votre aide. Bon week-end. 
Matthieu

matthieu71000, 2021-02-22:
Bonjour, le script a t il été mis à jour ? Car j'ai régulièrement des problèmes avec ce fichier /etc/shorewall/rules

Les lignes se retrouvent dupliquées. et ensuite message "VPN en répond pas"

Merci

closed 2021-05-28 by github-actions[bot]