Ysurac/openmptcprouter #1402: Setup WIFI AP on RPI4

pbapba512, 2020-11-20:
I'm tried to setup a WIFI AP on my RPI4 according to the documentation. Probably its my fault, but its not works for me. I followed the documentation step-by-step. As I see well after the setup, my WIFI radio is running in Client mode, of course I setted up as Access point:

[image]
[image]
[image]

The radio hardware is working anyway, its probably configuration error, but I can't figure out what is the problem. So my question is: how can I switch the radio to AP mode, instead of client mode?

thank you for your help!

  - OpenMPTCProuter version: v0.56.5
  - OpenMPTCProuter VPS version: 5.4.65-mptcp 0.1020
  - OpenMPTCProuter platform: RPI4, latest firmware (updated today)

Adorfer, 2020-11-21:
wifi5 seems highly buggy, specially if you try anything above HT20. 
No idea what is broken, but is likely an OpenwRT issue, which is better adressed in the openwrt forums, since there are already a few threads about that.

constant1010, 2020-11-21:
it works with this config of `/etc/config/wireless`

`config wifi-device 'radio0'
    option type 'mac80211'
    option path 'platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
    option legacy_rates '0'
    option country 'US'
    option hwmode '11g'
    option channel '7'
    option htmode 'HT20'
    option short_gi_40 '0'`

pbapba512, 2020-11-21:
Thank you! Menawhile I found the solution, yes, its buggy. With legacy 2.4G settings works well. But all of other modes (N, 5G, etc) are broken.

Again, Adorfer and Constant1010, thank you for the help!

pbapba512, 2020-11-22:
Today we did a 2 hours long SRT livestream via WIFI interface, and it was stable. So with legacy 2.4G settings works well, and all the others are broken, buggy. I close this issue, becuase its partially solved, but there is some bug with the RPI WIFI handling.

closed 2020-11-22 by pbapba512