Ysurac/openmptcprouter #1403: Packages missing

Maverick78de, 2020-11-22:
I tried to install packages to use bonding (LAN Trunking) but the kmod-bonding package from openwrt is missing in your repo.
Please provide this package too
The luci package for bonding is available.

` opkg install kmod-bonding
Unknown package 'kmod-bonding'.
Collected errors:
 * opkg_install_cmd: Cannot install package kmod-bonding.
root@MPTCP-GW:~# opkg list | grep bond
luci-proto-bonding - git-20.282.09254-d4901f3 - Support for Link Aggregation (Channel Bonding)
proto-bonding - 2020-03-30-1 - This package contains the channel bonding proto handler for netifd
yggdrasil - 0.3.15-1 - Yggdrasil builds end-to-end encrypted networks with IPv6. Beyond the similarities with cjdns is a different routing algorithm. This globally-agreed spanning tree uses greedy routing in a metric space. Back-pressure routing techniques allow advanced link aggregation bonding on per-stream basis. In turn, a single stream will span across multiple network interfaces simultaneously with much greater throughput.
root@MPTCP-GW:~# opkg install luci-proto-bonding
Installing luci-proto-bonding (git-20.282.09254-d4901f3) to root...
Downloading http://packages.openmptcprouter.com/v0.56.4/aarch64_cortex-a53/luci/luci-proto-bonding_git-20.282.09254-d4901f3_all.ipk
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-proto-bonding:
 *      kmod-bonding
 * opkg_install_cmd: Cannot install package luci-proto-bonding.
`

Ysurac, 2020-11-28:
```
opkg install kmod-bonding
Installing kmod-bonding (5.4.69-1) to root...
Downloading http://download.openmptcprouter.com:80/release/v0.56.5/rpi4/targets/bcm27xx/bcm2711/packages/kmod-bonding_5.4.69-1_aarch64_cortex-a72.ipk
Configuring kmod-bonding.
```
You may need to refresh package list, but the package exist.

Maverick78de, 2020-11-28:
Thx, I found the problem. I used sysupgrade image to update to 0.56.5 from 0.56.4 and the /etc/opkg/distfeeds.conf wasn't updated to the new version path.

closed 2021-03-03 by github-actions[bot]