Ysurac/openmptcprouter #1404: Ping instable Discord ou perte de co en jeu

Anoratsu, 2020-11-22:
Bonjour,

Expected Behavior
Pouvoir jouer et communiquer correctement

Current Behavior
Actuellement mon ping est instable, normalement vers 50/60ms il monte parfois à 4 000ms en faisant des pics. Cela à pour conséquence une mauvaise communication vocale. Cela se produit quand les réglages sont les suivants : shadowsocks/chacha20/glorytun tcp

Si j'utilise v2ray, le ping sur discord est OK donc plus de problèmes de voix robotiques par contre impossible de me connecter à mes jeux (escape from tarkov, golf it, par exemple).

Possible Solution
Aucune idée

Steps to Reproduce the Problem
Réglages : shadowsocks/chacha20/glorytun tcp => ping instable sur discord
Réglage : v2ray => ping OK mais pas moyen de se connecter aux jeux (escape from tarkov, golf it...)

Specifications
OpenMPTCProuter version: 0.56.5
OpenMPTCProuter VPS version: 0.1022
OpenMPTCProuter platform: Raspberry Pi 4 Model B Rev 1.1

Anoratsu, 2020-11-22:
Logs : 

Sun Nov 22 12:12:08 2020 daemon.info dnsmasq-dhcp[26312]: read /etc/ethers - 0 addresses
Sun Nov 22 12:12:08 2020 user.notice omr-bypass: OMR-ByPass is running
Sun Nov 22 12:12:08 2020 user.notice omr-bypass: Reload dnsmasq...
Sun Nov 22 12:12:09 2020 daemon.info dnsmasq[26312]: read /etc/hosts - 4 addresses
Sun Nov 22 12:12:09 2020 daemon.info dnsmasq[26312]: read /tmp/hosts/odhcpd - 0 addresses
Sun Nov 22 12:12:09 2020 daemon.info dnsmasq[26312]: read /tmp/hosts/dhcp.cfg01411c - 5 addresses
Sun Nov 22 12:12:09 2020 daemon.info dnsmasq-dhcp[26312]: read /etc/ethers - 0 addresses
Sun Nov 22 12:12:09 2020 user.notice omr-bypass: OMR-ByPass is running
Sun Nov 22 12:12:09 2020 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Sun Nov 22 12:12:27 2020 daemon.info dnsmasq-dhcp[26312]: DHCPREQUEST(eth0) 192.168.100.101 74:da:88:e1:61:4b
Sun Nov 22 12:12:27 2020 daemon.info dnsmasq-dhcp[26312]: DHCPACK(eth0) 192.168.100.101 74:da:88:e1:61:4b Archer_C6
Sun Nov 22 12:12:36 2020 user.notice omr-tracker: Launching...
Sun Nov 22 12:12:36 2020 user.notice omr-tracker: Launched
Sun Nov 22 12:12:37 2020 daemon.info glorytun: starting glorytun vpn instance vpn
Sun Nov 22 12:12:37 2020 user.notice omr-bypass: Starting OMR-ByPass...
Sun Nov 22 12:12:38 2020 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Sun Nov 22 12:12:38 2020 user.notice omr-bypass: Restart dnsmasq...
Sun Nov 22 12:12:38 2020 daemon.info dnsmasq[26312]: exiting on receipt of SIGTERM
Sun Nov 22 12:12:38 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Nov 22 12:12:38 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Nov 22 12:12:38 2020 user.notice dnsmasq: Allowing RFC1918 responses for domain plex.direct
Sun Nov 22 12:12:38 2020 user.notice omr-bypass: OMR-ByPass is running
Sun Nov 22 12:12:38 2020 daemon.info dnsmasq[5837]: started, version 2.82 cachesize 150
Sun Nov 22 12:12:38 2020 daemon.info dnsmasq[5837]: DNS service limited to local subnets
Sun Nov 22 12:12:38 2020 daemon.info dnsmasq[5837]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Sun Nov 22 12:12:38 2020 daemon.info dnsmasq-dhcp[5837]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Sun Nov 22 12:12:38 2020 daemon.info dnsmasq[5837]: using only locally-known addresses for domain test
Sun Nov 22 12:12:38 2020 daemon.info dnsmasq[5837]: using only locally-known addresses for domain onion
Sun Nov 22 12:12:38 2020 daemon.info dnsmasq[5837]: using only locally-known addresses for domain localhost
Sun Nov 22 12:12:38 2020 daemon.info dnsmasq[5837]: using only locally-known addresses for domain local
Sun Nov 22 12:12:38 2020 daemon.info dnsmasq[5837]: using only locally-known addresses for domain invalid
Sun Nov 22 12:12:38 2020 daemon.info dnsmasq[5837]: using only locally-known addresses for domain bind
Sun Nov 22 12:12:38 2020 daemon.info dnsmasq[5837]: using only locally-known addresses for domain use-application-dns.net
Sun Nov 22 12:12:38 2020 daemon.info dnsmasq[5837]: using nameserver 127.0.0.1#5353
Sun Nov 22 12:12:38 2020 daemon.info dnsmasq[5837]: using only locally-known addresses for domain lan
Sun Nov 22 12:12:38 2020 daemon.info dnsmasq[5837]: read /etc/hosts - 4 addresses
Sun Nov 22 12:12:38 2020 daemon.info dnsmasq[5837]: read /tmp/hosts/odhcpd - 0 addresses
Sun Nov 22 12:12:38 2020 daemon.info dnsmasq[5837]: read /tmp/hosts/dhcp.cfg01411c - 5 addresses
Sun Nov 22 12:12:38 2020 daemon.info dnsmasq-dhcp[5837]: read /etc/ethers - 0 addresses
Sun Nov 22 12:12:59 2020 daemon.info dnsmasq-dhcp[5837]: DHCPREQUEST(eth0) 192.168.100.101 74:da:88:e1:61:4b
Sun Nov 22 12:12:59 2020 daemon.info dnsmasq-dhcp[5837]: DHCPACK(eth0) 192.168.100.101 74:da:88:e1:61:4b Archer_C6
Sun Nov 22 12:13:31 2020 daemon.info dnsmasq-dhcp[5837]: DHCPREQUEST(eth0) 192.168.100.101 74:da:88:e1:61:4b
Sun Nov 22 12:13:31 2020 daemon.info dnsmasq-dhcp[5837]: DHCPACK(eth0) 192.168.100.101 74:da:88:e1:61:4b Archer_C6
Sun Nov 22 12:14:04 2020 daemon.info dnsmasq-dhcp[5837]: DHCPREQUEST(eth0) 192.168.100.101 74:da:88:e1:61:4b
Sun Nov 22 12:14:04 2020 daemon.info dnsmasq-dhcp[5837]: DHCPACK(eth0) 192.168.100.101 74:da:88:e1:61:4b Archer_C6
Sun Nov 22 12:14:36 2020 daemon.info dnsmasq-dhcp[5837]: DHCPREQUEST(eth0) 192.168.100.101 74:da:88:e1:61:4b
Sun Nov 22 12:14:36 2020 daemon.info dnsmasq-dhcp[5837]: DHCPACK(eth0) 192.168.100.101 74:da:88:e1:61:4b Archer_C6
Sun Nov 22 12:15:08 2020 daemon.info dnsmasq-dhcp[5837]: DHCPREQUEST(eth0) 192.168.100.101 74:da:88:e1:61:4b
Sun Nov 22 12:15:08 2020 daemon.info dnsmasq-dhcp[5837]: DHCPACK(eth0) 192.168.100.101 74:da:88:e1:61:4b Archer_C6
Sun Nov 22 12:15:40 2020 daemon.info dnsmasq-dhcp[5837]: DHCPREQUEST(eth0) 192.168.100.101 74:da:88:e1:61:4b
Sun Nov 22 12:15:40 2020 daemon.info dnsmasq-dhcp[5837]: DHCPACK(eth0) 192.168.100.101 74:da:88:e1:61:4b Archer_C6
Sun Nov 22 12:16:12 2020 daemon.info dnsmasq-dhcp[5837]: DHCPREQUEST(eth0) 192.168.100.101 74:da:88:e1:61:4b
Sun Nov 22 12:16:12 2020 daemon.info dnsmasq-dhcp[5837]: DHCPACK(eth0) 192.168.100.101 74:da:88:e1:61:4b Archer_C6
Sun Nov 22 12:16:44 2020 daemon.info dnsmasq-dhcp[5837]: DHCPREQUEST(eth0) 192.168.100.101 74:da:88:e1:61:4b
Sun Nov 22 12:16:44 2020 daemon.info dnsmasq-dhcp[5837]: DHCPACK(eth0) 192.168.100.101 74:da:88:e1:61:4b Archer_C6
[... cut ...]
Sun Nov 22 13:13:00 2020 daemon.info dnsmasq-dhcp[7111]: DHCPACK(eth0) 192.168.100.101 74:da:88:e1:61:4b Archer_C6
Sun Nov 22 13:13:32 2020 daemon.info dnsmasq-dhcp[7111]: DHCPREQUEST(eth0) 192.168.100.101 74:da:88:e1:61:4b
Sun Nov 22 13:13:32 2020 daemon.info dnsmasq-dhcp[7111]: DHCPACK(eth0) 192.168.100.101 74:da:88:e1:61:4b Archer_C6
Sun Nov 22 13:14:04 2020 daemon.info dnsmasq-dhcp[7111]: DHCPREQUEST(eth0) 192.168.100.101 74:da:88:e1:61:4b
Sun Nov 22 13:14:04 2020 daemon.info dnsmasq-dhcp[7111]: DHCPACK(eth0) 192.168.100.101 74:da:88:e1:61:4b Archer_C6

Ysurac, 2020-11-22:
L'archer C6 a l'air d'avoir du mal à obtenir une adresse par DHCP.
Pour les jeux il faudrait que je teste, mais il n'y a que "Golf it!" pouvant fonctionner sur Steam avec Proton sous Linux. Je vais cependant attendre une solde car il est à 9€ ;)

Anoratsu, 2020-11-22:
Oui en effet, je comprends pas car il semble fonctionner correctement.

Je ne sais pas comment je pourrais faire des tests. Est ce que ce soucis de ping peut venir de ma connexion adsl ou 4g qui n'est pas stable ?

Ysurac, 2020-11-22:
Le ping passe par le VPN, tout comme l'UDP quand le proxy shadowsocks est utilisé.
Quand V2Ray est utilisé, il n'y a plus que le ping ICMP qui passe par le VPN, donc il doit être stable vu qu'il sert plus à grand chose.
Il est possible d'essayer un autre VPN, dans certains cas glorytun TCP n'est pas le plus stable.

Par contre j'aimerais bien savoir pourquoi les jeux ne passent pas quand v2ray est utilisé...

Anoratsu, 2020-11-22:
Si je désactive le proxy et le vpn, je bénéficie plus de l'agrégation ? Même question si je désactive soit l'un soit l'autre ?
Quel VPN serait le plus stable ?

Ysurac, 2020-11-22:
Si le proxy et le VPN désactivé, oui plus d'agrégation. Si juste le proxy désactivé l'agrégation se fait via le VPN (plus lent), si juste le VPN désactivé, alors uniquement TCP est activé dans le cas de shadowsocks, et uniquement TCP et UDP dans le cas de V2Ray.
Pour le VPN, tout dépend l'utilisation au final, un VPN peut être stable dans une configuration et utilisation et moins dans une autre.

Anoratsu, 2020-11-22:
Ok je vois. En plus pour le VPN je ne peux pas faire n'importe quoi, ma compagne est en télétravail et dans certains cas son accès distant ne fonctionne pas (j'avais fait une issue pour ça).

Je fais des essais un peu au hasard, je viens de mettre l'adresse mac de mon PC dans omrbypass sur l'adsl uniquement et depuis mon ami ne m'entend plus en robot sur Discord. Le vocal semble stable. La 4G pourrait être la source du problème ?

Ysurac, 2020-11-22:
Si beaucoup de paquets sont perdus oui c'est un soucis.
Dans réseau->MPTCP certains paramètres peuvent être modifiés pour améliorer un peu les choses parfois.

Anoratsu, 2020-11-22:
A mon avis ma 4G doit pas être d'une excellente qualité donc surement de la perte de paquet.
Ok je vois le menu mais ni connaissant rien je ne sais pas quoi changer ....

Ysurac, 2020-11-22:
Tout dépend de ce que tu as déjà ;)
Mais sinon en scheduler tu devrais avoir BLEST, et en TCP Congestion control cubic. Du moins c'est ce qui est maintenant mis par défaut sur la dernière version (donc hors mise à jour). Tu peux aussi monter un peu le "Tentatives Multipath TCP SYN".

Anoratsu, 2020-11-22:
[image]
[image]

J'ai ça comme réglages (par défaut logiquement)

Ysurac, 2020-11-22:
Met donc BLEST en planificateur et cubic en contrôle de gestion, ça devrait déjà améliorer un peu les choses.

Anoratsu, 2020-11-22:
Ok je fais ça. Est ce que je change la valeur de "Tentatives Multipath TCP SYN"?

Ysurac, 2020-11-22:
Éventuellement en montant un peu, selon le résultat des autres changements.

Anoratsu, 2020-11-22:
Cela semble beaucoup plus stable. Plus de voix robotique mais il y a quand même des pics dans le ping. Je vais faire un test sur les jeux.
Tentatives Multipath TCP SYN cela peut jouer sur ces pics ?

Anoratsu, 2020-11-22:
Ok j'ai rien dit, j'ai lancé Escape from tarkov ça s'est envolé (plus de 3000ms) donc la belle voix robotique. Je continue de voir si c'était juste un "cas isolé"

Ysurac, 2020-11-22:
SQM peut être utilisé pour éviter la saturation d'une connexion et l'envolé du Ping. Ca se configure dans Système->OpenMPTCProuter.

Anoratsu, 2020-11-22:
Je l'active sur les 2 ou que l'ADSL ? Vu que d'après le message il faut éviter sur la 4G

Ysurac, 2020-11-22:
Essaye déjà sur l'ADSL, si pas de changements sur les 2.

Anoratsu, 2020-11-22:
ok je fais ça ;)
SQM sur adsl mais je n'ai pas touché "Tentatives Multipath TCP SYN" car je ne sais pas quoi mettre comme valeur

Corrado56, 2020-11-22:
Hi Just my experience.
May be cubic is more stable but there ia a performance penalty over BBR look at the following results:
Red box CUBIC, Green box BBR. 
Few minutes between the tests, no other parameter changed
[image]

Ysurac, 2020-11-22:
True, cubic is slower but often more stable for bad connections. BBR is faster but really not stable if connections are bad.
I think I will add some profiles in OpenMPTCProuter, so user can choose what is best with stable connections and with unstable connections.

Anoratsu, 2020-11-22:
En effet d'après un speedtest j'ai moins de puissance mais pour le moment c'est plus stable que cela soit sur Discord ou en jeu.

Randdrick, 2020-11-23:
Bonjour,
Pour jouer tu n'as pas besoin d'avoir du débit, mais une connexion stable.
Pour savoir si ton problème de ping vient ou non de ta connexion internet, mais ton PC de jeux en byPass sur ta connexion ADSL. ( Tu dois obtenir un ping stable, vue que tu ne passeras plus par le VPN, ni même le proxy).
Si ton ping est instable, c'est ta box ( ou ta connexion ADSL qui pose problème). Redémarres ta box et vois avec ton opérateur si le problème persiste.
Concernant les connexions 4G (  via un téléphone) elles n'ont jamais été réputées pour être stables. Elles le sont un peu plus si tu possèdes une box 4G, bien positionnée avec une antenne. Une bonne connexion 4 G doit te permettre d'avoir un ping à peu près équivalent à 30 à 40 ms, avec une gigue la plus faible possible ( La gigue, c'est en gros la variation de latence). Tu peux la mesurer grâce au site Nperf .
Comme toi, j'ai eu a un moment donné un problème de latence ( dû à une perte de paquets). Je pense que la cause était lié à mon VPS, bien que j'en ai jamais eu la preuve. Afin de résoudre ce problème, j'ai :
- Vérifié que ma connexion la plus table (ADSL en principe) soit bien définie en maitre
- Que le plugin utilisé pour l'obfuscation soit simple-obfs / http et non V2ray ( V2ray avec l'obfuscation activée pose des problèmes de ressources, du moins sur le RPI3)
- Que la sélection de l'interface maître soit en mode "Equilibrage"

J'utilise pour ma part, une connexion ADSL ( qui est en maitre par défaut), une connexion 4G box et une 4G via un téléphone portable ( pour avoir un peu plus de débit). J'ai réglé le Multipath TCP Scheduler sur ECF pour avoir plus de stabilité. Mon débit, lui n'a pas bougé.

Concernant Discord et autre logiciels du même type (teams, Teamspeak, etc) je te conseil de faire un ByPass sur ton interface réseau maitre grâce à l'OMR-ByPASS. En effet, le plus important encore une fois, est d'avoir de la stabilité ( et non du débit) pour ce type d'utilisation.

Enfin, la stabilité de ton VPS est importante aussi.  Pour t'en assurer, mets ton PC en ByPAss et fais un ping -t son adresse IP. Si tu as une pertes de paquets ( ou un ping instable ) contact ton hébergeur et ouvres un ticket avec les résultats de ton ping. (Si tu es chez FirstHeberg ton VPS sera changé de Filer)

Anoratsu, 2020-11-23:
Bonjour,
Mon installation est la suivante :
- ADSL
- 4G via une box 4G
- VPS chez Milkywan (ovh en secours)

Avant d'effectuer les modifications précédentes j'ai testé de bypass le pc sur l'adsl et en effet c'était plus stable. Avec les modifs cela reste stable sauf quelques pics qui sont bien plus rare qu'avant.

ECF serait donc mieux que BLEST ?
Pour le mode "equilibrage" je ne sais pas où trouver ce paramètre. Idem pour l'offuscation.

Pour bypass discord tu as juste utilisé le truc pré configuré de omrbypass dans service ?

Mon VPS n'est normalement pas la cause vu que le problème était présent sur les 2 que j'utilise.

Sur nperf avec les réglages que j'ai actuellement j'obtiens ceci comme résultat :
[image]

Randdrick, 2020-11-23:
`Avant d'effectuer les modifications précédentes j'ai testé de bypass le pc sur l'adsl et en effet c'était plus stable. Avec les modifs cela reste stable sauf quelques pics qui sont bien plus rare qu'avant.`
Oui c'est normal. En principe les pics doivent être très rares sur ta connexion ADSL (Après toute va dépendre de ton fournisseur d'accès. Orange par exemple privilégie une latence stable par rapport au débit. SFR, c'est plus le débit.. ) . Ta latence et ta gigue sont bonnes ( d'après ce que je peux voir voir sur ta capture d'écran). 

 `ECF serait donc mieux que BLEST ?`
Pour avoir testé les deux, oui pour moi ECF est plus performant et stable. Après, je n'ai pas fais de mesures précises pouvant le démontrer formellement.

`Pour bypass discord tu as juste utilisé le truc pré configuré de omrbypass dans service ?`
Le BYPass dans le service. Depuis la version 0.56.3 je crois, il n'y a plus ce soucis d'OMR.

`Pour le mode "equilibrage" je ne sais pas où trouver ce paramètre. Idem pour l'offuscation.`
Ils sont tous les deux dans le menu _Système->OpenMPTCPRouter->Paramètres avancés_.
- Le plugin de brouillage se trouve dans les paramètres de ton **VPS Maitre**.
- Le mode équilibrage, dans la sélection de l'interface maître - Autres paramètres

Anoratsu, 2020-11-23:
Après plusieurs nperf j'ai BLEST légèrement plus performant que ECF.
J'ai mis Discord dans le bypass sur l'adsl
et passé le mode sur équilibrage. Le brouillage n'était pas activé, je le laisse désactivé
Je suis chez Free pour l'adsl et la 4G

Randdrick, 2020-11-23:
Laisses cette configuration et surveilles maintenant la qualité de ta connexion.
Free n'est pas réputé être le plus stable ( surtout si tu es en dehors d'une grande ville). Après, c'est comme tout. La qualité de ta connexion va dépendre de la distance à laquelle tu te trouves de ton DSLAM. Tu peux aussi avoir des problèmes sur ta ligne cuivre. Surveilles au niveau de ta box, si tu n'aurais pas des erreurs de synchronisation et/ou perte de paquets. 
Il en va de même pour ta connexion 4G. Si tu es dans une zone de réception difficile ( à savoir des obstacles entre ton domicile et l'antenne, des murs trop épais, et/ou à la limite du cône d'émission de l'antenne, si tu captes ta 4G sur cellule défaillante, etc.. tout cela peut nuire à la qualité de ta 4 G. Même le temps peut avoir une influence, selon la direction du vent… . Dès fois, il suffit juste de déplacer la box de quelques centimètres pour que tout change.

Voici les liens de quelques sites qui pourraient t'intéresser :
- https://www.zoneadsl.com/test-debit-internet/4g.html
- https://www.cartoradio.fr/index.html#/

Anoratsu, 2020-11-23:
Oui je vais surveiller ma connexion.
Je sais pour Free mais en étant chez eux ça me permet d'avoir la data illimitée sur la 4G. Après j'ai bien positionné l'antenne extérieure pour la 4G et j'ai une vue dégagée (sommet de coteaux) donc ça va de ce côté.

Merci en tout cas pour ton aide, en espérant que ma connexion suive :)

Anoratsu, 2020-11-27:
Bonsoir,
Après plusieurs jours de tests, ma connexion est plus stable qu'avant l'ouverture de ce ticket. J'ai cependant encore des problèmes sur Discord malgré que Discord soit dans omrbypass. Cela voudrait dire que le soucis de stabilité vient de la co servant de bypass (adsl) ?

Ysurac, 2020-11-28:
Oui. Sauf si le routeur est complètement saturé, à voir avec le load average.

Anoratsu, 2020-11-28:
Le load average que je trouve sur le panel openmtcp ?
Ca serait ça : Charge moyenne：0.54, 0.40, 0.38
Si le soucis vient de la co, il faudrait que je la redémarre plus souvent j'imagine ? Ou bien c'est inutile.

Ysurac, 2020-11-28:
Pas de soucis de load average la.
Donc oui ça doit venir de la connexion. Redémarrer le routeur change rien, mais la box de la connexion oui pour refaire une synchro. Par exemple mon ADSL était synchro à 210Kb/s en rédémarrant la box je passe à 2817Kb/s ;)

Anoratsu, 2020-11-28:
Oui évidement quand je parlais de redémarrer c'était bien redémarrer la box ADSL (et pourquoi pas la 4G dans la foulée, ça devrait pas lui faire de mal ?).
Si j'ai des soucis de synchro est ce que ça serait bien de programmer par exemple un reboot quotidien à l'aide de prises connectées ? Histoire que les box se resynchronisent chaque jour.

Ysurac, 2020-11-28:
Oui ou mieux si la box a une API permettant de vérifier sa synchro et de lancer un redémarrage.
Ma vieille Freebox permet de voir la synchro, mais je pense que c'est pas possible de redémarrer via API, par contre c'est sûrement possible pour mon routeur 4G Huawei. Donc bonne idée ;)

Anoratsu, 2020-11-28:
J'ai la freebox crystal toute simple puis bon je suis trop rouillé pour jouer avec des api ! Je vais tester avec des prises connectées je verrais ce que ça donne comme résultats sur ma co

kevin39, 2020-11-28:
FYI

> routeur 4G Huawei

J'ai un reboot auto sur mon B715s-23c, j'ai utilisé ce script : https://github.com/mkorz/b618reboot

Anoratsu, 2020-11-28:
> FYI
> 
> > routeur 4G Huawei
> 
> J'ai un reboot auto sur mon B715s-23c, j'ai utilisé ce script : https://github.com/mkorz/b618reboot

J'ai le même routeur mais je ne pense pas avoir l'environnement pour mettre en place ce script

Anoratsu, 2020-12-01:
Mise en place des prises connectées avec reboot quotidien de mes box ADSL et 4G. On va voir ce que cela donne.

closed 2021-03-07 by github-actions[bot]