Ysurac/openmptcprouter #1405: Problem to add 2nd connection

tof30, 2020-11-22:
## Expected Behavior

Hi
I wan't to connect my internet box, my smartphone and an 4G router to my openmptcprouteur (raspberry 3B)
## Current Behavior

It's ok to connect my box and my smartphone but if i connect the 4G router i lost box connection
Can you help me ?

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.56.5 r0+14653-a439f1bb47
  - OpenMPTCProuter platform: Raspberry Pi 3 Model B Rev 1.2

Network configuration
network.loopback=interface
network.loopback.ifname='lo'
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.loopback.multipath='off'
network.loopback.macaddr='00:00:00:00:00:00'
network.loopback.metric='7'
network.globals=globals
network.globals.ula_prefix='fd6e:7dc3:d47e::/48'
network.globals.multipath='enable'
network.globals.mptcp_path_manager='fullmesh'
network.globals.mptcp_scheduler='blest'
network.globals.congestion='cubic'
network.globals.mptcp_checksum='0'
network.globals.mptcp_debug='0'
network.globals.mptcp_syn_retries='2'
network.globals.mptcp_fullmesh_num_subflows='1'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.lan=interface
network.lan.ifname='eth0'
network.lan.proto='static'
network.lan.ipaddr='192.168.100.1'
network.lan.netmask='255.255.255.0'
network.lan.ip6assign='60'
network.lan.ipv6='0'
network.lan.delegate='0'
network.lan.multipath='off'
network.lan.ip4table='lan'
network.lan.macaddr='xx:xx:xx:xx:xx:xx'
network.lan.modalias='usb:v0424pEC00d0200dcFFdsc00dp01icFFisc00ipFFin00'
network.lan.product='424/ec00/200'
network.lan.metric='8'
network.lan.defaultroute='0'
network.lan.peerdns='0'
network.lan.label='Param lan'
network.lan.addlatency='0'
network.lan_rule=rule
network.lan_rule.lookup='lan'
network.lan_rule.priority='100'
network.wan1=interface
network.wan1.proto='static'
network.wan1.ip4table='wan'
network.wan1.multipath='master'
network.wan1.defaultroute='0'
network.wan1.type='macvlan'
network.wan1.ifname='wan1'
network.wan1.masterintf='eth0'
network.wan1.peerdns='0'
network.wan1.ipv6='0'
network.wan1.metric='9'
network.wan1.macaddr='xx:xx:xx:xx:xx:xx'
network.wan1.label='Freebox'
network.wan1.ipaddr='192.168.0.200'
network.wan1.netmask='255.255.255.0'
network.wan1.gateway='192.168.0.254'
network.wan3=interface
network.wan3.proto='static'
network.wan3.type='macvlan'
network.wan3.masterintf='eth0'
network.wan3.ip4table='wan'
network.wan3.multipath='on'
network.wan3.defaultroute='0'
network.wan3.ifname='wan3'
network.wan3.metric='13'
network.wan3.peerdns='0'
network.wan3.ipv6='0'
network.wan3.macaddr='xx:xx:xx:xx:xx:xx'
network.wan3.label='Routeur'
network.wan3.ipaddr='192.168.21.100'
network.wan3.netmask='255.255.255.0'
network.wan3.gateway='192.168.21.1'
network.wan2=interface
network.wan2.ip4table='wan'
network.wan2.multipath='on'
network.wan2.defaultroute='0'
network.wan2.peerdns='0'
network.wan2.ipv6='0'
network.wan2.metric='10'
network.wan2.macaddr='xx:xx:xx:xx:xx:xx'
network.wan2.masterintf='wlan0'
network.wan2.ifname='wlan0'
network.wan2.proto='dhcp'
network.wan2.label='4G'
network.wan2.modalias='sdio:c00v02D0dA9A6'
network.wan1_dev=device
network.wan1_dev.name='wan1'
network.wan1_dev.type='macvlan'
network.wan1_dev.ifname='eth0'
network.wan1_dev.macaddr='xx:xx:xx:xx:xx:xx'
network.Routeur=interface
network.Routeur.proto='static'
network.Routeur.type='macvlan'
network.Routeur.ifname='Routeur'
network.Routeur.multipath='off'
network.Routeur.metric='15'
network.wan3_dev=device
network.wan3_dev.name='wan3'
network.wan3_dev.type='macvlan'
network.wan3_dev.ifname='eth0'
network.wan3_dev.macaddr='xx:xx:xx:xx:xx:xx'

Ysurac, 2020-11-22:
I don't see any error, but you should check if there is no IP conflict.
What is the router ? USB or ethernet based ?

tof30, 2020-11-22:
It's an ethernet router . It s a TP link AC750

closed 2021-02-26 by github-actions[bot]