Ysurac/openmptcprouter #1406: compiling ipq4019 error.

logernLee, 2020-11-22:
Hello!

I had some problems compiling ipq4019, and I chose P&W R619AC 128M. There are problems in mkimage, as follows:

mkdir -p /home/openmptcprouter/myplatform/source/bin/targets/ipq40xx/generic /home/openmptcprouter/myplatform/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp
rm -rf /home//openmptcprouter/myplatform/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/json_info_files
/home/openmptcprouter/myplatform/source/staging_dir/host/bin/mksquashfs4 /home/openmptcprouter/myplatform/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx /home/openmptcprouter/myplatform/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/root.squashfs -nopad -noappend -root-owned -comp xz -Xpreset 9 -Xe -Xlc 0 -Xlp 2 -Xpb 2 -Xbcj arm -b 256k -p '/dev d 755 0 0' -p '/dev/console c 600 0 0 5 1' -processors 1
Parallel mksquashfs: Using 1 processor
Creating 4.0 filesystem on /home/openmptcprouter/myplatform/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/root.squashfs, block size 262144.
Pseudo file "/dev" exists in source filesystem "/home/openmptcprouter/myplatform/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/dev".
Ignoring, exclude it (-e/-ef) to override.
[=============================================================/] 8856/8856 100%

Exportable Squashfs 4.0 filesystem, xz compressed, data block size 262144
	compressed data, compressed metadata, compressed fragments,
	compressed xattrs, compressed ids
	duplicates are removed
Filesystem size 37868.38 Kbytes (36.98 Mbytes)
	25.47% of uncompressed filesystem size (148684.19 Kbytes)
Inode table size 58514 bytes (57.14 Kbytes)
	18.62% of uncompressed inode table size (314287 bytes)
Directory table size 81008 bytes (79.11 Kbytes)
	33.94% of uncompressed directory table size (238688 bytes)
Number of duplicate files found 1660
Number of inodes 9553
Number of files 8523
Number of fragments 223
Number of symbolic links  548
Number of device nodes 1
Number of fifo nodes 0
Number of socket nodes 0
Number of directories 481
Number of ids (unique uids + gids) 1
Number of uids 1
	root (0)
Number of gids 1
	root (0)
rm -f /home/openmptcprouter/myplatform/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openmptcprouter-v0.56.5-3e459a2-r0+14653-a439f1b-ipq40xx-generic-p2w_r619ac-128m-initramfs-fit-zImage.itb
cp /home/openmptcprouter/myplatform/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/zImage-initramfs /home/openmptcprouter/myplatform/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openmptcprouter-v0.56.5-3e459a2-r0+14653-a439f1b-ipq40xx-generic-p2w_r619ac-128m-initramfs-fit-zImage.itb
/home/openmptcprouter/myplatform/source/scripts/mkits.sh -D p2w_r619ac-128m -o /home/openmptcprouter/myplatform/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openmptcprouter-v0.56.5-3e459a2-r0+14653-a439f1b-ipq40xx-generic-p2w_r619ac-128m-initramfs-fit-zImage.itb.its -k /home/openmptcprouter/myplatform/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openmptcprouter-v0.56.5-3e459a2-r0+14653-a439f1b-ipq40xx-generic-p2w_r619ac-128m-initramfs-fit-zImage.itb -d /home/openmptcprouter/myplatform/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/linux-5.4.69/arch/arm/boot/dts/qcom-ipq4019-r619ac-128m.dtb -C none -a 0x80208000 -e 0x80208000  -c config@10 -A arm -v 5.4.69
PATH=/home/openmptcprouter/myplatform/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/linux-5.4.69/scripts/dtc:/home/openmptcprouter/myplatform/source/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-8.4.0_musl_eabi/bin:/home/openmptcprouter/myplatform/source/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-8.4.0_musl_eabi/bin:/home/openmptcprouter/myplatform/source/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-8.4.0_musl_eabi/bin:/home/openmptcprouter/myplatform/source/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-8.4.0_musl_eabi/bin:/home/openmptcprouter/myplatform/source/staging_dir/host/bin:/home/openmptcprouter/myplatform/source/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-8.4.0_musl_eabi/bin:/home/openmptcprouter/myplatform/source/staging_dir/host/bin:/home/openmptcprouter/myplatform/source/staging_dir/host/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin mkimage -f /home/openmptcprouter/myplatform/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openmptcprouter-v0.56.5-3e459a2-r0+14653-a439f1b-ipq40xx-generic-p2w_r619ac-128m-initramfs-fit-zImage.itb.its /home/openmptcprouter/myplatform/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openmptcprouter-v0.56.5-3e459a2-r0+14653-a439f1b-ipq40xx-generic-p2w_r619ac-128m-initramfs-fit-zImage.itb.new
sh: 1: Syntax error: Unterminated quoted string
mkimage: Can't open /home/openmptcprouter/myplatform/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openmptcprouter-v0.56.5-3e459a2-r0+14653-a439f1b-ipq40xx-generic-p2w_r619ac-128m-initramfs-fit-zImage.itb.new.tmp: No such file or directory
Makefile:734: recipe for target '/home/openmptcprouter/myplatform/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openmptcprouter-v0.56.5-3e459a2-r0+14653-a439f1b-ipq40xx-generic-p2w_r619ac-128m-initramfs-fit-zImage.itb' failed
make[5]: *** [/home/openmptcprouter/myplatform/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openmptcprouter-v0.56.5-3e459a2-r0+14653-a439f1b-ipq40xx-generic-p2w_r619ac-128m-initramfs-fit-zImage.itb] Error 255
make[5]: Leaving directory '/home/openmptcprouter/myplatform/source/target/linux/ipq40xx/image'
Makefile:23: recipe for target 'install' failed
make[4]: *** [install] Error 2
make[4]: Leaving directory '/home/openmptcprouter/myplatform/source/target/linux/ipq40xx'
Makefile:13: recipe for target 'install' failed
make[3]: *** [install] Error 2
make[3]: Leaving directory '/home/openmptcprouter/myplatform/source/target/linux'
time: target/linux/install#239.51#29.69#316.42
target/Makefile:23: recipe for target 'target/linux/install' failed
make[2]: *** [target/linux/install] Error 2
make[2]: Leaving directory '/home/openmptcprouter/myplatform/source'
target/Makefile:19: recipe for target '/home/openmptcprouter/myplatform/source/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.target_install' failed
make[1]: *** [/home/openmptcprouter/myplatform/source/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/openmptcprouter/myplatform/source'
/home/openmptcprouter/myplatform/source/include/toplevel.mk:234: recipe for target 'world' failed
make: *** [world] Error 2

thanks for your reply.

adricoues, 2020-11-23:
Hi, 

I'm encoutering the same issue with this result : 

`make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/***/***/***/OpenMPTCP/openmptcprouter/BCM47XX/source/include/toplevel.mk:236 : world] Erreur 1`

Thanks for your reply.

suyuan168, 2020-11-24:
仔细检查自己的编译环境并且进行修改

closed 2021-02-28 by github-actions[bot]