Ysurac/openmptcprouter #1409: How to use iperf3 in OMR VPS?

ccmks, 2020-11-23:
I am trying to test my WAN connection speed against the VPS for OMR, how can I run iperf3? I saw the iperf3 is listening on port 65400. I am running windows machine to  run iperf3, but when I try to run iperf3 -c vps-ip-address -p 65400, I got error.

Please advise.

Thank you

Ysurac, 2020-11-23:
The iperf server on the VPS part need a password and a key.
```omr-iperf``` is available on the router to simplify the usage.

ccmks, 2020-11-23:
How can I iperf on specific WAN interface?

ccmks, 2020-11-23:
The problem I have right now is I am trying to figure out what's bottleneck one of my WAN connection. My WAN5 is supposed to get 70 Mbps download, but when using OMR, only 5 Mbps being used. I would like to run iperf on WAN5 against the VPS to see if the connection between WAN5 to VPS is being throttle...

ccmks, 2020-11-23:
If I do omr-test-speed eth5, it will get aggregate connection result. Plus omr-test-speed only run speedtest towards public server, not to my VPS

Ysurac, 2020-11-23:
I forgot to bind to an interface for omr-test-speed, I will fix that for next release.
To test speed from VPS you can transfert a file with scp for example.

closed 2021-02-28 by github-actions[bot]