Ysurac/openmptcprouter #1411: VPN Tunnel Down

Johnr24, 2020-11-24:
## Expected Behavior

VPN Tunnel to work 

## Current Behavior

VPN tunnel is not working despite being enabled 
## Possible Solution

## Steps to Reproduce the Problem

  1. ?
  2.?
  3.?

## Context (Environment)

Here are all my settings 
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_protection='0'
dhcp.@dnsmasq[0].server='/lan/' '/use-application-dns.net/' '10.255.255.1#53' '10.255.254.1#53' '127.0.0.1#5353' '127.0.0.1#5353' '127.0.0.1#5353' '127.0.0.1#5353' '127.0.0.1#5353' '127.0.0.1#5353'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1' '/spotify.ac/omr_dst_bypass_eth1,omr6_dst_bypass_eth1,omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.ac/omr_dst_bypass_eth1,omr6_dst_bypass_eth1,omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.ae/omr_dst_bypass_eth1,omr6_dst_bypass_eth1,omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.ae/omr_dst_bypass_eth1,omr6_dst_bypass_eth1,omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.africa/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.africa/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.africa/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.ag/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.ag/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.ag/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.ai/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.ai/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.ai/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.am/omr_dst_bypass_eth1,omr6_dst_bypass_eth1,omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.am/omr_dst_bypass_eth1,omr6_dst_bypass_eth1,omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.app/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.app/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.app/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.ar/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.ar/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.ar/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.arab/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.arab/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.arab/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.as/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.as/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.as/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.asia/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.asia/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.asia/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.at/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.at/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.at/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.audio/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.audio/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.audio/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.author/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.author/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.author/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.aws/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.aws/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.aws/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.az/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.az/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.az/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.ba/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.ba/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.ba/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.band/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.band/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.band/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.bar/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.bar/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.bar/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.bayern/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.bayern/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.bayern/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.be/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.be/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.be/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.bet/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.bet/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.bet/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.bg/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.bg/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.bg/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.bi/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.bi/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.bi/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.bio/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.bio/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.bio/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.biz/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.biz/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.biz/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.black/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.black/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.black/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.blog/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.blog/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.blog/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.blue/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.blue/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.blue/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.bo/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.bo/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.bo/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.bs/omr_dst_bypass_eth1,omr6_dst_bypass_eth1,omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.bs/omr_dst_bypass_eth1,omr6_dst_bypass_eth1,omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.by/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.by/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.by/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.bz/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.bz/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.bz/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.ca/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.ca/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.ca/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.cam/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.cam/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.cam/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify.car/omr_dst_bypass_eth1,omr6_dst_bypass_eth1' '/spotify [... cut ...]
v2ray.omrrouting.domain='full:omr.lan'
v2ray.omrroutingo=routing_rule
v2ray.omrroutingo.type='field'
v2ray.omrroutingo.inbound_tag='omrbridge'
v2ray.omrroutingo.outbound_tag='out'
v2ray.main_routing=routing
v2ray.main_routing.enabled='1'
v2ray.main_routing.rules='omrrouting' 'omrroutingo'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='eth0' 'eth1' 'eth5' 'eth3' 'eth4' 'tun0'

## Specifications

  - OpenMPTCProuter version:  0.56.5
  - OpenMPTCProuter VPS version:  0.1022.5.4.74
  - OpenMPTCProuter platform:        x86-64

Ysurac, 2020-11-24:
Can you put here the content of file ```/etc/shorewall/rules``` on the VPS ?

Johnr24, 2020-11-24:
Cheers mate, Here they are.

######################################################################################################################################################################################################
#ACTION         SOURCE          DEST            PROTO   DEST    SOURCE
     ORIGINAL        RATE            USER/   MARK    CONNLIMIT       TIME
         HEADERS         SWITCH          HELPER
#                                                       PORT    PORT(S)
    DEST            LIMIT           GROUP
?SECTION ALL
?SECTION ESTABLISHED
?SECTION RELATED
?SECTION INVALID
?SECTION UNTRACKED
?SECTION NEW

#       Don't allow connection pickup from the net
#
Invalid(DROP)   net             all             tcp
#
#       Accept DNS connections from the firewall to the network
#
DNS(ACCEPT)     $FW             net
#
#       Allow Ping from/to the VPN
#
Ping(ACCEPT)    vpn             $FW
Ping(ACCEPT)    $FW             vpn
#
#       Allow Ping from the firewall to the network
#
Ping(ACCEPT)    $FW             net
#
# Drop Ping from the "bad" net zone.. and prevent your log from being
flooded..
#
#Ping(DROP)     net             $FW
Ping(ACCEPT)    net             $FW
#
#       Accept connection from port > 65000 for shadowsocks and glorytun on
the firewall
#
ACCEPT          net             $FW             tcp     65000-65535
ACCEPT          net             $FW             udp     65000-65535
#
#       Accept connection from SSH to the firewall
#
ACCEPT          net             $FW             tcp     65222
#
#       DHCP forward to the VPN from the firewall
#
DHCPfwd(ACCEPT) $FW             vpn
#
#       Redirect all port from 1 to 64999 to the VPN client from the network
#
DNAT            net             vpn:$OMR_ADDR   tcp     1-64999
DNAT            net             vpn:$OMR_ADDR   udp     1-64999
ACCEPT          net             $FW             tcp     65001   # OMR
openmptcprouter open glorytun port tcp
ACCEPT          net             $FW             udp     65001   # OMR
openmptcprouter open glorytun port udp
ACCEPT          net             $FW             tcp     65101   # OMR
openmptcprouter open shadowsocks port tcp
ACCEPT          net             $FW             udp     65101   # OMR
openmptcprouter open shadowsocks port udp
DNAT            net             vpn:$OMR_ADDR   udp     1194    # OMR
openmptcprouter redirect router 1194 port udp
DNAT            net             vpn:$OMR_ADDR   tcp     1935    # OMR
openmptcprouter redirect router 1935 port tcp
DNAT            net             vpn:$OMR_ADDR   udp     1935    # OMR
openmptcprouter redirect router 1935 port udp
DNAT            net             vpn:$OMR_ADDR   tcp     8123    # OMR
openmptcprouter redirect router 8123 port tcp
DNAT            net             vpn:$OMR_ADDR   udp     8123    # OMR
openmptcprouter redirect router 8123 port udp
DNAT            net             vpn:$OMR_ADDR   tcp     80      # OMR
openmptcprouter redirect router 80 port tcp

-- 
John Rogers
---
Independent Colourist
http://johnrogerscolour.co.uk/
07774 347417
---
General Terms and Conditions
<https://johnrogerscolour.co.uk/terms-and-conditions> - Booking Terms and
Conditions
<https://johnrogerscolour.co.uk/bookings-and-terms-and-conditions>
---

Ysurac, 2020-11-24:
You can try snapshot image, to check if it's fixed (or not): https://download.openmptcprouter.com/develop/5.4/x86_64/targets/x86/64/

Johnr24, 2020-11-24:
Unfortunatly that snapshot image is not working for me,

Ysurac, 2020-11-24:
What is the issue you ? It doesn't boot ?

Johnr24, 2020-11-24:
Essentially yes.

Ysurac, 2020-11-24:
I tested the image using virtualbox and no issue. What is the image used ?

Johnr24, 2020-11-24:
It boots but then when I copy my settings over it will no longer boot

Johnr24, 2020-11-25:
Closing the issue as VPN tunnel is now working

At the time of the crash I was trying to stream alot of webrtc, which OMR does not deal as well with I am wondering if my VPS provider host world saw this and dropped the ports temporarily on the router,

Johnr24, 2020-12-02:
Hey @Ysurac I'm reopening this as the issue has come back, is their anything I can check with the hosting provider to see if they are limiting something?

closed 2021-03-07 by github-actions[bot]