Ysurac/openmptcprouter #1412: VOIP and Openmptcprouter

matthieu71000, 2020-11-24:
Hello,

I have a question about VOIP.

I am using OVH VOIP in SIP with g729 codec.
If I only use the OVH-ADSL line without Openmptcprouter I have no quality problem for calls.

When I use Openmptcprouter for my telephony, it is not correct.

My configuration: RPI3 + with ORANGE-ADSL, FREE-ADSL, OVH-ADSL and FREE-4G, with an OVH VPS.

Each ADSL gives DL: 13Mbits / UP: 950Mbits
4G gives: 60Mbits / 20 Mbits

I am using Glorytun TCP with ChaCha20 encryption.

First question, is it possible not to use encryption, and can this improve performance?
Second question, do you have any ideas for advice on how to properly configure the QOS for VOIP-OVH?

Thank you

Matthieu

Adorfer, 2020-11-25:
ovh VPS without dedicated cpu cores suffer from heavy jitter

matthieu71000, 2020-11-25:
> ovh VPS without dedicated cpu cores suffer from heavy jitter

ok so if I disable encryption, is it better ?

Adorfer, 2020-11-25:
I would not expect that.
But if it helps, why not. Strange anyways!

matthieu71000, 2020-11-25:
> I would not expect that.
> But if it helps, why not. Strange anyways!

No it's just a question, I don't test.
Do you think it could be better whith no encription ?

thanks

ndaptardar, 2020-11-26:
I dont think there is a way to disable encryption on Glorytun ... but do check out and let us know

Adorfer, 2020-11-26:
> Do you think it could be better whith no encription ?

No, i do not think so. 
I stand by my first reply.

DeeJayMX, 2020-11-30:
Just use only your OVH ADSL for your voip with OMR Bypass

closed 2021-03-05 by github-actions[bot]