Ysurac/openmptcprouter #1413: How much extra data used?

ndaptardar, 2020-11-24:
I am comparing OMR to Speedify. In one of the blogs, it was mentioned that Speedify uses lot more than needed, and it was confirmed by their customer support as well. May be as much as 50%. Example: if I am downloading 1MB file, actual data used is 1.5MB. Main reason - dropped packets or wastage of packets etc.

Do we have some idea how much extra data used by OMR?
By VPN if we know e.g. by MLVPN, by Glorytun, By DSVPN, By OpenVPN?
Does it matter type of connection? Like 4G data uses 10% more v/s DSL uses 5% more?

Ysurac, 2020-11-24:
Some data is used by encryption, but not a lot.
More data can be used if many dropped packets because of TCP retransmission, but not more than any router/client.
Data is also used to check connection status, but it's only some ICMP ping by default.
VPN are used for ICMP and UDP if shadowsocks proxy is used.
So not really much data are used than a normal router.

ndaptardar, 2020-11-26:
OMR uses VPN or Shadowsocks proxy, right? Not both at a time, or it does use both? If both, then wont it encrypt twice?

Ysurac, 2020-11-26:
It use both at same time. As I said, shadowsocks proxy is used for TCP, and the VPN for UDP & ICMP. If traffic is TCP it only use the proxy, so it's encrypted only once.

ndaptardar, 2020-11-26:
> It use both at same time. As I said, shadowsocks proxy is used for TCP, and the VPN for UDP & ICMP. If traffic is TCP it only use the proxy, so it's encrypted only once.

Oh OK. My bad!

closed 2020-11-26 by ndaptardar