Ysurac/openmptcprouter #1414: Boradband is not performing as good as 4G?

ndaptardar, 2020-11-24:
I have tried combining 3 4G connections, and they are doing OK. Am setting them as DHCP in wizard. Am getting 60%-90% of combined speed. All of the 3 4G give anywhere between 2-3Mbps to 8-10Mbps.
But when I am combining 2 broadbands on wifi (1 cable that gives 30Mbps, and 1 DSL that gives 6Mbps), I barely get 2-3Mbps combined. Any reason for this? Again I am using DHCP setting. I get same result when I combine broadband and 4G.

Any thoughts?

Ysurac, 2020-11-24:
Using wifi is often a very bad idea.

ndaptardar, 2020-11-25:
You mean WiFi of broadbands instead of wired broadband , right? Or WiFi in general, including 4G?

Ysurac, 2020-11-25:
I was speaking about wifi of any WAN, but any connection that can lost many packets is a bad idea, so it's the same for a bad 4G.
Wifi often lost many packets, so it's why it's a bad idea in most case.

closed 2020-11-25 by ndaptardar