Ysurac/openmptcprouter #1416: Architecture and technology documentation

ndaptardar, 2020-11-24:
**Is your feature request related to a problem? Please describe.**
This is a request to have more documentation on OMR.

**Describe the solution you'd like**
It would be good to understand internal architecture of OMR and VPS.
Some questions:
1. What kinds of VPN are supported, when to use which
2. What all firewalls are supported
3. What all kinds of congestion control logics supported, and when to use which
4. Connection diagram of tunnels
5. Different command line utilities given in OMR (like speed test)

I know its a tall order, but thought to put it out there :)
Thanks in advance.

suyuan168, 2020-11-26:
no~

Adorfer, 2020-11-26:
it would help a lot. 
There are now so many options to configure the OMR concerning tunnels and/or encryption and/or additions VPN, that i lost track about the pro/cons of the different combinations and how they might interfere with each other.

closed 2021-03-01 by github-actions[bot]