Ysurac/openmptcprouter #1417: installation Orange Pi 3

Roman001006, 2020-11-24:
Hi, is there any support for Orange Pi 3 microcomputer? Will it be in the future? Thanks!
Orange Pi 3 website: www.orangepi.org

Ysurac, 2020-11-24:
There is no support for Orange PI 3, and no idea for the future.
I only support hardware that I own (so it will only be supported if I receive a Orange PI 3 or if I buy one ;) ).

It's maybe possible to compile it yourself: https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform

Roman001006, 2020-11-24:
> Нет поддержки Orange PI 3 и нет никаких представлений о будущем.
> Я поддерживаю только оборудование, которое у меня есть (так что оно будет поддерживаться только в том случае, если я получу Orange PI 3 или куплю его;)).
> 
> Возможно, можно скомпилировать его самостоятельно: https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform

If I install the x86 version on the micro sd card, It will work?

Ysurac, 2020-11-24:
No. A dedicated image is needed (and Orange PI is based on ARM).

Roman001006, 2020-11-24:
> Нет. Требуется специальный образ (и Orange PI основан на ARM).

thanks!

closed 2021-02-28 by github-actions[bot]