Ysurac/openmptcprouter #1419: Firewall restriction from IP Source

mycanaletto, 2020-11-25:
Hello,
I've been a bit of a flirt without success.
I would like to create a rule to accept incoming traffic only from certain IP ranges (Cloudflare).

Ideally, I would have liked to do Geo IP but I don't think that's possible.

Merci ;-)

kevin39, 2020-11-25:
Hi,
What's your problem with firewall rules ? What did you try ?

mycanaletto, 2020-11-25:
Where do I put the authorized source IPs?

[image]
Here I can't

Ysurac, 2020-11-25:
In Advanced settings tab, source IP address.

mycanaletto, 2020-11-25:
I must be awake badly again!!!

Thanks to you.

No Geo IP ideas?

mycanaletto, 2020-11-25:
Except that in advanced the combo only shows internal IPs and it's not possible to add them.... (I would have to add all the CloudFlare ranges)
[image]

kevin39, 2020-11-25:
I'm not sure to understand you correctly.
How do you want to filter if you do not want to enter IP ranges ?

Not tested but CIDR format seems to be allowed.

Ysurac, 2020-11-25:
You can enter manually an ip at the end.
I don't remember if subnet is supported...
You can also directly edit ```/etc/shorewall/rules``` on the VPS (without putting comment at end of line). It may also support geoip, but I need to check if I added support in kernel for that...

mycanaletto, 2020-11-25:
In the meantime it is not possible to use "Firewall - Custom Rules"?

Ysurac, 2020-11-25:
no...
Changes in the router are applied in shorewall on the VPS, so traffic is blocked on VPS side and is not forwarded to the router when not needed.
And it's easy to edit shorewall rules, so why doing something more difficult that will give a bad result ?

mycanaletto, 2020-11-25:
Yes it can be a solution.

But finally I send the port to a synology's reverse proxy which has a GeoIP blocker and where I can also easily add Cloudflare addresses.

In the absolute it would of course be better to be able to block in GeoIP on openMPTCP, but I know it's not simple, Unifi has been in beta on this subject for more than two years without making any progress for example...

Thanks for your work !

closed 2021-03-01 by github-actions[bot]