Ysurac/openmptcprouter #1420: Erreur kernel et instabilités

kdesroches89, 2020-11-25:
Bonjour,
Depuis plusieurs semaines et depuis la version MTCP 0.56.5 sous Rasberry 4B et 0.1022 en noyau 5.4.74-mtcp en noyau sur du Debian 10 des problèmes  internet sur un VPS OVH et aussi AWS . 

Dans les logs je trouve ca : 

Wed Nov 25 14:37:45 2020 kern.err kernel: [12515.753905] mptcp_close: Close of m
eta_sk with tok 0x9e41a7f1
Wed Nov 25 14:37:46 2020 kern.err kernel: [12515.782176] mptcp_del_sock: Removin
g subsock tok 0x9e41a7f1 pi:1 state 7 is_meta? 0
Wed Nov 25 14:37:46 2020 kern.err kernel: [12515.833840] mptcp_del_sock: Removin
g subsock tok 0x9e41a7f1 pi:2 state 7 is_meta? 0
Wed Nov 25 14:37:46 2020 kern.err kernel: [12515.841501] mptcp_sock_destruct des
troying meta-sk token 0x9e41a7f1

Par ailleurs je vois également que ShadowSock à des soucis de temps en temps ou que le VPN s'arrete (message d'erreur dans les informations réseaux) .

J'ai tenté de réinitaliser ainsi que de refaire une machine sur AWS mais rien y fait et j'ai toujours les instabilités . 

L'agrégation est faite avec une Freebox et un modem router 4G Lynksys sous opérateur Free . 

Merci pour d'avance pour l'aide 
Kilian

Ysurac, 2020-11-25:
Il serait vraiment plus pratique d'utiliser le template...

Il faudrait voir dans Etat->Journal système ce qu'il y a quand une déconnexion est visible.

kdesroches89, 2020-11-25:
Effectivement et voici mes logs : 

`Wed Nov 25 15:09:11 2020 kern.err kernel: [14401.000999] mptcp_add_sock: token 0x5bae6cfa pi 2, src_addr:0.0.0.0:0 dst_addr:0.0.0.0:0
Wed Nov 25 15:09:11 2020 kern.err kernel: [14401.009126] __mptcp_init4_subsockets: token 0x5bae6cfa pi 2 src_addr:192.168.0.2:0 dst_addr:15.188.215.124:65228 ifidx: 12
Wed Nov 25 15:09:11 2020 kern.err kernel: [14401.069122] mptcp_close: Close of meta_sk with tok 0x5bae6cfa
Wed Nov 25 15:09:11 2020 kern.err kernel: [14401.085811] mptcp_del_sock: Removing subsock tok 0x5bae6cfa pi:1 state 7 is_meta? 0
Wed Nov 25 15:09:11 2020 kern.err kernel: [14401.136562] mptcp_del_sock: Removing subsock tok 0x5bae6cfa pi:2 state 7 is_meta? 0
Wed Nov 25 15:09:11 2020 kern.err kernel: [14401.144229] mptcp_sock_destruct destroying meta-sk token 0x5bae6cfa
Wed Nov 25 15:09:12 2020 kern.err kernel: [14402.211263] mptcp_alloc_mpcb: created mpcb with token 0x41bafb12
Wed Nov 25 15:09:12 2020 kern.err kernel: [14402.217289] mptcp_add_sock: token 0x41bafb12 pi 1, src_addr:192.168.3.2:60831 dst_addr:15.188.215.124:65500
Wed Nov 25 15:09:12 2020 kern.err kernel: [14402.281123] mptcp_add_sock: token 0x41bafb12 pi 2, src_addr:0.0.0.0:0 dst_addr:0.0.0.0:0
Wed Nov 25 15:09:12 2020 kern.err kernel: [14402.289242] __mptcp_init4_subsockets: token 0x41bafb12 pi 2 src_addr:192.168.0.2:0 dst_addr:15.188.215.124:65500 ifidx: 12
Wed Nov 25 15:09:12 2020 kern.err kernel: [14402.340467] mptcp_close: Close of meta_sk with tok 0x41bafb12
Wed Nov 25 15:09:12 2020 kern.err kernel: [14402.375193] mptcp_del_sock: Removing subsock tok 0x41bafb12 pi:1 state 7 is_meta? 0
Wed Nov 25 15:09:12 2020 kern.err kernel: [14402.426107] mptcp_del_sock: Removing subsock tok 0x41bafb12 pi:2 state 7 is_meta? 0
Wed Nov 25 15:09:12 2020 kern.err kernel: [14402.433769] mptcp_sock_destruct destroying meta-sk token 0x41bafb12
Wed Nov 25 15:09:14 2020 daemon.info dnsmasq[10577]: 2850 192.168.100.1/33124 query[A] bing.com from 192.168.100.1
Wed Nov 25 15:09:14 2020 daemon.info dnsmasq[10577]: 2850 192.168.100.1/33124 cached bing.com is 13.107.21.200
Wed Nov 25 15:09:14 2020 daemon.info dnsmasq[10577]: 2850 192.168.100.1/33124 cached bing.com is 204.79.197.200
Wed Nov 25 15:09:14 2020 daemon.info dnsmasq[10577]: 2851 192.168.100.1/33124 query[AAAA] bing.com from 192.168.100.1
Wed Nov 25 15:09:14 2020 daemon.info dnsmasq[10577]: 2851 192.168.100.1/33124 cached bing.com is 2620:1ec:c11::200
Wed Nov 25 15:09:14 2020 kern.err kernel: [14404.331309] mptcp_alloc_mpcb: created mpcb with token 0xa00d53bb
Wed Nov 25 15:09:14 2020 kern.err kernel: [14404.337332] mptcp_add_sock: token 0xa00d53bb pi 1, src_addr:192.168.3.2:17977 dst_addr:15.188.215.124:65228
Wed Nov 25 15:09:14 2020 kern.err kernel: [14404.380299] mptcp_add_sock: token 0xa00d53bb pi 2, src_addr:0.0.0.0:0 dst_addr:0.0.0.0:0
Wed Nov 25 15:09:14 2020 kern.err kernel: [14404.388464] __mptcp_init4_subsockets: token 0xa00d53bb pi 2 src_addr:192.168.0.2:0 dst_addr:15.188.215.124:65228 ifidx: 12
Wed Nov 25 15:09:14 2020 kern.err kernel: [14404.444945] mptcp_close: Close of meta_sk with tok 0xa00d53bb
Wed Nov 25 15:09:14 2020 kern.err kernel: [14404.468266] mptcp_del_sock: Removing subsock tok 0xa00d53bb pi:1 state 7 is_meta? 0
Wed Nov 25 15:09:14 2020 kern.err kernel: [14404.517992] mptcp_del_sock: Removing subsock tok 0xa00d53bb pi:2 state 7 is_meta? 0
Wed Nov 25 15:09:14 2020 kern.err kernel: [14404.525686] mptcp_sock_destruct destroying meta-sk token 0xa00d53bb
Wed Nov 25 15:09:15 2020 kern.err kernel: [14405.390603] mptcp_alloc_mpcb: created mpcb with token 0x39a52c29
Wed Nov 25 15:09:15 2020 kern.err kernel: [14405.396622] mptcp_add_sock: token 0x39a52c29 pi 1, src_addr:192.168.3.2:60837 dst_addr:15.188.215.124:65500
Wed Nov 25 15:09:15 2020 kern.err kernel: [14405.459623] mptcp_add_sock: token 0x39a52c29 pi 2, src_addr:0.0.0.0:0 dst_addr:0.0.0.0:0
Wed Nov 25 15:09:15 2020 kern.err kernel: [14405.467743] __mptcp_init4_subsockets: token 0x39a52c29 pi 2 src_addr:192.168.0.2:0 dst_addr:15.188.215.124:65500 ifidx: 12
Wed Nov 25 15:09:15 2020 kern.err kernel: [14405.523071] mptcp_close: Close of meta_sk with tok 0x39a52c29
Wed Nov 25 15:09:15 2020 kern.err kernel: [14405.553758] mptcp_del_sock: Removing subsock tok 0x39a52c29 pi:2 state 7 is_meta? 0
Wed Nov 25 15:09:15 2020 kern.err kernel: [14405.576701] mptcp_del_sock: Removing subsock tok 0x39a52c29 pi:1 state 7 is_meta? 0
Wed Nov 25 15:09:15 2020 kern.err kernel: [14405.584372] mptcp_sock_destruct destroying meta-sk token 0x39a52c29
Wed Nov 25 15:09:16 2020 daemon.info dnsmasq[10577]: 2852 192.168.100.175/51191 query[A] cdn.samsungcloudsolution.com from 192.168.100.175
Wed Nov 25 15:09:16 2020 daemon.info dnsmasq[10577]: 2852 192.168.100.175/51191 cached cdn.samsungcloudsolution.com is <CNAME>
Wed Nov 25 15:09:16 2020 daemon.info dnsmasq[10577]: 2852 192.168.100.175/51191 cached cdn.samsungcloudsolution.com.edgesuite.net is <CNAME>
Wed Nov 25 15:09:16 2020 daemon.info dnsmasq[10577]: 2852 192.168.100.175/51191 cached a1963.w73.akamai.net is 95.101.183.163
Wed Nov 25 15:09:16 2020 daemon.info dnsmasq[10577]: 2852 192.168.100.175/51191 cached a1963.w73.akamai.net is 95.101.183.154
Wed Nov 25 15:09:16 2020 daemon.info dnsmasq[10577]: 2852 192.168.100.175/51191 cached a1963.w73.akamai.net is 95.101.183.138
Wed Nov 25 15:09:16 2020 daemon.info dnsmasq[10577]: 2852 192.168.100.175/51191 cached a1963.w73.akamai.net is 95.101.183.176
[... cut ...]
Wed Nov 25 15:09:51 2020 daemon.info dnsmasq[10577]: 2914 192.168.100.127/57459 query[A] aghome.kikimath.net from 192.168.100.127
Wed Nov 25 15:09:51 2020 daemon.info dnsmasq[10577]: 2914 192.168.100.127/57459 /tmp/hosts/dhcp.cfg01411c aghome.kikimath.net is 192.168.100.1
Wed Nov 25 15:09:51 2020 daemon.info dnsmasq[10577]: 2915 192.168.100.127/51498 query[AAAA] aghome.kikimath.net from 192.168.100.127
Wed Nov 25 15:09:51 2020 daemon.info dnsmasq[10577]: 2915 192.168.100.127/51498 /tmp/hosts/dhcp.cfg01411c aghome.kikimath.net is fd29:375f:e011::1`

kdesroches89, 2020-11-25:
@Ysurac voici mes logs au moment d'une déconnexion . Je ne saisis pas vraiment ce qui se passe avec OMR-Bypass :
[logbug.log](https://github.com/Ysurac/openmptcprouter/files/5597278/logbug.log)

Ysurac, 2020-11-25:
Pas de fichiers attachés svp, je dois télécharger à chaque fois que je veux voir l'issue et c'est rapidement pénible.

kdesroches89, 2020-11-25:
Pas de soucis : 

Wed Nov 25 15:25:43 2020 daemon.err omr-tracker[10693]: [info] DNS disabled: main_dns
Wed Nov 25 15:25:43 2020 daemon.info dnsmasq[10577]: 4027 192.168.100.198/51841 query[A] push.services.mozilla.com from 192.168.100.198
Wed Nov 25 15:25:43 2020 daemon.info dnsmasq[10577]: 4027 192.168.100.198/51841 cached push.services.mozilla.com is <CNAME>
Wed Nov 25 15:25:43 2020 daemon.info dnsmasq[10577]: 4027 192.168.100.198/51841 cached autopush.prod.mozaws.net is 34.211.185.195
Wed Nov 25 15:25:43 2020 daemon.info dnsmasq[10577]: 4022 192.168.100.102/30976 reply netcomv2.netatmo.net is <CNAME>
Wed Nov 25 15:25:43 2020 daemon.info dnsmasq[10577]: 4022 192.168.100.102/30976 reply nv2.netatmo.net is <CNAME>
Wed Nov 25 15:25:43 2020 daemon.info dnsmasq[10577]: 4022 192.168.100.102/30976 reply nv2.trafficmanager.net is 13.94.189.183
Wed Nov 25 15:25:43 2020 daemon.info dnsmasq[10577]: 4028 192.168.100.127/53901 query[A] alive.github.com from 192.168.100.127
Wed Nov 25 15:25:43 2020 daemon.info dnsmasq[10577]: 4028 192.168.100.127/53901 forwarded alive.github.com to 127.0.0.1
Wed Nov 25 15:25:43 2020 daemon.info dnsmasq[10577]: 4029 192.168.100.102/26595 query[AAAA] alive.github.com from 192.168.100.102
Wed Nov 25 15:25:43 2020 daemon.info dnsmasq[10577]: 4029 192.168.100.102/26595 forwarded alive.github.com to 127.0.0.1
Wed Nov 25 15:25:43 2020 daemon.info v2ray: Setting transparent proxy on port: 1897
Wed Nov 25 15:25:43 2020 daemon.err omr-tracker[10693]: [info] Setting transparent proxy on port: 1897
Wed Nov 25 15:25:43 2020 daemon.info v2ray: Transparent proxy mode: default
Wed Nov 25 15:25:43 2020 daemon.err omr-tracker[10693]: [info] Transparent proxy mode: default
Wed Nov 25 15:25:43 2020 user.notice v2ray: add rules
Wed Nov 25 15:25:43 2020 daemon.info dnsmasq[10577]: 4029 192.168.100.102/26595 reply alive.github.com is <CNAME>
Wed Nov 25 15:25:43 2020 daemon.info dnsmasq[10577]: 4030 192.168.100.127/59102 query[AAAA] live.github.com from 192.168.100.127
Wed Nov 25 15:25:43 2020 daemon.info dnsmasq[10577]: 4030 192.168.100.127/59102 forwarded live.github.com to 127.0.0.1
Wed Nov 25 15:25:43 2020 daemon.info dnsmasq[10577]: 4028 192.168.100.127/53901 reply alive.github.com is <CNAME>
Wed Nov 25 15:25:43 2020 daemon.info dnsmasq[10577]: 4028 192.168.100.127/53901 reply live.github.com is 140.82.114.25
Wed Nov 25 15:25:43 2020 daemon.info dnsmasq[10577]: 4031 192.168.100.127/42094 query[AAAA] live.github.com from 192.168.100.127
Wed Nov 25 15:25:43 2020 daemon.info dnsmasq[10577]: 4031 192.168.100.127/42094 forwarded live.github.com to 127.0.0.1
Wed Nov 25 15:25:44 2020 kern.err kernel: [15393.867589] mptcp_address_worker could not find id
Wed Nov 25 15:25:44 2020 user.notice v2ray: Reload omr-bypass rules
Wed Nov 25 15:25:44 2020 daemon.info dnsmasq[10577]: 4032 192.168.100.127/34927 query[A] aghome.kikimath.net from 192.168.100.127
Wed Nov 25 15:25:44 2020 daemon.info dnsmasq[10577]: 4032 192.168.100.127/34927 /tmp/hosts/dhcp.cfg01411c aghome.kikimath.net is 192.168.100.1
Wed Nov 25 15:25:44 2020 daemon.info dnsmasq[10577]: 4033 192.168.100.127/51461 query[AAAA] aghome.kikimath.net from 192.168.100.127
Wed Nov 25 15:25:44 2020 daemon.info dnsmasq[10577]: 4033 192.168.100.127/51461 /tmp/hosts/dhcp.cfg01411c aghome.kikimath.net is fd29:375f:e011::1
Wed Nov 25 15:25:44 2020 user.notice omr-bypass: Starting OMR-ByPass...
Wed Nov 25 15:25:44 2020 daemon.info dnsmasq[10577]: 4034 192.168.100.127/58013 query[A] push.services.mozilla.com from 192.168.100.127
Wed Nov 25 15:25:44 2020 daemon.info dnsmasq[10577]: 4034 192.168.100.127/58013 cached push.services.mozilla.com is <CNAME>
Wed Nov 25 15:25:44 2020 daemon.info dnsmasq[10577]: 4034 192.168.100.127/58013 cached autopush.prod.mozaws.net is 34.211.185.195
Wed Nov 25 15:25:44 2020 daemon.info dnsmasq[10577]: 4035 192.168.100.127/37148 query[AAAA] push.services.mozilla.com from 192.168.100.127
Wed Nov 25 15:25:44 2020 daemon.info dnsmasq[10577]: 4035 192.168.100.127/37148 forwarded push.services.mozilla.com to 127.0.0.1
Wed Nov 25 15:25:44 2020 daemon.info dnsmasq[10577]: 4035 192.168.100.127/37148 reply push.services.mozilla.com is <CNAME>
Wed Nov 25 15:25:44 2020 daemon.info dnsmasq[10577]: 4036 192.168.100.127/41208 query[AAAA] autopush.prod.mozaws.net from 192.168.100.127
Wed Nov 25 15:25:44 2020 daemon.info dnsmasq[10577]: 4036 192.168.100.127/41208 forwarded autopush.prod.mozaws.net to 127.0.0.1
Wed Nov 25 15:25:44 2020 daemon.info dnsmasq[10577]: 4037 192.168.100.127/42347 query[AAAA] autopush.prod.mozaws.net from 192.168.100.127
Wed Nov 25 15:25:44 2020 daemon.info dnsmasq[10577]: 4037 192.168.100.127/42347 forwarded autopush.prod.mozaws.net to 127.0.0.1
Wed Nov 25 15:25:44 2020 daemon.info dnsmasq[10577]: 4038 192.168.100.127/56732 query[AAAA] live.github.com from 192.168.100.127
Wed Nov 25 15:25:44 2020 daemon.info dnsmasq[10577]: 4038 192.168.100.127/56732 forwarded live.github.com to 127.0.0.1
Wed Nov 25 15:25:44 2020 daemon.info dnsmasq[10577]: 4039 192.168.100.127/50902 query[A] aghome.kikimath.net from 192.168.100.127
Wed Nov 25 15:25:44 2020 daemon.info dnsmasq[10577]: 4039 192.168.100.127/50902 /tmp/hosts/dhcp.cfg01411c aghome.kikimath.net is 192.168.100.1
Wed Nov 25 15:25:44 2020 daemon.info dnsmasq[10577]: 4040 192.168.100.127/59176 query[AAAA] aghome.kikimath.net from 192.168.100.127
[... cut ...]
Wed Nov 25 15:25:46 2020 daemon.info dnsmasq[10577]: 4081 192.168.100.102/52348 query[A] aghome.kikimath.net from 192.168.100.102
Wed Nov 25 15:25:46 2020 daemon.info dnsmasq[10577]: 4081 192.168.100.102/52348 /tmp/hosts/dhcp.cfg01411c aghome.kikimath.net is 192.168.100.1
Wed Nov 25 15:25:46 2020 daemon.info dnsmasq[10577]: 4082 192.168.100.102/46867 query[AAAA] aghome.kikimath.net from 192.168.100.102
Wed Nov 25 15:25:46 2020 daemon.info dnsmasq[10577]: 4082 192.168.100.102/46867 /tmp/hosts/dhcp.cfg01411c aghome.kikimath.net is fd29:375f:e011::1

Ysurac, 2020-11-25:
Le proxy V2Ray est utilisé et je vois OpenVPN, MPTCP over VPN est utilisé ? Les clefs n'ont pas l'air d'être à jour.
Si c'est le cas, je doute que V2Ray fonctionne correctement avec MPTCP over VPN, je n'ai pas encore configuré v2ray pour ça.

kdesroches89, 2020-11-25:
@Ysurac j'avoue qu'au départ j'utilisais ShadowSock mais récemment vu l'instabilité j'ai changé avec V2Ray , ça n'a rien changé . 
Ce que je ne comprend pas dans le contexte ou pourtant dans Système > OpenMPTCPRouter > paramètres avancées je met bien la récupération forcée des clés serveurs . 

Je viens de remettre ShadowSocks pour serveur proxy par défaut en forçant la MaJ des clés . Sinon pour répondre a la question , oui OpenVPN, MPTCP over VPN sont utilisés .

Ysurac, 2020-11-25:
Y'a trop de log niveau dnsmasq, la journalisation des requêtes est activée ?

kdesroches89, 2020-11-25:
@Ysurac oui car c'était mieux pour essayer de comprendre le problème et debug .
je viens de le désactiver .

kdesroches89, 2020-11-25:
@Ysurac voici les logs sans la journalisation : 

Wed Nov 25 19:45:38 2020 kern.err kernel: [30988.886370] mptcp_del_sock: Removing subsock tok 0x227c31e0 pi:2 state 7 is_meta? 0
Wed Nov 25 19:45:38 2020 kern.err kernel: [30989.001917] mptcp_del_sock: Removing subsock tok 0x1862bea6 pi:1 state 7 is_meta? 0
Wed Nov 25 19:45:38 2020 kern.err kernel: [30989.053459] mptcp_del_sock: Removing subsock tok 0x1862bea6 pi:2 state 7 is_meta? 0
Wed Nov 25 19:45:38 2020 kern.err kernel: [30989.061130] mptcp_sock_destruct destroying meta-sk token 0x1862bea6
Wed Nov 25 19:45:39 2020 kern.err kernel: [30989.204626] mptcp_del_sock: Removing subsock tok 0x227c31e0 pi:1 state 7 is_meta? 0
Wed Nov 25 19:45:39 2020 kern.err kernel: [30989.212304] mptcp_sock_destruct destroying meta-sk token 0x227c31e0
Wed Nov 25 19:45:39 2020 kern.err kernel: [30989.705319] mptcp_alloc_mpcb: created mpcb with token 0xe65ee1ef
Wed Nov 25 19:45:39 2020 kern.err kernel: [30989.711344] mptcp_add_sock: token 0xe65ee1ef pi 1, src_addr:192.168.3.2:19301 dst_addr:51.91.11.146:65500
Wed Nov 25 19:45:39 2020 kern.err kernel: [30989.720989] mptcp_alloc_mpcb: created mpcb with token 0x9158cfe4
Wed Nov 25 19:45:39 2020 kern.err kernel: [30989.727011] mptcp_add_sock: token 0x9158cfe4 pi 1, src_addr:127.0.0.1:1101 dst_addr:192.168.3.2:19301
Wed Nov 25 19:45:39 2020 kern.err kernel: [30989.751940] mptcp_add_sock: token 0xe65ee1ef pi 2, src_addr:0.0.0.0:0 dst_addr:0.0.0.0:0
Wed Nov 25 19:45:39 2020 kern.err kernel: [30989.760061] __mptcp_init4_subsockets: token 0xe65ee1ef pi 2 src_addr:192.168.0.2:0 dst_addr:51.91.11.146:65500 ifidx: 12
Wed Nov 25 19:45:39 2020 kern.err kernel: [30989.955683] mptcp_alloc_mpcb: created mpcb with token 0xd0ae6714
Wed Nov 25 19:45:39 2020 kern.err kernel: [30989.961703] mptcp_add_sock: token 0xd0ae6714 pi 1, src_addr:192.168.3.2:23763 dst_addr:15.188.215.124:65101
Wed Nov 25 19:45:40 2020 kern.err kernel: [30990.340709] mptcp_add_sock: token 0xd0ae6714 pi 2, src_addr:0.0.0.0:0 dst_addr:0.0.0.0:0
Wed Nov 25 19:45:40 2020 kern.err kernel: [30990.348821] __mptcp_init4_subsockets: token 0xd0ae6714 pi 2 src_addr:192.168.0.2:0 dst_addr:15.188.215.124:65101 ifidx: 12
Wed Nov 25 19:45:40 2020 kern.err kernel: [30990.396494] mptcp_del_sock: Removing subsock tok 0xad3453c pi:2 state 7 is_meta? 0
Wed Nov 25 19:45:40 2020 kern.err kernel: [30990.538828] mptcp_alloc_mpcb: created mpcb with token 0x5741964
Wed Nov 25 19:45:40 2020 kern.err kernel: [30990.544760] mptcp_add_sock: token 0x5741964 pi 1, src_addr:192.168.3.2:23767 dst_addr:15.188.215.124:65101
Wed Nov 25 19:45:40 2020 kern.err kernel: [30990.649758] mptcp_del_sock: Removing subsock tok 0xad3453c pi:1 state 7 is_meta? 0
Wed Nov 25 19:45:40 2020 kern.err kernel: [30990.657343] mptcp_sock_destruct destroying meta-sk token 0xad3453c
Wed Nov 25 19:45:40 2020 kern.err kernel: [30990.706379] mptcp_close: Close of meta_sk with tok 0xe65ee1ef
Wed Nov 25 19:45:40 2020 kern.err kernel: [30990.712202] mptcp_close: Close of meta_sk with tok 0xd0ae6714
Wed Nov 25 19:45:40 2020 kern.err kernel: [30990.712227] mptcp_del_sock: Removing subsock tok 0xe65ee1ef pi:2 state 7 is_meta? 0
Wed Nov 25 19:45:40 2020 kern.err kernel: [30990.718246] mptcp_close: Close of meta_sk with tok 0x9158cfe4
Wed Nov 25 19:45:40 2020 kern.err kernel: [30990.725639] mptcp_del_sock: Removing subsock tok 0xe65ee1ef pi:1 state 7 is_meta? 0
Wed Nov 25 19:45:40 2020 kern.err kernel: [30990.731636] mptcp_del_sock: Removing subsock tok 0x9158cfe4 pi:1 state 7 is_meta? 0
Wed Nov 25 19:45:40 2020 kern.err kernel: [30990.739579] mptcp_sock_destruct destroying meta-sk token 0xe65ee1ef
Wed Nov 25 19:45:40 2020 kern.err kernel: [30990.740109] mptcp_del_sock: Removing subsock tok 0xd0ae6714 pi:1 state 7 is_meta? 0
Wed Nov 25 19:45:40 2020 kern.err kernel: [30990.746727] mptcp_sock_destruct destroying meta-sk token 0x9158cfe4
Wed Nov 25 19:45:40 2020 kern.err kernel: [30990.834684] mptcp_add_sock: token 0x5741964 pi 2, src_addr:0.0.0.0:0 dst_addr:0.0.0.0:0
Wed Nov 25 19:45:40 2020 kern.err kernel: [30990.842732] __mptcp_init4_subsockets: token 0x5741964 pi 2 src_addr:192.168.0.2:0 dst_addr:15.188.215.124:65101 ifidx: 12
Wed Nov 25 19:45:40 2020 kern.err kernel: [30990.854509] mptcp_close: Close of meta_sk with tok 0x5741964
Wed Nov 25 19:45:40 2020 kern.err kernel: [30990.942200] mptcp_del_sock: Removing subsock tok 0xd0ae6714 pi:2 state 7 is_meta? 0
Wed Nov 25 19:45:40 2020 kern.err kernel: [30990.949899] mptcp_sock_destruct destroying meta-sk token 0xd0ae6714
Wed Nov 25 19:45:40 2020 kern.err kernel: [30990.974033] mptcp_del_sock: Removing subsock tok 0x5741964 pi:1 state 7 is_meta? 0
Wed Nov 25 19:45:41 2020 kern.err kernel: [30991.075440] mptcp_del_sock: Removing subsock tok 0x5741964 pi:2 state 7 is_meta? 0
Wed Nov 25 19:45:41 2020 kern.err kernel: [30991.083025] mptcp_sock_destruct destroying meta-sk token 0x5741964
Wed Nov 25 19:45:41 2020 kern.err kernel: [30991.716128] mptcp_alloc_mpcb: created mpcb with token 0xc3d5dbd3
Wed Nov 25 19:45:41 2020 kern.err kernel: [30991.722159] mptcp_add_sock: token 0xc3d5dbd3 pi 1, src_addr:192.168.3.2:19309 dst_addr:51.91.11.146:65500
Wed Nov 25 19:45:41 2020 kern.err kernel: [30991.731803] mptcp_alloc_mpcb: created mpcb with token 0xdf578e29
Wed Nov 25 19:45:41 2020 kern.err kernel: [30991.737817] mptcp_add_sock: token 0xdf578e29 pi 1, src_addr:127.0.0.1:1101 dst_addr:192.168.3.2:19309
[... cut ...]
Wed Nov 25 19:45:56 2020 daemon.warn dnsmasq-dhcp[8935]: not giving name kikimath.kikimath.net to the DHCP lease of 192.168.100.198 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 169.254.115.125
Wed Nov 25 19:45:56 2020 daemon.warn dnsmasq-dhcp[8935]: not giving name Mathilde-tel.kikimath.net to the DHCP lease of 192.168.100.116 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.100.226
Wed Nov 25 19:45:56 2020 user.notice omr-bypass: OMR-ByPass is running

Ysurac, 2020-11-25:
Il faudrait aussi le mode debug de MPTCP.
Je vois que ça passe à un serveur de backup car le serveur maitre ne répond pas, mais MPTCP over VPN ne supporte pas d'avoir plusieurs serveurs actuellement (cas non prévu, il faudrait que je récupère la clef OpenVPN lors du changement de serveur, je vais ajouter ça pour la prochaine version).
Il y a aussi ss-tunnel qui a été activé alors qu'il ne devrait pas.

kdesroches89, 2020-11-25:
@Ysurac  Comment on passe MTCP en mode debug ?

Ysurac, 2020-11-25:
Je voulais dire qu'il est activé la, dans MPTCP, "Multipath Debug".

kdesroches89, 2020-11-25:
@Ysurac je viens de passer MPTCP en mode non debug et repasser l'ensemble sur le master .
J'ai désactivé ss-tunnel : 

[image]

kdesroches89, 2020-11-25:
@Ysurac nouveau log avec un test de téléchargement qui coupe au bout de 5 min sur un fichier de 705 Mo : 

Wed Nov 25 20:31:25 2020 daemon.info /usr/bin/ss-redir[25860]: tcp port reuse enabled
Wed Nov 25 20:31:25 2020 daemon.info /usr/bin/ss-redir[25860]: UDP relay enabled
Wed Nov 25 20:31:25 2020 daemon.info /usr/bin/ss-local[25858]: listening at 127.0.0.1:1111
Wed Nov 25 20:31:25 2020 daemon.info /usr/bin/ss-redir[25860]: udp port reuse enabled
Wed Nov 25 20:31:25 2020 daemon.info /usr/bin/ss-local[25858]: tcp port reuse enabled
Wed Nov 25 20:31:25 2020 daemon.info /usr/bin/ss-local[25858]: udprelay enabled
Wed Nov 25 20:31:25 2020 daemon.info /usr/bin/ss-local[25858]: udp port reuse enabled
Wed Nov 25 20:31:25 2020 daemon.info /usr/bin/ss-redir[25860]: running from root user
Wed Nov 25 20:31:25 2020 daemon.info /usr/bin/ss-local[25858]: running from root user
Wed Nov 25 20:31:25 2020 daemon.info /usr/bin/ss-redir[25862]: tcp port reuse enabled
Wed Nov 25 20:31:25 2020 daemon.info /usr/bin/ss-redir[25864]: enable TCP no-delay
Wed Nov 25 20:31:25 2020 daemon.info /usr/bin/ss-redir[25864]: initializing ciphers... chacha20-ietf-poly1305
Wed Nov 25 20:31:25 2020 daemon.info /usr/bin/ss-redir[25864]: listening at 0.0.0.0:1101
Wed Nov 25 20:31:25 2020 daemon.info /usr/bin/ss-redir[25864]: tcp port reuse enabled
Wed Nov 25 20:31:25 2020 daemon.info /usr/bin/ss-redir[25864]: UDP relay enabled
Wed Nov 25 20:31:25 2020 daemon.info /usr/bin/ss-redir[25864]: udp port reuse enabled
Wed Nov 25 20:31:25 2020 daemon.info /usr/bin/ss-redir[25864]: running from root user
Wed Nov 25 20:31:25 2020 daemon.info /usr/bin/ss-redir[25862]: UDP relay enabled
Wed Nov 25 20:31:25 2020 daemon.info /usr/bin/ss-redir[25862]: udp port reuse enabled
Wed Nov 25 20:31:25 2020 daemon.info /usr/bin/ss-redir[25862]: running from root user
Wed Nov 25 20:31:25 2020 user.info : using tcp fast open
Wed Nov 25 20:31:25 2020 daemon.info /usr/bin/ss-redir[25863]: enable TCP no-delay
Wed Nov 25 20:31:25 2020 daemon.info /usr/bin/ss-redir[25863]: initializing ciphers... chacha20-ietf-poly1305
Wed Nov 25 20:31:25 2020 daemon.info /usr/bin/ss-redir[25863]: listening at 0.0.0.0:1101
Wed Nov 25 20:31:25 2020 daemon.info /usr/bin/ss-redir[25863]: tcp port reuse enabled
Wed Nov 25 20:31:25 2020 daemon.info /usr/bin/ss-redir[25863]: UDP relay enabled
Wed Nov 25 20:31:25 2020 daemon.info /usr/bin/ss-redir[25863]: udp port reuse enabled
Wed Nov 25 20:31:25 2020 daemon.info /usr/bin/ss-redir[25863]: running from root user
Wed Nov 25 20:31:25 2020 daemon.info /usr/bin/ss-tunnel[25865]: initializing ciphers... chacha20-ietf-poly1305
Wed Nov 25 20:31:25 2020 daemon.info /usr/bin/ss-tunnel[25865]: listening at 0.0.0.0:5353
Wed Nov 25 20:31:25 2020 daemon.err /usr/bin/ss-tunnel[25865]: bind: Address in use
Wed Nov 25 20:31:25 2020 daemon.err /usr/bin/ss-tunnel[25865]: bind() error
Wed Nov 25 20:31:26 2020 daemon.err openvpn(omr)[3104]: event_wait : Interrupted system call (code=4)
Wed Nov 25 20:31:26 2020 daemon.notice openvpn(omr)[3104]: /sbin/ifconfig tun0 0.0.0.0
Wed Nov 25 20:31:26 2020 daemon.notice netifd: Network device 'tun0' link is down
Wed Nov 25 20:31:26 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Wed Nov 25 20:31:26 2020 daemon.notice netifd: Interface 'omrvpn' is now down
Wed Nov 25 20:31:26 2020 user.notice ModemManager: hotplug: remove network interface tun0: event processed
Wed Nov 25 20:31:26 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=tun0, subsystem=net
Wed Nov 25 20:31:26 2020 daemon.notice openvpn(omr)[3104]: /usr/libexec/openvpn-hotplug down omr tun0 1500 1562 10.255.252.6 10.255.252.5 init
Wed Nov 25 20:31:26 2020 daemon.notice openvpn(omr)[3104]: SIGTERM[hard,] received, process exiting
Wed Nov 25 20:31:26 2020 daemon.notice netifd: Interface 'omrvpn' is disabled
Wed Nov 25 20:31:26 2020 daemon.notice openvpn(omr)[26203]: OpenVPN 2.4.9 aarch64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
Wed Nov 25 20:31:26 2020 daemon.notice openvpn(omr)[26203]: library versions: OpenSSL 1.1.1h  22 Sep 2020, LZO 2.10
Wed Nov 25 20:31:26 2020 daemon.warn openvpn(omr)[26203]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Wed Nov 25 20:31:26 2020 daemon.warn openvpn(omr)[26203]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Wed Nov 25 20:31:26 2020 daemon.notice openvpn(omr)[26203]: TCP/UDP: Preserving recently used remote address: [AF_INET]51.91.11.146:65301
Wed Nov 25 20:31:26 2020 daemon.notice openvpn(omr)[26203]: Attempting to establish TCP connection with [AF_INET]51.91.11.146:65301 [nonblock]
Wed Nov 25 20:31:26 2020 user.notice omr-tracker: Launching...
Wed Nov 25 20:31:26 2020 daemon.info /usr/bin/ss-tunnel[26310]: initializing ciphers... chacha20-ietf-poly1305
Wed Nov 25 20:31:26 2020 daemon.info /usr/bin/ss-tunnel[26310]: listening at 0.0.0.0:5353
Wed Nov 25 20:31:26 2020 daemon.err /usr/bin/ss-tunnel[26310]: bind: Address in use
Wed Nov 25 20:31:26 2020 daemon.err /usr/bin/ss-tunnel[26310]: bind() error
Wed Nov 25 20:31:26 2020 user.notice omr-tracker: Launched
Wed Nov 25 20:31:26 2020 daemon.info vnstatd[14411]: Info: SIGTERM received, exiting.
Wed Nov 25 20:31:26 2020 daemon.info vnstatd[26969]: Info: vnStat daemon 2.6 started. (pid:26969 uid:0 gid:0 64-bit)
[... cut ...]
Wed Nov 25 20:43:36 2020 daemon.warn dnsmasq-dhcp[10909]: not giving name kikimath.kikimath.net to the DHCP lease of 192.168.100.198 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 169.254.115.125
Wed Nov 25 20:43:36 2020 daemon.warn dnsmasq-dhcp[10909]: not giving name Mathilde-tel.kikimath.net to the DHCP lease of 192.168.100.116 because the name exists in /tmp/hosts/dhcp.cfg01411c with address 192.168.100.226
Wed Nov 25 20:43:36 2020 user.notice omr-bypass: OMR-ByPass is running

On dirait que shadowsocks a toujours un problème . En gras ce qui semble bizarre . @Ysurac  votre avis ?

Ysurac, 2020-11-25:
Je vois encore un changement de serveur à  20:43:03.
Sinon visiblement les IPs/domaines d'omr-tracker ont été modifiés, peut être d'autres valeurs ce qui peut entraîner une détection trop précoce d'un soucis.

kdesroches89, 2020-11-25:
Oui la rebascule sur le master et non le backup . 
Le problème actuel des instabilités persistes toujours . 
Que dois je faire @Ysurac ? Merci

kdesroches89, 2020-11-25:
@Ysurac voici des logs un peu plus clair : 

Wed Nov 25 22:16:18 2020 daemon.info /usr/bin/ss-redir[1889]: tcp port reuse enabled
Wed Nov 25 22:16:18 2020 daemon.info /usr/bin/ss-redir[1889]: UDP relay enabled
Wed Nov 25 22:16:18 2020 daemon.info /usr/bin/ss-redir[1889]: udp port reuse enabled
Wed Nov 25 22:16:18 2020 daemon.info /usr/bin/ss-redir[1889]: running from root user
Wed Nov 25 22:16:18 2020 daemon.info /usr/bin/ss-redir[1887]: enable TCP no-delay
Wed Nov 25 22:16:18 2020 daemon.info /usr/bin/ss-redir[1887]: initializing ciphers... chacha20-ietf-poly1305
Wed Nov 25 22:16:18 2020 daemon.info /usr/bin/ss-redir[1887]: listening at 0.0.0.0:1100
Wed Nov 25 22:16:18 2020 daemon.info /usr/bin/ss-redir[1887]: tcp port reuse enabled
Wed Nov 25 22:16:18 2020 daemon.info /usr/bin/ss-redir[1887]: UDP relay enabled
Wed Nov 25 22:16:18 2020 daemon.info /usr/bin/ss-redir[1887]: udp port reuse enabled
Wed Nov 25 22:16:18 2020 daemon.info /usr/bin/ss-redir[1887]: running from root user
Wed Nov 25 22:16:18 2020 daemon.err openvpn(omr)[20998]: event_wait : Interrupted system call (code=4)
Wed Nov 25 22:16:18 2020 daemon.notice openvpn(omr)[20998]: /sbin/ifconfig tun0 0.0.0.0
Wed Nov 25 22:16:18 2020 daemon.notice netifd: Network device 'tun0' link is down
Wed Nov 25 22:16:18 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Wed Nov 25 22:16:18 2020 daemon.notice netifd: Interface 'omrvpn' is now down
Wed Nov 25 22:16:18 2020 user.notice ModemManager: hotplug: remove network interface tun0: event processed
Wed Nov 25 22:16:18 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=tun0, subsystem=net
Wed Nov 25 22:16:18 2020 daemon.notice openvpn(omr)[20998]: /usr/libexec/openvpn-hotplug down omr tun0 1500 1562 10.255.252.10 10.255.252.9 init
Wed Nov 25 22:16:18 2020 daemon.notice netifd: Interface 'omrvpn' is disabled
Wed Nov 25 22:16:18 2020 daemon.notice openvpn(omr)[20998]: SIGTERM[hard,] received, process exiting
Wed Nov 25 22:16:18 2020 daemon.notice openvpn(omr)[2161]: OpenVPN 2.4.9 aarch64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
Wed Nov 25 22:16:18 2020 daemon.notice openvpn(omr)[2161]: library versions: OpenSSL 1.1.1h  22 Sep 2020, LZO 2.10
Wed Nov 25 22:16:18 2020 daemon.warn openvpn(omr)[2161]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Wed Nov 25 22:16:18 2020 daemon.warn openvpn(omr)[2161]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Wed Nov 25 22:16:18 2020 daemon.notice openvpn(omr)[2161]: TCP/UDP: Preserving recently used remote address: [AF_INET]15.188.215.124:65301
Wed Nov 25 22:16:18 2020 daemon.notice openvpn(omr)[2161]: Attempting to establish TCP connection with [AF_INET]15.188.215.124:65301 [nonblock]
Wed Nov 25 22:16:19 2020 daemon.notice openvpn(omr)[2161]: TCP connection established with [AF_INET]15.188.215.124:65301
Wed Nov 25 22:16:19 2020 daemon.notice openvpn(omr)[2161]: TCP_CLIENT link local: (not bound)
Wed Nov 25 22:16:19 2020 daemon.notice openvpn(omr)[2161]: TCP_CLIENT link remote: [AF_INET]15.188.215.124:65301
Wed Nov 25 22:16:19 2020 daemon.err openvpn(omr)[2161]: VERIFY ERROR: depth=1, error=self signed certificate in certificate chain: CN=ChangeMe
Wed Nov 25 22:16:19 2020 daemon.err openvpn(omr)[2161]: OpenSSL: error:1416F086:SSL routines:tls_process_server_certificate:certificate verify failed
Wed Nov 25 22:16:19 2020 daemon.err openvpn(omr)[2161]: TLS_ERROR: BIO read tls_read_plaintext error
Wed Nov 25 22:16:19 2020 daemon.err openvpn(omr)[2161]: TLS Error: TLS object -> incoming plaintext read error
Wed Nov 25 22:16:19 2020 daemon.err openvpn(omr)[2161]: TLS Error: TLS handshake failed
Wed Nov 25 22:16:19 2020 daemon.err openvpn(omr)[2161]: Fatal TLS error (check_tls_errors_co), restarting
Wed Nov 25 22:16:19 2020 daemon.notice openvpn(omr)[2161]: SIGUSR1[soft,tls-error] received, process restarting
Wed Nov 25 22:16:23 2020 user.notice OMR-Tracker-Server: Master server up, set it back
Wed Nov 25 22:16:23 2020 user.notice OMR-Tracker-Server: 15.188.215.124 - 51.91.11.146
Wed Nov 25 22:16:23 2020 user.notice Shadowsocks: Rules DOWN

kdesroches89, 2020-11-26:
@Ysurac bonjour, il semblerai que ca soit un problème de certificat SSL lié a OpenVPN , vous confirmez ? 
Merci

Ysurac, 2020-11-26:
Comme je l'ai déjà dis, le soucis c'est que 2 serveurs VPS sont présents, ce n'est pas compatible avec MPTCP over VPN.
Ca le sera dès la prochaine version.

kdesroches89, 2020-11-26:
@Ysurac  OK , voyez vous d'autres anomalies ou m'indiquer ce qui faudrai que je fasse pour retrouver une certaine stabilité ? 

Merci

Ysurac, 2020-11-26:
Ne pas utiliser MPTCP over VPN qui n'est pas souvent utile ou alors supprimer totalement le second serveur VPS.

kdesroches89, 2020-11-27:
Merci @Ysurac pour l'aide , je clos l'incident dans l'attente de la nouvelle version 
A dispo

closed 2020-11-27 by kdesroches89