Ysurac/openmptcprouter #1421: Attribute a public IP Fullstack to server hosted behind OMR

Kaffeine75, 2020-11-25:
## Expected Behavior
I have add second public IP to my VPS. I would like dedicated it to a server hosted behind OMR.

## Current Behavior
I have followed documentation on OMR website:
https://github.com/Ysurac/openmptcprouter/wiki/VPS-Multi-IPs

Everything is ok on my primary IP and all status is green.

On my server hosted behind OMR:
- server --> outside --> second ip is ok (curl ifconfig.co return well my second IP)
- outside --> internal server --> not work

After tests, the problem seems that ports of second IP are not forward of VPS to router.

So, I need help to solve this.

Thanks in advance.

## Specifications

  - OpenMPTCProuter version: 0.56.5
  - OpenMPTCProuter VPS version: 0.1022 hosted on Proxmox VM with two virtual network card
  - OpenMPTCProuter platform: x86_64

OMR File in VPS:
> {
> "port": 65500,
> "users": [
> {
> "admin": {
> "username": "admin",
> "user_password": "xxxxxxx",
> "permissions": "admin",
> "disabled": false
> },
> "openmptcprouter": {
> "username": "openmptcprouter",
> "user_password": "xxxxxxxx",
> "shadowsocks_port": 65101,
> "disabled": false,
> "lanips": [
> "192.168.100.1/255.255.255.0"
> ],
> "vpnremoteip": "10.255.255.2",
> "vpnlocalip": "10.255.255.1",
> "ula": "xxxx:xxxx:xxxx::/48",
> "vpn": "glorytun_tcp",
> "gre_tunnels": {
> "gre-user0-ip0": {
> "shadowsocks_port": "65102",
> "local_ip": "10.255.249.1",
> "remote_ip": "10.255.249.2",
> "public_ip": "xx.xx.xx.1"
> },
> "gre-user0-ip1": {
> "shadowsocks_port": "65103",
> "local_ip": "10.255.249.5",
> "remote_ip": "10.255.249.6",
> "public_ip": "xx.xx.xx.2"
> }
> },
> "v2ray": {
> "key": "xxxxx",
> "port": "65228"
> }
> }
> }
> ],
> "debug": false,
> "allips": [
> "xx.xx.xx.1",
> "xx.xx.xx.2"
> ],
> "ipv4": "xx.xx.xx.1",
> "hostname": "ano.host.my",
> "lastchange": 1604197314.7052019
> }

Ysurac, 2020-11-26:
Can you give me the file ```/etc/shorewall/rules``` from the VPS ?

Kaffeine75, 2020-11-26:
To be complete, I zipped you the shorewall full directory.
[shorewall.zip](https://github.com/Ysurac/openmptcprouter/files/5603296/shorewall.zip)

Ysurac, 2020-11-26:
I really don't like to download file for an issue, it's difficult to easily follow an issue with that.

Kaffeine75, 2020-11-26:
`/etc/shorewall/rules`
> #
> # Shorewall version 4.0 - Sample Rules File for two-interface configuration.
> # Copyright (C) 2006-2014,2007 by the Shorewall Team
> #
> # This library is free software; you can redistribute it and/or
> # modify it under the terms of the GNU Lesser General Public
> # License as published by the Free Software Foundation; either
> # version 2.1 of the License, or (at your option) any later version.
> #
> # See the file README.txt for further details.
> #------------------------------------------------------------------------------
> # For information about entries in this file, type "man shorewall-rules"
> ######################################################################################################################################################################################################
> #ACTION         SOURCE          DEST            PROTO   DEST    SOURCE          ORIGINAL        RATE            USER/   MARK    CONNLIMIT       TIME            HEADERS         SWITCH          HELPER
> #                                                       PORT    PORT(S)         DEST            LIMIT           GROUP
> ?SECTION ALL
> ?SECTION ESTABLISHED
> ?SECTION RELATED
> ?SECTION INVALID
> ?SECTION UNTRACKED
> ?SECTION NEW
> 
> #       Don't allow connection pickup from the net
> #
> Invalid(DROP)   net             all             tcp
> #
> #       Accept DNS connections from the firewall to the network
> #
> DNS(ACCEPT)     $FW             net
> #
> #       Allow Ping from/to the VPN
> #
> Ping(ACCEPT)    vpn             $FW
> Ping(ACCEPT)    $FW             vpn
> #
> #       Allow Ping from the firewall to the network
> #
> Ping(ACCEPT)    $FW             net
> #
> # Drop Ping from the "bad" net zone.. and prevent your log from being flooded..
> #
> #Ping(DROP)     net             $FW
> Ping(ACCEPT)    net             $FW
> #
> #	Accept connection from port > 65000 for shadowsocks and glorytun on the firewall
> #
> ACCEPT          net             $FW             tcp     65000-65535
> ACCEPT          net             $FW             udp     65000-65535
> #
> #	Accept connection from SSH to the firewall
> #
> ACCEPT          net             $FW             tcp     65222
> #
> #	DHCP forward to the VPN from the firewall
> #
> DHCPfwd(ACCEPT)	$FW		vpn
> #
> #	Redirect all port from 1 to 64999 to the VPN client from the network
> #
> DNAT		net		vpn:$OMR_ADDR	tcp	1-64999
> DNAT		net		vpn:$OMR_ADDR	udp	1-64999
> DNAT		net		vpn:$OMR_ADDR	tcp	0-64999	-	10.255.249.6	# OMR openmptcprouter redirect router 0-64999 port tcp to 10.255.249.6
> DNAT		net		vpn:$OMR_ADDR	udp	0-64999	-	10.255.249.6	# OMR openmptcprouter redirect router 0-64999 port udp to 10.255.249.6
> ACCEPT		net		$FW		tcp	65101	# OMR openmptcprouter open shadowsocks port tcp
> ACCEPT		net		$FW		udp	65101	# OMR openmptcprouter open shadowsocks port udp
> DNAT		net		vpn:$OMR_ADDR	tcp	0-64999	-	10.255.249.6	# OMR openmptcprouter redirect router 0-64999 port tcp to 10.255.249.6
> DNAT		net		vpn:$OMR_ADDR	udp	0-64999	-	10.255.249.6	# OMR openmptcprouter redirect router 0-64999 port udp to 10.255.249.6
> DNAT		net		vpn:$OMR_ADDR	tcp	0-64999	-	10.255.249.6	# OMR openmptcprouter redirect router 0-64999 port tcp to 10.255.249.6
> DNAT		net		vpn:$OMR_ADDR	udp	0-64999	-	10.255.249.6	# OMR openmptcprouter redirect router 0-64999 port udp to 10.255.249.6
> DNAT		net		vpn:$OMR_ADDR	tcp	21	# OMR openmptcprouter redirect router 21 port tcp
> DNAT		net		vpn:$OMR_ADDR	udp	21	# OMR openmptcprouter redirect router 21 port udp
> DNAT		net		vpn:$OMR_ADDR	tcp	80	# OMR openmptcprouter redirect router 80 port tcp
> DNAT		net		vpn:$OMR_ADDR	udp	80	# OMR openmptcprouter redirect router 80 port udp
> DNAT		net		vpn:$OMR_ADDR	tcp	25	# OMR openmptcprouter redirect router 25 port tcp
> DNAT		net		vpn:$OMR_ADDR	udp	25	# OMR openmptcprouter redirect router 25 port udp
> DNAT		net		vpn:$OMR_ADDR	tcp	110	# OMR openmptcprouter redirect router 110 port tcp
> DNAT		net		vpn:$OMR_ADDR	udp	110	# OMR openmptcprouter redirect router 110 port udp
> DNAT		net		vpn:$OMR_ADDR	tcp	143	# OMR openmptcprouter redirect router 143 port tcp
> DNAT		net		vpn:$OMR_ADDR	udp	143	# OMR openmptcprouter redirect router 143 port udp
> DNAT		net		vpn:$OMR_ADDR	tcp	443	# OMR openmptcprouter redirect router 443 port tcp
> DNAT		net		vpn:$OMR_ADDR	udp	443	# OMR openmptcprouter redirect router 443 port udp
> DNAT		net		vpn:$OMR_ADDR	tcp	465	# OMR openmptcprouter redirect router 465 port tcp
> DNAT		net		vpn:$OMR_ADDR	udp	465	# OMR openmptcprouter redirect router 465 port udp
> DNAT		net		vpn:$OMR_ADDR	tcp	993	# OMR openmptcprouter redirect router 993 port tcp
> DNAT		net		vpn:$OMR_ADDR	udp	993	# OMR openmptcprouter redirect router 993 port udp
> DNAT		net		vpn:$OMR_ADDR	tcp	995	# OMR openmptcprouter redirect router 995 port tcp
> DNAT		net		vpn:$OMR_ADDR	udp	995	# OMR openmptcprouter redirect router 995 port udp
> DNAT		net		vpn:$OMR_ADDR	tcp	0-64999	-	10.255.249.6	# OMR openmptcprouter redirect router 0-64999 port tcp to 10.255.249.6
> DNAT		net		vpn:$OMR_ADDR	udp	0-64999	-	10.255.249.6	# OMR openmptcprouter redirect router 0-64999 port udp to 10.255.249.6

Ysurac, 2020-11-26:
You should remove duplicated rules (a bug in the API that will be solved in next release) and restart shorewall on the VPS (```/etc/init.d/shorewall restart```)
Then on the router you can check via SSH using tcpdump if you have something on the interface with 10.255.249.6 (```tcpdump -i <interface>```)

Kaffeine75, 2020-11-27:
I have remove duplicate rules. After change, I have well shorewall restart.
```
#
# Shorewall version 4.0 - Sample Rules File for two-interface configuration.
# Copyright (C) 2006-2014,2007 by the Shorewall Team
# This library is free software; you can redistribute it and/or
# modify it under the terms of the GNU Lesser General Public
# License as published by the Free Software Foundation; either
# version 2.1 of the License, or (at your option) any later version.
# See the file README.txt for further details.
#------------------------------------------------------------------------------
# For information about entries in this file, type "man shorewall-rules"
######################################################################################################################################################################################################
#ACTION         SOURCE          DEST            PROTO   DEST    SOURCE          ORIGINAL        RATE            USER/   MARK    CONNLIMIT       TIME            HEADERS         SWITCH          HELPER
#                                                       PORT    PORT(S)         DEST            LIMIT           GROUP
?SECTION ALL
?SECTION ESTABLISHED
?SECTION RELATED
?SECTION INVALID
?SECTION UNTRACKED
?SECTION NEW

#       Don't allow connection pickup from the net
Invalid(DROP)   net             all             tcp
#       Accept DNS connections from the firewall to the network
DNS(ACCEPT)     $FW             net
#       Allow Ping from/to the VPN
Ping(ACCEPT)    vpn             $FW
Ping(ACCEPT)    $FW             vpn
#       Allow Ping from the firewall to the network
Ping(ACCEPT)    $FW             net
# Drop Ping from the "bad" net zone.. and prevent your log from being flooded..
#Ping(DROP)     net             $FW
Ping(ACCEPT)    net             $FW
#	Accept connection from port > 65000 for shadowsocks and glorytun on the firewall
ACCEPT          net             $FW             tcp     65000-65535
ACCEPT          net             $FW             udp     65000-65535
#	Accept connection from SSH to the firewall
ACCEPT          net             $FW             tcp     65222
#	DHCP forward to the VPN from the firewall
DHCPfwd(ACCEPT)	$FW		vpn
#	Redirect all port from 1 to 64999 to the VPN client from the network
DNAT		net		vpn:$OMR_ADDR	tcp	1-64999
DNAT		net		vpn:$OMR_ADDR	udp	1-64999
DNAT		net		vpn:$OMR_ADDR	tcp	0-64999	-	10.255.249.6	# OMR openmptcprouter redirect router 0-64999 port tcp to 10.255.249.6
DNAT		net		vpn:$OMR_ADDR	udp	0-64999	-	10.255.249.6	# OMR openmptcprouter redirect router 0-64999 port udp to 10.255.249.6
ACCEPT		net		$FW		tcp	65101	# OMR openmptcprouter open shadowsocks port tcp
ACCEPT		net		$FW		udp	65101	# OMR openmptcprouter open shadowsocks port udp
DNAT		net		vpn:$OMR_ADDR	tcp	21	# OMR openmptcprouter redirect router 21 port tcp
DNAT		net		vpn:$OMR_ADDR	udp	21	# OMR openmptcprouter redirect router 21 port udp
DNAT		net		vpn:$OMR_ADDR	tcp	80	# OMR openmptcprouter redirect router 80 port tcp
DNAT		net		vpn:$OMR_ADDR	udp	80	# OMR openmptcprouter redirect router 80 port udp
DNAT		net		vpn:$OMR_ADDR	tcp	25	# OMR openmptcprouter redirect router 25 port tcp
DNAT		net		vpn:$OMR_ADDR	udp	25	# OMR openmptcprouter redirect router 25 port udp
DNAT		net		vpn:$OMR_ADDR	tcp	110	# OMR openmptcprouter redirect router 110 port tcp
DNAT		net		vpn:$OMR_ADDR	udp	110	# OMR openmptcprouter redirect router 110 port udp
DNAT		net		vpn:$OMR_ADDR	tcp	143	# OMR openmptcprouter redirect router 143 port tcp
DNAT		net		vpn:$OMR_ADDR	udp	143	# OMR openmptcprouter redirect router 143 port udp
DNAT		net		vpn:$OMR_ADDR	tcp	443	# OMR openmptcprouter redirect router 443 port tcp
DNAT		net		vpn:$OMR_ADDR	udp	443	# OMR openmptcprouter redirect router 443 port udp
DNAT		net		vpn:$OMR_ADDR	tcp	465	# OMR openmptcprouter redirect router 465 port tcp
DNAT		net		vpn:$OMR_ADDR	udp	465	# OMR openmptcprouter redirect router 465 port udp
DNAT		net		vpn:$OMR_ADDR	tcp	993	# OMR openmptcprouter redirect router 993 port tcp
DNAT		net		vpn:$OMR_ADDR	udp	993	# OMR openmptcprouter redirect router 993 port udp
DNAT		net		vpn:$OMR_ADDR	tcp	995	# OMR openmptcprouter redirect router 995 port tcp
DNAT		net		vpn:$OMR_ADDR	udp	995	# OMR openmptcprouter redirect router 995 port udp

```

Result for  `tcpdump -i gre4-omrip1gre -v` (Ctrl+c after two minutes):
```
root@OpenMPTCProuter:~# tcpdump -i gre4-omrip1gre -v
tcpdump: listening on gre4-omrip1gre, link-type LINUX_SLL (Linux cooked v1), capture size 262144 bytes
^C
0 packets captured
2 packets received by filter
0 packets dropped by kernel
```

Ysurac, 2020-11-27:
I think I see the error, you need in firewall as "External IP address" use the public IP you want to use on the VPS.
I need to test and take screenshot for the doc ;)

Ysurac, 2020-11-27:
There is also an issue with gre tunnel, this will be solved with next VPS script update.

closed 2021-03-03 by github-actions[bot]