Ysurac/openmptcprouter #1423: All links got a shot down/up

kevin39, 2020-11-28:
Hi,
## Current Behavior
All my links got a short down/up.
There are two 4G links and one DSL link. It seems impossible that everythings goes down at same time.

## Possible Solution
Don't know. Maybe be an update if a fix is already know :)

I know I'm not up to date but I saw many users with weird behavior so I decided to wait.
Is it a bug ? Is a fix already available ?

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.55.1 r0+14022-18b7d87a8f
  - OpenMPTCProuter VPS version: 0.1017
  - OpenMPTCProuter platform: RPI4

---

Sat Nov 28 20:18:57 2020 user.notice post-tracking-post-tracking: wan1 (wan1) switched off because ping from 192.168.1.99 error 
Sat Nov 28 20:27:34 2020 daemon.err /usr/bin/ss-redir[22167]: remote recv: Operation timed out
Sat Nov 28 20:27:49 2020 user.notice post-tracking-post-tracking: wan3 (wan3) switched off because gateway down and ping from 192.168.3.99 error 
Sat Nov 28 20:27:49 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.3.2 dev wan3
Sat Nov 28 20:27:49 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.3.2 dev wan3
Sat Nov 28 20:27:49 2020 user.notice post-tracking-post-tracking: wan2 (wan2) switched off because gateway down and ping from 192.168.8.99 error (80.67.169.12,80.67.169.40)
Sat Nov 28 20:27:49 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.8.1 dev wan2
Sat Nov 28 20:27:49 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.8.1 dev wan2
Sat Nov 28 20:28:27 2020 user.notice post-tracking-post-tracking: wan2 (wan2) switched up
Sat Nov 28 20:28:29 2020 user.notice post-tracking-post-tracking: wan3 (wan3) switched up

Ysurac, 2020-11-28:
Old release, I only support latest release.
If they are on same switch, check the switch.

kevin39, 2020-11-30:
Switch misconfigured, sorry :)
So, problem fixed.

closed 2020-11-30 by kevin39