Ysurac/openmptcprouter #1425: IpV6 configuration FirstHeberg

Xamamule, 2020-11-29:
## Expected Behavior
Operate IPv6 after activating the option

## Current Behavior
I have activated in "Common server settings" the Enable IPv6 option.
After that I get the message "No IPv6 access" in the "Network overview"
I have the impression that the problem comes from the vps hosted at "FirstHeberg"
How to know if the vps is correctly configured?

## Specifications

  - OpenMPTCProuter version: 0.56.5
  - OpenMPTCProuter VPS version: 0.1022 5.4.74-mtcp
  - OpenMPTCProuter platform: x86_64

[image]

Thanks for your help !

Maxime.

Ysurac, 2020-11-29:
The VPS need to have IPv6 configured.
Check on the VPS the result of ```ip -6 a``` and that you can ping an ipv6 address, like ```ping6 www.google.com```.

Xamamule, 2020-11-29:
I am unable to understand the result of the command

thanks for the help

~# ip -6 a
1: lo: <LOOPBACK,UP,LOWER_UP,80000> mtu 65536 state UNKNOWN qlen 1000
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UP qlen 1000
    inet6 2a03:75c0:39:e291::1/64 scope global
       valid_lft forever preferred_lft forever
    inet6 fe80::28a:56ff:fe04:52c/64 scope link
       valid_lft forever preferred_lft forever
4: gt-udp-tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 state UP qlen 100
    inet6 fe80::35c2:f589:1fc4:5570/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
5: dsvpn0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 9000 state UNKNOWN qlen 500
    inet6 64:ff9b::aff:fb01 peer 64:ff9b::aff:fb02/96 scope global
       valid_lft forever preferred_lft forever
    inet6 fe80::3949:e53:1fc4:8f33/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
6: mlvpn0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1444 state UNKNOWN qlen 500
    inet6 fe80::2503:cd01:de2a:a012/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
7: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 state UNKNOWN qlen 100
    inet6 fe80::ded7:3ade:85e7:863e/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
8: tun1: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 state UNKNOWN qlen 100
    inet6 fe80::d5e0:7439:ff09:56c8/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
13: gt-tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 state UNKNOWN qlen 500
    inet6 fe80::5d74:ffbe:ccdd:42b2/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
16: omr-6in4-user0@NONE: <POINTOPOINT,NOARP,UP,LOWER_UP,80000> mtu 1480 state UNKNOWN qlen 1000
    inet6 fe80::aff:fc01/64 scope link
       valid_lft forever preferred_lft forever
    inet6 fe80::a00:1/126 scope link
       valid_lft forever preferred_lft forever

----------------------------------------------
~# ping6 www.google.com
PING www.google.com(par10s39-in-x04.1e100.net (2a00:1450:4007:807::2004)) 56 data bytes
^C
--- www.google.com ping statistics ---
11 packets transmitted, 0 received, 100% packet loss, time 231ms

Ysurac, 2020-11-29:
You have a public IPv6: ```inet6 2a03:75c0:39:e291::1/64 scope global```
But if you can't ping there is a problem. Maybe a problem with route, what is the result of ```ip -6 r``` ?

Xamamule, 2020-11-29:
~# ip -6 r
::1 dev lo proto kernel metric 256 pref medium
64:ff9b::/96 dev dsvpn0 proto kernel metric 256 pref medium
2a03:75c0:39:e291::/64 dev eth0 proto kernel metric 256 pref medium
fdee:a92c:af41::/48 via fe80::a00:2 dev omr-6in4-user0 metric 1024 pref medium
fe80::a00:0/126 dev omr-6in4-user0 proto kernel metric 256 pref medium
fe80::/64 dev eth0 proto kernel metric 256 pref medium
fe80::/64 dev dsvpn0 proto kernel metric 256 pref medium
fe80::/64 dev mlvpn0 proto kernel metric 256 pref medium
fe80::/64 dev tun0 proto kernel metric 256 pref medium
fe80::/64 dev tun1 proto kernel metric 256 pref medium
fe80::/64 dev gt-udp-tun0 proto kernel metric 256 pref medium
fe80::/64 dev gt-tun0 proto kernel metric 256 pref medium
fe80::/64 dev omr-6in4-user0 proto kernel metric 256 pref medium
default via fe80::1 dev eth0 metric 1024 onlink pref medium

Ysurac, 2020-11-29:
Configuration seems to be ok.

Xamamule, 2020-11-29:
Ok make a ticket to firstheberg. I'm waiting.

Randdrick, 2020-11-30:
I am at firstheberg and I had the same problem. I reinstalled my VPS with a Debian 10, and now it works.

Xamamule, 2020-12-01:
J'ai déjà reinstallé mon VPS avec Debian 10 et le problème persiste. J'attend une réponse de firstheberg

Xamamule, 2020-12-02:
Hello everyone.
Goods news.
I don't know what has changed. The firstheberg team stepped in and now the following command is working from the VPS server.

ping6 ipv6.google.com
PING ipv6.google.com(par10s33-in-x0e.1e100.net (2a00:1450:4007:816::200e)) 56 data bytes
64 bytes from par10s33-in-x0e.1e100.net (2a00:1450:4007:816::200e): icmp_seq=1 ttl=119 time=4.21 ms
64 bytes from par10s33-in-x0e.1e100.net (2a00:1450:4007:816::200e): icmp_seq=2 ttl=119 time=4.06 ms
64 bytes from par10s33-in-x0e.1e100.net (2a00:1450:4007:816::200e): icmp_seq=3 ttl=119 time=4.18 ms
64 bytes from par10s33-in-x0e.1e100.net (2a00:1450:4007:816::200e): icmp_seq=4 ttl=119 time=4.04 ms
^C
--- ipv6.google.com ping statistics ---
4 packets transmitted, 4 received, 0% packet loss, time 7ms
rtt min/avg/max/mdev = 4.036/4.122/4.214/0.108 ms

[image]

Now that the ping v6 is working on the vps. Do I have to initialize something on my OpenMTCProuter server?

Thanks for your help !

Maxime.

Xamamule, 2020-12-02:
The result of this command has changed.

I can't tell if it's better or worse than before

ip -6 a
1: lo: <LOOPBACK,UP,LOWER_UP,80000> mtu 65536 state UNKNOWN qlen 1000
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UP qlen 1000
    inet6 2a03:75c0:39:e291::1/64 scope global
       valid_lft forever preferred_lft forever
    inet6 fe80::28a:56ff:fe04:52c/64 scope link
       valid_lft forever preferred_lft forever
3: dsvpn0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 9000 state UNKNOWN qlen 500
    inet6 64:ff9b::aff:fb01 peer 64:ff9b::aff:fb02/96 scope global
       valid_lft forever preferred_lft forever
    inet6 fe80::b428:f56e:4018:fd38/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
5: gt-udp-tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 state UP qlen 100
    inet6 fe80::6198:6016:45a4:31f1/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
8: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 state UNKNOWN qlen 100
    inet6 fe80::7b3a:2f2c:6ac9:258a/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
9: tun1: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 state UNKNOWN qlen 100
    inet6 fe80::5360:5d66:d2de:a4a0/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
10: mlvpn0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1444 state UNKNOWN qlen 500
    inet6 fe80::afc6:a850:53ca:3878/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
16: omr-6in4-user0@NONE: <POINTOPOINT,NOARP,UP,LOWER_UP,80000> mtu 1480 state UNKNOWN qlen 1000
    inet6 fe80::aff:ff01/64 scope link
       valid_lft forever preferred_lft forever
    inet6 fe80::a00:1/126 scope link
       valid_lft forever preferred_lft forever
18: gt-tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 state UNKNOWN qlen 500
    inet6 fe80::28ca:369d:d75e:f600/64 scope link stable-privacy
       valid_lft forever preferred_lft forever

Xamamule, 2020-12-03:
I found an error in the kernel.log.

[ 3522.785315] Dead loop on virtual device 6in4-omr6in4, fix it urgently!

Someone know how resolve it ?

Thanks

Xamamule, 2020-12-03:
Hi,

Firstheberg has changed the vlan configuration. Now I have no more problem to ping6 from the vps.

However, I still get the error message "No Ipv6 access"

I keep looking.

Xamamule, 2020-12-04:
Hi,

I re-installed the vps  and the OpenMPTCProuter server 
Now everything works.
Perfect !!

closed 2020-12-05 by Xamamule