Ysurac/openmptcprouter #1426: what happens if i give the OMR a different source

rifaterdemsahin, 2020-11-29:
instead of giving only one direct modem if i connect one line to another multiwan router.

OMR shows online on draytek but i am not sure. As now it has dsl and lte in a roundrobin version with sticky sessions

what is my down side ?

[image]

[image]

Referenced multi wan router
https://www.broadbandbuyer.com/products/30263-draytek-v2862n-k/?gclid=Cj0KCQiAqo3-BRDoARIsAE5vnaLFHhU5wfZbdvATnT7TbD40_OrV-NWvz0k6QlMt4J06pNEtnG_ThcoaAlqWEALw_wcB

Ysurac, 2020-11-30:
I'm not sure that load balancer will really work in this case, but you will tell me ;)

AndreaLeidi, 2020-11-30:
[image]

[image]

Works like a charm on my Vigor 2926.

rifaterdemsahin, 2020-12-11:
it does also work with my vigor as well.I think it is not healthy though right ?

AndreaLeidi, 2020-12-11:
I see no other way if you want to add a bonding device to a LAN built around a Draytek load balancing router but for any reason you can't remove the existing hardware or change the wiring. 
The only unhealty result is adding a little more latency, I guess.

closed 2021-03-17 by github-actions[bot]