Ysurac/openmptcprouter #1429: No VPS connectivity if master link is down

Maverick78de, 2020-12-03:
v0.56.5

If the master link is down, the whole vps connectivity is unavailable, see screenshot. The output switch to direct connection through LTE. I don't know if this is really wanted?

[image]

Ysurac, 2020-12-03:
Please use the template...
Check in State->System log why it's not using the working wan.

Maverick78de, 2020-12-03:
`Thu Dec  3 17:20:04 2020 daemon.info procd: - init complete -
Thu Dec  3 17:21:50 2020 daemon.err glorytun[11747]: read: Operation timed out
Thu Dec  3 17:21:50 2020 daemon.info glorytun[11747]: STOPPED tun0
Thu Dec  3 17:22:04 2020 daemon.err /usr/bin/ss-local[19032]: getpeername: Socket not connected
Thu Dec  3 17:22:06 2020 user.notice post-tracking-post-tracking: omrvpn down because gateway down
Thu Dec  3 17:22:06 2020 user.notice post-tracking-post-tracking: Tunnel down use ShadowSocks for UDP
Thu Dec  3 17:22:06 2020 user.notice Shadowsocks: Rules DOWN
Thu Dec  3 17:22:07 2020 user.notice Shadowsocks: Rules UP
Thu Dec  3 17:22:07 2020 daemon.err omr-tracker[11133]: Flush terminated
Thu Dec  3 17:22:07 2020 user.notice Shadowsocks: Reload omr-bypass rules
Thu Dec  3 17:22:07 2020 user.notice omr-bypass: Starting OMR-ByPass...
Thu Dec  3 17:22:10 2020 user.notice post-tracking-post-tracking: wan1 (VLAN24) switched off because gateway down and ping from 192.168.24.2 error (8.8.8.8,80.67.169.12,8.8.4.4)
Thu Dec  3 17:22:10 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.24.1 dev VLAN24
Thu Dec  3 17:22:19 2020 daemon.err /usr/bin/ss-local[19032]: getpeername: Socket not connected
Thu Dec  3 17:22:23 2020 user.notice omr-bypass: Reload dnsmasq...
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[25017]: exiting on receipt of SIGTERM
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[7333]: started, version 2.82 cachesize 150
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[7333]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq-dhcp[7333]: DHCP, IP range 192.168.33.2 -- 192.168.33.253, lease time 7d
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq-dhcp[7333]: DHCP, IP range 192.168.189.2 -- 192.168.189.253, lease time 12h
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq-dhcp[7333]: DHCP, IP range 192.168.42.2 -- 192.168.42.253, lease time 7d
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[7333]: using only locally-known addresses for domain test
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[7333]: using only locally-known addresses for domain onion
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[7333]: using only locally-known addresses for domain localhost
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[7333]: using only locally-known addresses for domain local
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[7333]: using only locally-known addresses for domain invalid
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[7333]: using only locally-known addresses for domain bind
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[7333]: using nameserver 192.168.88.1#53
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[7333]: using nameserver 192.168.24.1#53
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[7333]: using nameserver 192.168.24.1#53 for domain fritz.box
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[7333]: using only locally-known addresses for domain mosnet
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[7333]: read /etc/hosts - 4 addresses
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[7333]: read /tmp/hosts/dhcp.cfg01411c - 36 addresses
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq-dhcp[7333]: read /etc/ethers - 0 addresses
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[7333]: read /etc/hosts - 4 addresses
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[7333]: read /tmp/hosts/dhcp.cfg01411c - 36 addresses
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq-dhcp[7333]: read /etc/ethers - 0 addresses
Thu Dec  3 17:22:24 2020 user.notice omr-bypass: OMR-ByPass is running
Thu Dec  3 17:22:24 2020 daemon.err glorytun[11747]: getpeername: Socket not connected
Thu Dec  3 17:22:24 2020 daemon.notice netifd: Network device 'tun0' link is down
Thu Dec  3 17:22:24 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Thu Dec  3 17:22:24 2020 daemon.notice netifd: Interface 'omrvpn' is now down
Thu Dec  3 17:22:24 2020 daemon.info glorytun: starting glorytun vpn instance vpn
Thu Dec  3 17:22:24 2020 user.notice ModemManager: hotplug: remove network interface tun0: event processed
Thu Dec  3 17:22:24 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=tun0, subsystem=net
Thu Dec  3 17:22:24 2020 daemon.info glorytun[7429]: INITIALIZED tun0
Thu Dec  3 17:22:24 2020 daemon.notice netifd: Interface 'omrvpn' is disabled
Thu Dec  3 17:22:24 2020 daemon.notice netifd: Interface 'omrvpn' is enabled
Thu Dec  3 17:22:24 2020 daemon.notice netifd: Interface 'omrvpn' is disabled
Thu Dec  3 17:22:24 2020 daemon.notice netifd: Interface 'omrvpn' is enabled
Thu Dec  3 17:22:24 2020 daemon.notice netifd: Network device 'tun0' link is up
Thu Dec  3 17:22:24 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Thu Dec  3 17:22:24 2020 daemon.notice netifd: Interface 'omrvpn' is setting up now
Thu Dec  3 17:22:24 2020 daemon.notice netifd: Interface 'omrvpn' is now up
Thu Dec  3 17:22:24 2020 user.notice ModemManager: hotplug: add network interface tun0: event processed
Thu Dec  3 17:22:24 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tun0, subsystem=net
Thu Dec  3 17:22:24 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Dec  3 17:22:24 2020 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
[... cut ...]
Thu Dec  3 17:23:21 2020 daemon.info omr-tracker-ss: Shadowsocks is down (can't contact via http 212.27.48.10, 198.27.92.1, 151.101.129.164, 77.88.55.77, 1.1.1.1)
Thu Dec  3 17:23:22 2020 user.notice Shadowsocks: Rules DOWN
Thu Dec  3 17:23:25 2020 daemon.err /usr/bin/ss-redir[19034]: remote recv: Host is unreachable
Thu Dec  3 17:23:25 2020 daemon.err /usr/bin/ss-redir[19033]: remote recv: Host is unreachable
Thu Dec  3 17:23:28 2020 daemon.info omr-tracker-ss: Server (168.119.230.67) seems down, no answer to ping
`

then every 5sec 
` /usr/bin/ss-local[19032]: getpeername: Socket not connected`

Ysurac, 2020-12-03:
Please don't attache file but paste here.

Maverick78de, 2020-12-03:
Thu Dec  3 17:21:50 2020 daemon.err glorytun[11747]: read: Operation timed out
Thu Dec  3 17:21:50 2020 daemon.info glorytun[11747]: STOPPED tun0
Thu Dec  3 17:22:04 2020 daemon.err /usr/bin/ss-local[19032]: getpeername: Socket not connected
Thu Dec  3 17:22:06 2020 user.notice post-tracking-post-tracking: omrvpn down because gateway down
Thu Dec  3 17:22:06 2020 user.notice post-tracking-post-tracking: Tunnel down use ShadowSocks for UDP
Thu Dec  3 17:22:06 2020 user.notice Shadowsocks: Rules DOWN
Thu Dec  3 17:22:07 2020 user.notice Shadowsocks: Rules UP
Thu Dec  3 17:22:07 2020 daemon.err omr-tracker[11133]: Flush terminated
Thu Dec  3 17:22:07 2020 user.notice Shadowsocks: Reload omr-bypass rules
Thu Dec  3 17:22:07 2020 user.notice omr-bypass: Starting OMR-ByPass...
Thu Dec  3 17:22:10 2020 user.notice post-tracking-post-tracking: wan1 (VLAN24) switched off because gateway down and ping from 192.168.24.2 error (8.8.8.8,80.67.169.12,8.8.4.4)
Thu Dec  3 17:22:10 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.24.1 dev VLAN24
Thu Dec  3 17:22:19 2020 daemon.err /usr/bin/ss-local[19032]: getpeername: Socket not connected
Thu Dec  3 17:22:23 2020 user.notice omr-bypass: Reload dnsmasq...
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[25017]: exiting on receipt of SIGTERM
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[7333]: started, version 2.82 cachesize 150
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[7333]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq-dhcp[7333]: DHCP, IP range 192.168.33.2 -- 192.168.33.253, lease time 7d
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq-dhcp[7333]: DHCP, IP range 192.168.189.2 -- 192.168.189.253, lease time 12h
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq-dhcp[7333]: DHCP, IP range 192.168.42.2 -- 192.168.42.253, lease time 7d
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[7333]: using only locally-known addresses for domain test
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[7333]: using only locally-known addresses for domain onion
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[7333]: using only locally-known addresses for domain localhost
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[7333]: using only locally-known addresses for domain local
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[7333]: using only locally-known addresses for domain invalid
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[7333]: using only locally-known addresses for domain bind
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[7333]: using nameserver 192.168.88.1#53
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[7333]: using nameserver 192.168.24.1#53
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[7333]: using nameserver 192.168.24.1#53 for domain fritz.box
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[7333]: using only locally-known addresses for domain mosnet
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[7333]: read /etc/hosts - 4 addresses
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[7333]: read /tmp/hosts/dhcp.cfg01411c - 36 addresses
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq-dhcp[7333]: read /etc/ethers - 0 addresses
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[7333]: read /etc/hosts - 4 addresses
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq[7333]: read /tmp/hosts/dhcp.cfg01411c - 36 addresses
Thu Dec  3 17:22:24 2020 daemon.info dnsmasq-dhcp[7333]: read /etc/ethers - 0 addresses
Thu Dec  3 17:22:24 2020 user.notice omr-bypass: OMR-ByPass is running
Thu Dec  3 17:22:24 2020 daemon.err glorytun[11747]: getpeername: Socket not connected
Thu Dec  3 17:22:24 2020 daemon.notice netifd: Network device 'tun0' link is down
Thu Dec  3 17:22:24 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Thu Dec  3 17:22:24 2020 daemon.notice netifd: Interface 'omrvpn' is now down
Thu Dec  3 17:22:24 2020 daemon.info glorytun: starting glorytun vpn instance vpn
Thu Dec  3 17:22:24 2020 user.notice ModemManager: hotplug: remove network interface tun0: event processed
Thu Dec  3 17:22:24 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=tun0, subsystem=net
Thu Dec  3 17:22:24 2020 daemon.info glorytun[7429]: INITIALIZED tun0
Thu Dec  3 17:22:24 2020 daemon.notice netifd: Interface 'omrvpn' is disabled
Thu Dec  3 17:22:24 2020 daemon.notice netifd: Interface 'omrvpn' is enabled
Thu Dec  3 17:22:24 2020 daemon.notice netifd: Interface 'omrvpn' is disabled
Thu Dec  3 17:22:24 2020 daemon.notice netifd: Interface 'omrvpn' is enabled
Thu Dec  3 17:22:24 2020 daemon.notice netifd: Network device 'tun0' link is up
Thu Dec  3 17:22:24 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Thu Dec  3 17:22:24 2020 daemon.notice netifd: Interface 'omrvpn' is setting up now
Thu Dec  3 17:22:24 2020 daemon.notice netifd: Interface 'omrvpn' is now up
Thu Dec  3 17:22:24 2020 user.notice ModemManager: hotplug: add network interface tun0: event processed
Thu Dec  3 17:22:24 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tun0, subsystem=net
Thu Dec  3 17:22:24 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Thu Dec  3 17:22:24 2020 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Thu Dec  3 17:22:29 2020 daemon.info omr-tracker-ss: Shadowsocks is down (can't contact via http 212.27.48.10, 198.27.92.1, 151.101.129.164)
[... cut ...]
Thu Dec  3 17:23:21 2020 daemon.info omr-tracker-ss: Shadowsocks is down (can't contact via http 212.27.48.10, 198.27.92.1, 151.101.129.164, 77.88.55.77, 1.1.1.1)
Thu Dec  3 17:23:22 2020 user.notice Shadowsocks: Rules DOWN
Thu Dec  3 17:23:25 2020 daemon.err /usr/bin/ss-redir[19034]: remote recv: Host is unreachable
Thu Dec  3 17:23:25 2020 daemon.err /usr/bin/ss-redir[19033]: remote recv: Host is unreachable
Thu Dec  3 17:23:28 2020 daemon.info omr-tracker-ss: Server (168.119.230.67) seems down, no answer to ping

Ysurac, 2020-12-03:
It's a new install or an update ?

Maverick78de, 2020-12-03:
It's an update from 0.56.4

Ysurac, 2020-12-03:
What is the result of ```ip r``` on the router via SSH or Services->Terminal when master is down ?

Maverick78de, 2020-12-03:
It is working now. I don't know but the order of interfaces in /etc/config/network was changes. WAN2 was above WAN1 (Master). I changed the order back to WAN1 and then WAN2, now it is working again.

Malaga82, 2020-12-04:
maybe this could help:
[image]
i setted this up, if master went down, the wan2 will be the master. as soon as wan1 will come up it will return master.

Maverick78de, 2020-12-05:
Thx I didn't recocnize this option

closed 2020-12-05 by Maverick78de