Ysurac/openmptcprouter #1430: Dashboard ok but speed still the same as ADSL

ijaouani, 2020-12-03:
First of all thanks a lot for the great work on this repo.

## Expected Behavior
I start to work on this project since one week and I finally have a status page (the dashboard) with everything green. I expect to have better speedtest with aggregation now.

## Current Behavior
The speedtest show me exactly the same value as ADSL.

## My setup

[image]

VPS : Debian hosted on Azure France Central Standard B1ms (1 vcpus, 2 GiB memory) 
OpenMPTCProuter installed on Synology NAS via Virtual Manager.
4G BOX : MR600
ADSL: Orange BOX (Sosh)
DHCP disabled on both 4G box and ADSL box.

## Connexions

NAS connected with LAN to orange box
4G Box connected with LAN to orange box
PC connected with LAN to orange box

## Specifications

  - OpenMPTCProuter version: v0.56.5 r0+14653-a439f1bb47
  - OpenMPTCProuter VPS version: 5.4.74-mptcp 0.1022
  - OpenMPTCProuter platform: openmptcprouter-v0.56.5-r0+14653-a439f1bb47-x86-64-generic-ext4-combined-efi.ova

Any ideas ?

Ysurac, 2020-12-03:
Check router and VPS cpu usage when you download a file.
Check also in Network->MPTCP, Bandwidth tab what you have when you download file.

ijaouani, 2020-12-03:
CPU usage is ok on both side.
But yes there is a problem because the bandwith tab is really really low (just few kbits/s) when I download a file from my PC. It feels like the download go through adsl directly and not through openMPTCP router. However, my ip is correct on my PC : 

[image]

Ysurac, 2020-12-03:
You have an IPv6 configured, I think this IP come from the ADSL, so any website that support IPv6 use the box directly.
You have to disable IPv6 on  the box if you want aggregation.

ijaouani, 2020-12-03:
Disabling ipv6 works ! I mean i am sure i am going through openMPTCP router now.
Bandwidth ADSL : 14 Mbits download - 1 Mbits upload
Bandwith 4G: 20 Mbits download/upload

Result : 5Mbits download and upload.

Maybe the CPU is the problem here :

[image]

Ysurac, 2020-12-03:
No, at least on the router no problem.

Ysurac, 2020-12-03:
As the router is on a VM, check also CPU usage on the host.

ijaouani, 2020-12-04:
After a new clean install everything seems to be fine on the setup side and the speedtest is as expected.
I close this issue.
Thanks for the nice tip about ipv6.

closed 2020-12-04 by ijaouani