Ysurac/openmptcprouter #1431: Launch a call on Microsoft Teams breaks the network

ijaouani, 2020-12-04:
## Expected Behavior
To be able able to perform call on Microsoft Teams.

## Current Behavior
Everytime I launch a call (even with the test bot) the network stop to work :

[image]

When I stop the call after a few seconds, all is green again.

## Specifications

- OpenMPTCProuter version: v0.56.5 r0+14653-a439f1bb47
- OpenMPTCProuter VPS version: 5.4.74-mptcp 0.1022
- OpenMPTCProuter platform: openmptcprouter-v0.56.5-r0+14653-a439f1bb47-x86-64-generic-ext4-combined-efi.ova

Ysurac, 2020-12-04:
Check VPS cpu load and VPS access, it seems to not answer to ping. What is the provider ?

ijaouani, 2020-12-04:
[image]

VPS : Debian hosted on Azure France Central Standard B1ms (1 vcpus, 2 GiB memory)
OpenMPTCProuter installed on Synology NAS via Virtual Manager.
4G BOX : MR600
ADSL: Orange BOX (Sosh)
DHCP disabled on both 4G box and ADSL box.

Connexions
NAS connected with LAN to orange box
4G Box connected with LAN to orange box
PC connected with LAN to orange box

ijaouani, 2020-12-04:
The thing I don't get is it's really specific to this app. When I download a big file everything work as expected.
I also try from an other PC to be sure same behavior.

This is the logs :

Fri Dec  4 10:52:21 2020 daemon.info dnsmasq-dhcp[10810]: DHCPREQUEST(eth0) 192.168.100.106 fc:f8:ae:60:c5:ed
Fri Dec  4 10:52:21 2020 daemon.info dnsmasq-dhcp[10810]: DHCPACK(eth0) 192.168.100.106 fc:f8:ae:60:c5:ed DESKTOP-I2BFNTI
Fri Dec  4 10:52:22 2020 daemon.info dnsmasq-dhcp[10810]: DHCPREQUEST(eth0) 192.168.100.106 fc:f8:ae:60:c5:ed
Fri Dec  4 10:52:22 2020 daemon.info dnsmasq-dhcp[10810]: DHCPACK(eth0) 192.168.100.106 fc:f8:ae:60:c5:ed DESKTOP-I2BFNTI
Fri Dec  4 10:52:34 2020 daemon.info dnsmasq-dhcp[10810]: DHCPREQUEST(eth0) 192.168.100.106 fc:f8:ae:60:c5:ed
Fri Dec  4 10:52:34 2020 daemon.info dnsmasq-dhcp[10810]: DHCPACK(eth0) 192.168.100.106 fc:f8:ae:60:c5:ed DESKTOP-I2BFNTI
Fri Dec  4 10:52:35 2020 daemon.warn odhcpd[4388]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Fri Dec  4 10:52:35 2020 daemon.err odhcpd[4388]: Failed to send to ff02::1%lan@eth0 (Permission denied)
Fri Dec  4 10:52:35 2020 daemon.info dnsmasq-dhcp[10810]: DHCPREQUEST(eth0) 192.168.100.106 fc:f8:ae:60:c5:ed
Fri Dec  4 10:52:35 2020 daemon.info dnsmasq-dhcp[10810]: DHCPACK(eth0) 192.168.100.106 fc:f8:ae:60:c5:ed DESKTOP-I2BFNTI
Fri Dec  4 10:52:48 2020 daemon.info dnsmasq-dhcp[10810]: DHCPREQUEST(eth0) 192.168.100.106 fc:f8:ae:60:c5:ed
Fri Dec  4 10:52:48 2020 daemon.info dnsmasq-dhcp[10810]: DHCPACK(eth0) 192.168.100.106 fc:f8:ae:60:c5:ed DESKTOP-I2BFNTI
Fri Dec  4 10:52:51 2020 daemon.warn odhcpd[4388]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Fri Dec  4 10:52:51 2020 daemon.err odhcpd[4388]: Failed to send to ff02::1%lan@eth0 (Permission denied)
Fri Dec  4 10:52:51 2020 daemon.info dnsmasq-dhcp[10810]: DHCPREQUEST(eth0) 192.168.100.106 fc:f8:ae:60:c5:ed
Fri Dec  4 10:52:51 2020 daemon.info dnsmasq-dhcp[10810]: DHCPACK(eth0) 192.168.100.106 fc:f8:ae:60:c5:ed DESKTOP-I2BFNTI
Fri Dec  4 10:53:07 2020 daemon.warn odhcpd[4388]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Fri Dec  4 10:53:07 2020 daemon.err odhcpd[4388]: Failed to send to ff02::1%lan@eth0 (Permission denied)
Fri Dec  4 10:53:08 2020 daemon.info dnsmasq-dhcp[10810]: DHCPREQUEST(eth0) 192.168.100.106 fc:f8:ae:60:c5:ed
Fri Dec  4 10:53:08 2020 daemon.info dnsmasq-dhcp[10810]: DHCPACK(eth0) 192.168.100.106 fc:f8:ae:60:c5:ed DESKTOP-I2BFNTI
Fri Dec  4 10:53:10 2020 daemon.info dnsmasq-dhcp[10810]: DHCPREQUEST(eth0) 192.168.100.106 fc:f8:ae:60:c5:ed
Fri Dec  4 10:53:10 2020 daemon.info dnsmasq-dhcp[10810]: DHCPACK(eth0) 192.168.100.106 fc:f8:ae:60:c5:ed DESKTOP-I2BFNTI
Fri Dec  4 10:53:20 2020 daemon.info dnsmasq-dhcp[10810]: DHCPREQUEST(eth0) 192.168.100.106 fc:f8:ae:60:c5:ed
Fri Dec  4 10:53:20 2020 daemon.info dnsmasq-dhcp[10810]: DHCPACK(eth0) 192.168.100.106 fc:f8:ae:60:c5:ed DESKTOP-I2BFNTI
Fri Dec  4 10:53:22 2020 daemon.info dnsmasq-dhcp[10810]: DHCPREQUEST(eth0) 192.168.100.106 fc:f8:ae:60:c5:ed
Fri Dec  4 10:53:22 2020 daemon.info dnsmasq-dhcp[10810]: DHCPACK(eth0) 192.168.100.106 fc:f8:ae:60:c5:ed DESKTOP-I2BFNTI
Fri Dec  4 10:53:23 2020 daemon.warn odhcpd[4388]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Fri Dec  4 10:53:23 2020 daemon.err odhcpd[4388]: Failed to send to ff02::1%lan@eth0 (Permission denied)
Fri Dec  4 10:53:34 2020 daemon.info dnsmasq-dhcp[10810]: DHCPREQUEST(eth0) 192.168.100.106 fc:f8:ae:60:c5:ed
Fri Dec  4 10:53:34 2020 daemon.info dnsmasq-dhcp[10810]: DHCPACK(eth0) 192.168.100.106 fc:f8:ae:60:c5:ed DESKTOP-I2BFNTI
Fri Dec  4 10:53:37 2020 daemon.info dnsmasq-dhcp[10810]: DHCPREQUEST(eth0) 192.168.100.106 fc:f8:ae:60:c5:ed
Fri Dec  4 10:53:37 2020 daemon.info dnsmasq-dhcp[10810]: DHCPACK(eth0) 192.168.100.106 fc:f8:ae:60:c5:ed DESKTOP-I2BFNTI
Fri Dec  4 10:53:39 2020 daemon.warn odhcpd[4388]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Fri Dec  4 10:53:39 2020 daemon.err odhcpd[4388]: Failed to send to ff02::1%lan@eth0 (Permission denied)
Fri Dec  4 10:53:45 2020 daemon.info dnsmasq-dhcp[10810]: DHCPREQUEST(eth0) 192.168.100.106 fc:f8:ae:60:c5:ed
Fri Dec  4 10:53:45 2020 daemon.info dnsmasq-dhcp[10810]: DHCPACK(eth0) 192.168.100.106 fc:f8:ae:60:c5:ed DESKTOP-I2BFNTI
Fri Dec  4 10:53:49 2020 daemon.info dnsmasq-dhcp[10810]: DHCPREQUEST(eth0) 192.168.100.106 fc:f8:ae:60:c5:ed
Fri Dec  4 10:53:49 2020 daemon.info dnsmasq-dhcp[10810]: DHCPACK(eth0) 192.168.100.106 fc:f8:ae:60:c5:ed DESKTOP-I2BFNTI
Fri Dec  4 10:53:51 2020 daemon.info dnsmasq-dhcp[10810]: DHCPREQUEST(eth0) 192.168.100.106 fc:f8:ae:60:c5:ed
Fri Dec  4 10:53:51 2020 daemon.info dnsmasq-dhcp[10810]: DHCPACK(eth0) 192.168.100.106 fc:f8:ae:60:c5:ed DESKTOP-I2BFNTI
Fri Dec  4 10:53:55 2020 daemon.warn odhcpd[4388]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Fri Dec  4 10:53:55 2020 daemon.err odhcpd[4388]: Failed to send to ff02::1%lan@eth0 (Permission denied)
Fri Dec  4 10:53:58 2020 daemon.err /usr/bin/ss-redir[12258]: remote recv: Operation timed out
Fri Dec  4 10:54:00 2020 daemon.err /usr/bin/ss-redir[12258]: remote recv: Operation timed out
[... cut ...]
Fri Dec  4 10:54:54 2020 daemon.info dnsmasq-dhcp[10810]: DHCPACK(eth0) 192.168.100.106 fc:f8:ae:60:c5:ed DESKTOP-I2BFNTI
Fri Dec  4 10:54:56 2020 daemon.info dnsmasq-dhcp[10810]: DHCPREQUEST(eth0) 192.168.100.106 fc:f8:ae:60:c5:ed
Fri Dec  4 10:54:56 2020 daemon.info dnsmasq-dhcp[10810]: DHCPACK(eth0) 192.168.100.106 fc:f8:ae:60:c5:ed DESKTOP-I2BFNTI
Fri Dec  4 10:54:59 2020 daemon.warn odhcpd[4388]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Fri Dec  4 10:54:59 2020 daemon.err odhcpd[4388]: Failed to send to ff02::1%lan@eth0 (Permission denied)

Ysurac, 2020-12-04:
I think Teams use UDP, it's why it's not the same problem when you download a file.
You can try to use enable SQM in the wizard and set download/upload speed.
It's still strange that the VPS doesn't answer to ping from the router...

ijaouani, 2020-12-04:
I think I have an idea :

[image]

[image]

It seems that the root cause is the VPS. When I launch MS Teams seems the CPU reach the top and the vps is unreachable (even with SSH). But when I download a file only few percent. Maybe Standard B1ms (1 vcpus, 2 GiB memory) is not enough ?

Ysurac, 2020-12-04:
I always had strange issues with Azure ;)
1 VCPU doesn't really say something, some provider share one core with 10 users (it's slow), some with 2 users (it's faster).

ijaouani, 2020-12-04:
I have upgraded the size of the VM to Standard D2s v3 (2 vcpus, 8 GiB memory) and exactly the same problem. 
When I launch a call I have exacty the image I sent in my first message and as soon as I hang up :
Sometine few seconds later everything is green. 
Sometime the vps is stuck for a while :

[image]

I also try to enable SQM for ADSL as you suggested but does not help for what I can see.

Where can I find some logs for the vps side ?

Ysurac, 2020-12-04:
In ```/var/log/daemon.log``` there is some log.
You have really high load average...

ijaouani, 2020-12-04:
I finally find a workaround. I created a new VPS (still on Azure) but with Ubuntu 20-04 instead of the recommanded debian and the issue just disappeared. I will keep the issue open until I am sure of it and then I will take care of closing it.

Thank you really much Ysurac for your help on this. Really appreciate.

closed 2020-12-09 by ijaouani