Ysurac/openmptcprouter #1432: Bloquer web à un appareil du réseau

JeedHome44, 2020-12-04:
## Expected Behavior
Comment bloquer internet à un appareil sur le réseau LAN? 

## Current Behavior

## Specifications

  - OpenMPTCProuter version: 55.6
  - OpenMPTCProuter VPS version: 22
  - OpenMPTCProuter platform: Pi 4B 2Go

Ysurac, 2020-12-04:
En lisant la doc : https://github.com/Ysurac/openmptcprouter/wiki/Block-client-Internet-access

JeedHome44, 2020-12-04:
Merci.
En fait, je ne veux pas que l'appareil soit mis à jour automatiquement.
C'est donc ça pour chacun?

JeedHome44, 2020-12-04:
Mais si je veux utiliser l'appareil qu'en local, je n'ai que la 2eme règle à faire non?

Ysurac, 2020-12-04:
Non les 2 règles. Après tu peux ne pas définir la passerelle ou une autre passerelle sur chaque machine via DHCP ou en statique. Pas de route par défaut par le routeur, pas d'accès à Internet.

closed 2021-03-10 by github-actions[bot]