Ysurac/openmptcprouter #1433: OMR VPS can use multiple CPU cores?

ndaptardar, 2020-12-05:
I was thinking if VPS is having 2 dedicated cores (instead of 1 shared core), will OMR provide faster data transfer?

Malaga82, 2020-12-05:
maybe.
if you have aes available on both router and vps and gcm as encryption it should use multithread and don't bottleneck.
but if you have a good cpu on your vps and didn't spike, it's not a problem.

user747, 2020-12-05:
just to add an example: 

vps script checks to see how many cores available
```
NBCPU=${NBCPU:-$(grep -c '^processor' /proc/cpuinfo | tr -d "\n")}
```
if more than 1, configure shadowsocks accordingly 
```
if [ "$NBCPU" -gt "1" ]; then
		for i in $(seq 2 NBCPU); do
			sed -i '0,/65101/ s/        "65101.*/&\n&/' /etc/shadowsocks-libev/manager.json
```

ndaptardar, 2020-12-05:
> just to add an example:
>
> vps script checks to see how many cores available
>
> ```
> NBCPU=${NBCPU:-$(grep -c '^processor' /proc/cpuinfo | tr -d "\n")}
> ```
>
> if more than 1, configure shadowsocks [...]

Great, so there is merit in using multi core. Thanks.

rifaterdemsahin, 2020-12-11:
i am using on a multicore device.
it is lighting fast

closed 2021-03-17 by github-actions[bot]