Ysurac/openmptcprouter #1434: Extend the aggregate network with Wan router

Xamamule, 2020-12-05:
## Expected Behavior
Extend the aggregate network created by OpenMPTCProuter

## Current Behavior
The speed behind the Asus router 1 is lower than the speed behind the 4g router.
I think my configuration is not correct.
On OpenMPTCProuter I have deactivated DCHP because I have to activate it on the Asus router to use AiMesh technology.

## Specifications
  - OpenMPTCProuter version: 0.56.5
  - OpenMPTCProuter VPS version: 0.1022 5.4.74
  - OpenMPTCProuter platform: x86_64
  - IPV6 Enabled

[image]

Can you give me some advice to improve the configuration?

Thanks

Randdrick, 2020-12-06:
Peharps it's beacause you have  an auto adaptive QOS activate on your Asus Router...

(Peut-être c'est parce que tu as du QOS "auto adaptatif" activé sur ton Routeur ASUS). Le QOS peut, entre autres choses, réguler ta bande passante pour réduire ta latence par exemple.

Xamamule, 2020-12-06:
Hello Randdrick

No the service auto adaptive QOS is not activate on my Asus Router.
I use an old Zotac Machine for OpenMPTCProuter  is it possible the Ethernet limit of thie machine is 100Mbits. 
How can i check this ?

Xamamule, 2020-12-06:
Result of the command ethtool eth0

The speed seems to be 1Gb.
So the limit is not here

[image]

Randdrick, 2020-12-06:
Use this tool : 

https://www.fr.paessler.com/performance-monitoring?msclkid=95f1aa8d814812c41d9bf5414365a3dd&utm_source=bing&utm_medium=cpc&utm_campaign=FRA_FR_Search-nonBrand_broad_2&utm_term=%2Bperformances%20%2Br%C3%A9seau&utm_content=performances-r%C3%A9seau

It's free for 30 days and may be it's help you to solve your problem

Xamamule, 2020-12-07:
Hello

I solve my problem. 
On the asus router, i changed the operation Mode from router mode to Access Point Ap mode.
DHCP is automatically disabled.
So I enabled DCHP on OpenMPTCProuter.

With this configuration I can continue to use aimesh between me two Asus routers
Ethernet and wifi are perfectly stable.

For the speed problem I also found.
The 4G server does not aggregate the different 700,1800 and 2600 Mhz ranges all the time.

I found a solution to force aggregation using [LTE H-Monitor](https://www.ltehmonitor.com/fr/) on a Synology
So the 4G router always uses the 4G + connection.

Thanks to Ysurac for the job and Randdrick for his help.

closed 2020-12-07 by Xamamule