Ysurac/openmptcprouter #1435: fresh update failed

numerobis08, 2020-12-06:
## Expected Behavior
factory udpate

## Current Behavior
i made a fresh install on VPS and router and since that it doesn't work anymore ( since i re installed VPS i didn't forget to use the new generated KEY but it doesn't seems to work. 
i have the following message on the status page : 

[image]

## Specifications

  - OpenMPTCProuter version: v0.56.5
  - OpenMPTCProuter VPS version: 0.1022
  - OpenMPTCProuter platform: (RPI4

numerobis08, 2020-12-06:
update : 
it seems to be working better ( at least i do have now an internet connection ) but we the update it seems to have new parameters that i don't know how to handle. Here is below the new status i have : 

[image]

what is strange is that the SFR ADSL Box seems to have same WAN IP than the 4G connection. 
Here is below how i configured the 4G router and the SFR ADSL Box , have I done something wrong ?

[image]

[image]

[image]

Ysurac, 2020-12-06:
wan1 and wan2 use same physical interface, in this case you must use Type MacVLAN.

numerobis08, 2020-12-06:
> wan1 and wan2 use same physical interface, in this case you must use Type MacVLAN.

Hello Ysurac , 
The Rpi4 is connected to a switch on which both 4G router and SFR ADSL BOX are connected that’s why I thought I had to choose Eth0. 
For my understanding in which case should we use ETh0 ? What is MacVLAN ? 
Thanks 😊

numerobis08, 2020-12-06:
> wan1 and wan2 use same physical interface, in this case you must use Type MacVLAN.

Hello i just did has you mentioned , and i works better (i read on GitHub the multipath should be set as master for the ADSL box as bitrate is more stable ( and set to "on" on the 4G router) , is that correct ? ) . just a question is it normal that time to times i have the following orange message regarding the VPS : 

[image]

because if i wait , it comes back to green as below (i'm using Firstheberg) : 

[image]

annoter topic is it any issue if it mentions " votre IP n'a pas été attribuée par ce routeur" ? 

Merci d'avance

Ysurac, 2020-12-06:
As it's indicated on the wizard, if you share same physical interface with multiples WAN, use MacVLAN. It's the default setting on a RPI.

Ysurac, 2020-12-06:
For the message,this can happen if this take too much time. I have many issues with Firstheberg...

numerobis08, 2020-12-06:
> For the message,this can happen if this take too much time. I have many issues with Firstheberg...

I think I’ll switch next month to Milkywan as I read good feedback.  
For multiparty the “master” should be on the adsl and only “on” on the 4G as it’s less stable ?

Ysurac, 2020-12-06:
true, master should be the more stable connection.

numerobis08, 2020-12-06:
Thanks a lot for the help :-) 
I think I can close that issue. 
Thank you

closed 2020-12-06 by numerobis08