Ysurac/openmptcprouter #1436: jffsrestore command does not reset the system back to vanilla install

sigkill, 2020-12-06:
Ran into a strange issue tonight with regards to how to reset openmptcprouter back to vanilla install on a Raspberry Pi 3B.

Following a guide for Openwrt, I was attempting to reset the OS back to what it was at when installed, however, it appear the support for this may not be present as of yet.

Following the guide here: https://openwrt.org/docs/guide-user/troubleshooting/failsafe_and_factory_reset

I realize I can take the sdcard out and reimage it, but I believe it would be ideal to have the firmware reset work via the ssh method too, as the pi was configured without a monitor when I ran into trouble.

It might be ideal to map a gpio switch to reset the pi too.

Anyhow, I hope this helps, I'll gather more info if needed.

Thanks!

MagmaiKH, 2020-12-08:
I just tried to do this and can confirm it did not reset the system.  
If you run with /dev/loop0 mounted it claims "/dev/loop0 is mounted as /overlay, only erasing files" but it will not erase .* "hidden" files (e.g. .ssh) .
If you umount /overlay then it claims" /dev/loop0 is not mounted. /dev/loop0 will be erased on next mount" but it isn't erased next mount.

I am using an APU4 (so x86_64).

```
/dev/root on /rom type squashfs (ro,relatime)
/dev/loop0 on /overlay type f2fs (rw,lazytime,noatime,background_gc=on,no_heap,user_xattr,inline_xattr,inline_data,inline_dentry,flush_merge,extent_cache,mode=adaptive,active_logs=6)
overlayfs:/overlay on / type overlay (rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work)
```

pylaterreur, 2026-10-07:
I checked this on a current image: an x86_64 squashfs build of develop (OpenWrt 328acd4e, fstools 2aa16407, f2fs overlay on `/dev/loop0`), in QEMU.

`firstboot -y` (which runs `factoryreset`) followed by a reboot does reset the configuration. The hostname I had set is back to the default, and a file in `/root` and a hidden file in `/etc` are gone. What survives is everything inside a hidden *directory*: `/root/.hidden-dir/key`, `/root/.hidden-dir/sub/key` and `/etc/visible-dir/.hidden-sub/file` are still there after the reboot. That's what @MagmaiKH saw with `.ssh`.

The cause is in OpenWrt's fstools. The overlay is wiped with `glob("<dir>/*")`, and `*` skips names that start with a dot. Both passes use it: the one `factoryreset` runs while the overlay is mounted, and the one `mount_root` runs at the next boot. A fix is proposed upstream in https://github.com/openwrt/fstools/pull/22. With that fix in `libfstools.so`, the same test leaves nothing behind.

Until then, `rm -rf /root/.ssh` (or whatever hidden directories you created) before the reset does the job. Reflashing the image works too.

I didn't test the second case from 2020 (unmounting `/overlay` first, then "will be erased on next mount"), since the overlay can't be unmounted on a running system.