Ysurac/openmptcprouter #1438: Access the NAS with port forwarding

Xamamule, 2020-12-07:
## Expected Behavior
Access the synology NAS by the IP of the VPS server

## Current Behavior
Cannot access the NAS with port forwarding

## Specifications

  - OpenMPTCProuter version: 0.56.5
  - OpenMPTCProuter VPS version: 0.1022 5.4.74-mptcp
  - OpenMPTCProuter platform: x86_64

I followed the tutorial "Port forwarding"
Here is my configuration

I open the port 5000 in the firevall.
19.168.100.1 is the ip of OpenMPTCProuter.
[image]

Then i set a SNAT rule 
[image]

The Vpn is OK in System->OpenMPTCProuter

rules seem OK on vps

#       Don't allow connection pickup from the net
#
Invalid(DROP)   net             all             tcp
#
#       Accept DNS connections from the firewall to the network
#
DNS(ACCEPT)     $FW             net
#
#       Allow Ping from/to the VPN
#
Ping(ACCEPT)    vpn             $FW
Ping(ACCEPT)    $FW             vpn
#
#       Allow Ping from the firewall to the network
#
Ping(ACCEPT)    $FW             net
#
# Drop Ping from the "bad" net zone.. and prevent your log from being flooded..
#
#Ping(DROP)     net             $FW
Ping(ACCEPT)    net             $FW
#
#	Accept connection from port > 65000 for shadowsocks and glorytun on the firewall
#
ACCEPT          net             $FW             tcp     65000-65535
ACCEPT          net             $FW             udp     65000-65535
#
#	Accept connection from SSH to the firewall
#
ACCEPT          net             $FW             tcp     65222
#
#	DHCP forward to the VPN from the firewall
#
DHCPfwd(ACCEPT)	$FW		vpn
#
#	Redirect all port from 1 to 64999 to the VPN client from the network
#
#DNAT		net		vpn:$OMR_ADDR	tcp	1-64999
#DNAT		net		vpn:$OMR_ADDR	udp	1-64999
ACCEPT		net		$FW		tcp	65101	# OMR openmptcprouter open shadowsocks port tcp
ACCEPT		net		$FW		udp	65101	# OMR openmptcprouter open shadowsocks port udp
ACCEPT		net		$FW		tcp	5000	# OMR openmptcprouter open router 5000 port tcp
DNAT		net		vpn:$OMR_ADDR	tcp	5000	# OMR openmptcprouter redirect router 5000 port tcp

params.vpn on VPS is ok too
VPS_ADDR=10.255.255.1
VPS_IFACE=gt-tun0
OMR_ADDR=10.255.255.2

and if i check on the router, if packets are coming on port 5000, i see this 

root@OpenMPTCProuter:~# tcpdump -i tun0 port 5000
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on tun0, link-type RAW (Raw IP), capture size 262144 bytes
21:32:33.361438 IP nor75-22_migr-88-162-121-225.fbx.proxad.net.54459 > 10.255.255.2.5000: Flags [S], seq 566676191, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
21:32:33.361621 IP 10.255.255.2.5000 > nor75-22_migr-88-162-121-225.fbx.proxad.net.54459: Flags [R.], seq 0, ack 566676192, win 0, length 0
21:32:33.362708 IP nor75-22_migr-88-162-121-225.fbx.proxad.net.54460 > 10.255.255.2.5000: Flags [S], seq 2087284841, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
21:32:33.362829 IP 10.255.255.2.5000 > nor75-22_migr-88-162-121-225.fbx.proxad.net.54460: Flags [R.], seq 0, ack 2087284842, win 0, length 0
21:32:33.624479 IP nor75-22_migr-88-162-121-225.fbx.proxad.net.54461 > 10.255.255.2.5000: Flags [S], seq 2880406045, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
21:32:33.624658 IP 10.255.255.2.5000 > nor75-22_migr-88-162-121-225.fbx.proxad.net.54461: Flags [R.], seq 0, ack 2880406046, win 0, length 0
21:32:33.920503 IP nor75-22_migr-88-162-121-225.fbx.proxad.net.54459 > 10.255.255.2.5000: Flags [S], seq 566676191, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
21:32:33.920684 IP 10.255.255.2.5000 > nor75-22_migr-88-162-121-225.fbx.proxad.net.54459: Flags [R.], seq 0, ack 1, win 0, length 0
21:32:33.921536 IP nor75-22_migr-88-162-121-225.fbx.proxad.net.54460 > 10.255.255.2.5000: Flags [S], seq 2087284841, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
21:32:33.921716 IP 10.255.255.2.5000 > nor75-22_migr-88-162-121-225.fbx.proxad.net.54460: Flags [R.], seq 0, ack 1, win 0, length 0
21:32:34.183719 IP nor75-22_migr-88-162-121-225.fbx.proxad.net.54461 > 10.255.255.2.5000: Flags [S], seq 2880406045, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
21:32:34.183902 IP 10.255.255.2.5000 > nor75-22_migr-88-162-121-225.fbx.proxad.net.54461: Flags [R.], seq 0, ack 1, win 0, length 0
21:32:34.480537 IP nor75-22_migr-88-162-121-225.fbx.proxad.net.54459 > 10.255.255.2.5000: Flags [S], seq 566676191, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
21:32:34.480714 IP 10.255.255.2.5000 > nor75-22_migr-88-162-121-225.fbx.proxad.net.54459: Flags [R.], seq 0, ack 1, win 0, length 0
21:32:34.480941 IP nor75-22_migr-88-162-121-225.fbx.proxad.net.54460 > 10.255.255.2.5000: Flags [S], seq 2087284841, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
21:32:34.481097 IP 10.255.255.2.5000 > nor75-22_migr-88-162-121-225.fbx.proxad.net.54460: Flags [R.], seq 0, ack 1, win 0, length 0
21:32:34.742274 IP nor75-22_migr-88-162-121-225.fbx.proxad.net.54461 > 10.255.255.2.5000: Flags [S], seq 2880406045, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
21:32:34.742447 IP 10.255.255.2.5000 > nor75-22_migr-88-162-121-225.fbx.proxad.net.54461: Flags [R.], seq 0, ack 1, win 0, length 0
21:32:35.038832 IP nor75-22_migr-88-162-121-225.fbx.proxad.net.54459 > 10.255.255.2.5000: Flags [S], seq 566676191, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
21:32:35.039053 IP 10.255.255.2.5000 > nor75-22_migr-88-162-121-225.fbx.proxad.net.54459: Flags [R.], seq 0, ack 1, win 0, length 0
21:32:35.039719 IP nor75-22_migr-88-162-121-225.fbx.proxad.net.54460 > 10.255.255.2.5000: Flags [S], seq 2087284841, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
21:32:35.039900 IP 10.255.255.2.5000 > nor75-22_migr-88-162-121-225.fbx.proxad.net.54460: Flags [R.], seq 0, ack 1, win 0, length 0
21:32:35.303341 IP nor75-22_migr-88-162-121-225.fbx.proxad.net.54461 > 10.255.255.2.5000: Flags [S], seq 2880406045, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
21:32:35.303524 IP 10.255.255.2.5000 > nor75-22_migr-88-162-121-225.fbx.proxad.net.54461: Flags [R.], seq 0, ack 1, win 0, length 0
21:32:35.600369 IP nor75-22_migr-88-162-121-225.fbx.proxad.net.54459 > 10.255.255.2.5000: Flags [S], seq 566676191, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
21:32:35.600544 IP 10.255.255.2.5000 > nor75-22_migr-88-162-121-225.fbx.proxad.net.54459: Flags [R.], seq 0, ack 1, win 0, length 0
21:32:35.601121 IP nor75-22_migr-88-162-121-225.fbx.proxad.net.54460 > 10.255.255.2.5000: Flags [S], seq 2087284841, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
21:32:35.601282 IP 10.255.255.2.5000 > nor75-22_migr-88-162-121-225.fbx.proxad.net.54460: Flags [R.], seq 0, ack 1, win 0, length 0
21:32:35.767876 IP nor75-22_migr-88-162-121-225.fbx.proxad.net.54462 > 10.255.255.2.5000: Flags [S], seq 933499842, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
21:32:35.768070 IP 10.255.255.2.5000 > nor75-22_migr-88-162-121-225.fbx.proxad.net.54462: Flags [R.], seq 0, ack 933499843, win 0, length 0
21:32:35.862404 IP nor75-22_migr-88-162-121-225.fbx.proxad.net.54461 > 10.255.255.2.5000: Flags [S], seq 2880406045, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
21:32:35.862590 IP 10.255.255.2.5000 > nor75-22_migr-88-162-121-225.fbx.proxad.net.54461: Flags [R.], seq 0, ack 1, win 0, length 0
21:32:36.315006 IP nor75-22_migr-88-162-121-225.fbx.proxad.net.54462 > 10.255.255.2.5000: Flags [S], seq 933499842, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
21:32:36.315220 IP 10.255.255.2.5000 > nor75-22_migr-88-162-121-225.fbx.proxad.net.54462: Flags [R.], seq 0, ack 1, win 0, length 0
21:32:36.860673 IP nor75-22_migr-88-162-121-225.fbx.proxad.net.54462 > 10.255.255.2.5000: Flags [S], seq 933499842, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
21:32:36.860848 IP 10.255.255.2.5000 > nor75-22_migr-88-162-121-225.fbx.proxad.net.54462: Flags [R.], seq 0, ack 1, win 0, length 0
21:32:37.420690 IP nor75-22_migr-88-162-121-225.fbx.proxad.net.54462 > 10.255.255.2.5000: Flags [S], seq 933499842, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
21:32:37.420876 IP 10.255.255.2.5000 > nor75-22_migr-88-162-121-225.fbx.proxad.net.54462: Flags [R.], seq 0, ack 1, win 0, length 0
21:32:37.981244 IP nor75-22_migr-88-162-121-225.fbx.proxad.net.54462 > 10.255.255.2.5000: Flags [S], seq 933499842, win 64240, options [mss 1460,nop,wscale 8,nop,nop,sackOK], length 0
21:32:37.981430 IP 10.255.255.2.5000 > nor75-22_migr-88-162-121-225.fbx.proxad.net.54462: Flags [R.], seq 0, ack 1, win 0, length 0
^C
40 packets captured
40 packets received by filter
0 packets dropped by kernel

In a browser if i check the adresse of ipvps:5000 i get the error ERR_CONNECTION_REFUSED.

Someone has an idea ?

Ysurac, 2020-12-08:
There is an error in SNAT, you shouldn't set any source port.

Xamamule, 2020-12-08:
I change the SNAT rule like this

[image]

I still have the same error if i test he adresse of ipvps:5000

Ysurac, 2020-12-08:
In the redirect port section, you redirect to 192.168.100.1 and not to 192.168.100.187, so traffic never try to go to 192.168.100.187 and SNAT rule is ignored.

Xamamule, 2020-12-08:
With the tutorial, I understood that it is necessary to make an forwarding port on the gateway  (ip OpenMTCProuter ) and then with the SNAT rules from the gateway  to the NAS.

Exemple to the tuto :
[image]

In my opinion I understood nothing

But if i redirect to 192.168.100.187 its very slow. I try again and i go back.

Ysurac, 2020-12-08:
I fixed the doc.

Xamamule, 2020-12-08:
So in my case, i does not need SNAT rule ?

Ysurac, 2020-12-08:
You may need it.

closed 2021-03-13 by github-actions[bot]