Ysurac/openmptcprouter #1440: Wrong EFI loader signature ... -- System halted

winkmichael, 2020-12-08:
## Expected Behavior
New system, x86 hardware using this image;
https://download.openmptcprouter.com/release/v0.56.5/x86_64/targets/x86/64/openmptcprouter-v0.56.5-r0+14653-a439f1bb47-x86-64-generic-ext4-combined-efi.img.gz

Ran the following;
gunzip openmptcprouter-*.img.gz
dd bs=4M if=openmptcprouter-v0.56.5-r0+14653-a439f1bb47-x86-64-generic-ext4-combined-efi.img of=/dev/sda conv=fsync

## Current Behavior

Fails to boot, with the following error;

Wrong EFI loader signature.
early console in extract_kernel
....
Decompressing Linux...
ZX-compressed data is corrupt

-- System halted

Photo of screen here: https://imgur.com/a/s5Wa7os

## Context (Environment)

x86 hardware, Dell R210, SSD hard drive.
I tried installing the image from a different system (attached drive to another computer) to image, and I tried using a Ubuntu live CD to image the system. The result was the same

## Specifications

  - OpenMPTCProuter openmptcprouter-v0.56.5-r0+14653-a439f1bb47-x86-64-generic-ext4-combined-efi.img
  
  - OpenMPTCProuter platform: x86_64 running on Dell r210

winkmichael, 2020-12-09:
I tried this same process using a Dell R320 with a HDD and exact same issue, exact same error. Hard drive is attached via SATA interface.

Ysurac, 2021-03-09:
Why are you using EFI ?

closed 2021-06-13 by github-actions[bot]