Ysurac/openmptcprouter #1441: Fresh install on Raspberry pi 3 // x86_64 - 192.168.100.1 doesn't reply

koopernick, 2020-12-08:
## Expected Behavior
Fresh install on raspberry pi 3. According to the FAQ, the UI should be available at 192.168.100.1

192.168.100.1 doesn't reply

## Current Behavior
I've installed => openmptcprouter-v0.56.5-r0+14653-a439f1bb47-bcm27xx-bcm2710-rpi-3-ext4-factory
on my SDcard for my Pi.

Booted the PI, here is the screenshot of the output => 
[image]

My router (dhcp server) "see" the device => 
[image]

but "http://192.168.100.1/" just times out.
"The server at 192.168.100.1 is taking too long to respond."

192.168.100.1 should display the UI

## Specifications

  - OpenMPTCProuter version: v0.56.5-r0+14653-a439f1bb47-bcm27xx-bcm2710-rpi-3-ext4-factory
  - OpenMPTCProuter VPS version: Not relevant here
  - OpenMPTCProuter platform: RPI3 (also tried with a dedicated hardware, quad port intel card, x86/x64)

Ysurac, 2020-12-08:
Check that you can ping and SSH to 192.168.100.1.
If not check that you have an IP on your client computer in 192.168.100.x network.

koopernick, 2020-12-08:
Thanks for the super fast reply, that was exactly my issue !

I had to change my router subnet in order to be able to manage the default IP of your router, and I didn't think at all to get a new local IP to match the range of your router.

Thanks again for the quick reply, and thanks a lot for your work !

closed 2020-12-08 by koopernick