Ysurac/openmptcprouter #1442: How to fix USB ETH device name so that the order of power on or connectio does not matter the routing

scienty, 2020-12-08:
I have few USB ETH devices that thether cellular internet  connection over USB ETH. My main goal is that (I think that is inline with what openmptcrouter is trying to accomplish) to achieve reliable connection over multiple service providers over remote 4G network still adding all individual bandwidth for an aggregated badnwidth/throughput. Being in India and few option in remote areas, FIrst choice is Airtel 4G/Relinace Jio 4G networks.  I have a RPI 3B+ where I am experimenting and the order is being placed for RPI 4 v1.2(B) for this experiment.

I have a Jio Fi 4G router(JMR815) that can thether internet over USB ETH. 
I also have two broadbad internet service providers (Airtel and ACT Fibernet). This is current situation in Urban aread, but pretty soon I am going to battle test  and be parepaired in very remote areas of India due to my personal choices. I realise that there are may people that are looking for a better siutaiton in the Pendemic time.

Since I work for a major networking company and my spouse too working for a software company, we togher need good combined bandwidth and reliable connection if we had to work from a remote area. Believe me, life is good in remote areas of India in this pendamic time.

When a new USB 4G device is connected. First time the device name allocated to it is eth0. if I had to restart the Pi and plug it again or some other device is thethering ver USB as ETH, the device name for the same device gets renamed to eth1. since at wan1 and wan2 interface we have a restriction to configure static IP address. this is going to missplac the connected device in different bridge/inteface group. What ever the interface/bridge configuration that is done from UI the last time does not host good for the next time if the order in which interfaces/USB devices deteted matters.
## Current Behavior

Every device that is hotplugged (hotplugg.d) should find it's own unique name. OpenMPTCRouter should be able to consider each wan interface independently either DHCP/Static IP lease and handle the upstread traffic propery.
## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter platform:

Ysurac, 2020-12-08:
This may be solved in snapshot image: https://github.com/Ysurac/openmptcprouter/wiki/Snapshots
Each interfaces are considered independently, but it's not always easy to know what USB device is where (some have same serial number).

gareylui, 2020-12-12:
+1, got the same situation here. I have 6 x USB 4G LTE. Every time reboot, I need to reconfigure the interface to match the real arrangement. May I know when this version can be delivered?

gareylui, 2020-12-23:
> This may be solved in snapshot image: https://github.com/Ysurac/openmptcprouter/wiki/Snapshots
> Each interfaces are considered independently, but it's not always easy to know what USB device is where (some have same serial number).

Hi Ysurac, May I know if this patch had released with v0.57? Thanks!

closed 2021-03-28 by github-actions[bot]