Ysurac/openmptcprouter #1446: My connection is blocked?

KaylFr, 2020-12-11:
Hello

Here I will need some advice,

I installed OMR on a Raspberry Pi 4
- A 4G Router (MR600)
- A freebox

I reach 50Mbits / s, and when I add a connection to this configuration a 4G connection and well this speed does not increase at all

My VPS is a Milkywan (also test with an OVH VPS)

I wonder where it can come from, possibility of solving this "problem"?

Thanks in advance

// --------- French

Bonjour

Voila j'aurai besoin de conseil,

J'ai installer OMR sur un Raspberry Pi 4
- Un Routeur 4G (MR600)
- Une freebox

J'atteint 50Mbits/s, et lorsque j'ajoute une connection a cet configuration une connection 4G et bien ce debit n'augmente pas du tous

Mon VPS est un Milkywan (tester aussi avec un VPS OVH)

Je me demande d'ou cela peut il bien venir, possibilité de regler ce "probleme" ?

Merci d'avance

Malaga82, 2020-12-11:
Hello,
in wizard try to check this:
[image]
then do a speedtest and check if speed is aggregated.
MPTCP could be blocked by ISP.

KaylFr, 2020-12-15:
Hello

[image]

I have check this box

Malaga82, 2020-12-16:
it's "normal", go to VPN, OpenVPN, you'll see a wan2 named VPN, stop and start it again.
then wait a bit.

KaylFr, 2020-12-16:
after activating the VPN, my speed caps at 50Mbits

KaylFr, 2020-12-16:
By cons when the 4G connection is alone (without VPN) the download goes up to 120Mbits (freebox connection deactivate)

Malaga82, 2020-12-16:
in /network/mptcp/bandwidth there is a graph of both WANS: did you see traffic for both wan1 and wan2?
is multipath enabled for the wan2 in wizard ( wan1 is master )?
[image]

closed 2021-03-22 by github-actions[bot]