Ysurac/openmptcprouter #1447: Can't boot the router from the hard disk, only from a usb pendrive.

VMax2, 2020-12-12:
## Expected Behavior

Boot the router from hard disk
## Current Behavior

I'm using a minipc x86_64 based; reding the documentation concerning the router installation for x86_64 platform, it's reported that the image may be copied into an HD too. What's happen to me is that if i flash the precompiled efi x86_64 image to a usb pendrive everything work as expected, instead, if i flash into the internal hard disk (that it's an eMMC) the system boot regulary, but it stuck waiting for the root device.
The log message is:
"Waiting for root device PARTUUID=xxxx-xxxx-xxxxx-xxxxxxxxx"

After that the only thing to do is to turn off the minipc.

## Possible Solution

Add the SATA driver to the kernel maybe?
## Steps to Reproduce the Problem

  1. boot with a live distro (like ubuntu 20.04 live for instance)
  2. download the image file to be flashed
  3. flash the image file with balena etcher (or whatever) choosing the hard disk as target
  4. when finished, reboot, remove the live distro, adjust the boot sequence in the bios and restart the system.

## Context (Environment) boot sequence stuck waiting for the root device

I would like to boot the router from the hard disk without the needs to use a usb pendrive but the image supplyed doesn't work for this purpose

## Specifications

  - OpenMPTCProuter version: 0.56.5-r0 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter platform: x86_64

Ysurac, 2020-12-13:
eMMC is not supported in current release. You can try snapshot image, this may be supported.

VMax2, 2020-12-13:
I see, ok, i will try it; do you plane to support it in the next release?

Ysurac, 2020-12-14:
Everything that is in snapshot will be in next release.

VMax2, 2020-12-14:
Ok thank you so much

closed 2021-03-19 by github-actions[bot]