Ysurac/openmptcprouter #1450: help me

ascomputer, 2020-12-13:
hello everyone I updated to version 0.56.5 and on the dedicated server I have 0.1022 55.4.74.mptcp
during use, all the wan are disconnected, can someone help me?

Sun Dec 13 15:32:30 2020 daemon.info omr-tracker-ss: Shadowsocks is down (can't contact via http 1.1.1.1, 1.0.0.1, 212.27.48.10)
Sun Dec 13 15:32:30 2020 user.notice omr-bypass: Reload dnsmasq...
Sun Dec 13 15:32:31 2020 daemon.info dnsmasq[12490]: read /etc/hosts - 4 addresses
Sun Dec 13 15:32:31 2020 daemon.info dnsmasq[12490]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Sun Dec 13 15:32:31 2020 daemon.info dnsmasq-dhcp[12490]: read /etc/ethers - 0 addresses
Sun Dec 13 15:32:31 2020 user.notice omr-bypass: OMR-ByPass is running
Sun Dec 13 15:32:31 2020 user.info : using tcp fast open
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-redir[16958]: enable TCP no-delay
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-redir[16958]: initializing ciphers... chacha20-ietf-poly1305
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-redir[16958]: listening at 0.0.0.0:1101
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-redir[16958]: tcp port reuse enabled
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-redir[16958]: UDP relay enabled
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-redir[16958]: udp port reuse enabled
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-redir[16958]: running from root user
Sun Dec 13 15:32:31 2020 user.info : using tcp fast open
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-local[16954]: using tcp fast open
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-redir[16955]: enable TCP no-delay
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-local[16954]: enable TCP no-delay
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-local[16954]: initializing ciphers... chacha20-ietf-poly1305
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-local[16954]: listening at 127.0.0.1:1111
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-local[16954]: tcp port reuse enabled
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-local[16954]: udprelay enabled
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-local[16954]: udp port reuse enabled
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-local[16954]: running from root user
Sun Dec 13 15:32:31 2020 user.info : using tcp fast open
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-redir[16957]: enable TCP no-delay
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-redir[16957]: initializing ciphers... chacha20-ietf-poly1305
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-redir[16957]: listening at 0.0.0.0:1101
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-redir[16957]: tcp port reuse enabled
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-redir[16957]: UDP relay enabled
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-redir[16957]: udp port reuse enabled
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-redir[16957]: running from root user
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-redir[16955]: resolving hostname to IPv6 address first
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-redir[16955]: initializing ciphers... chacha20-ietf-poly1305
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-redir[16955]: listening at 0.0.0.0:1100
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-redir[16955]: tcp port reuse enabled
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-redir[16955]: UDP relay enabled
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-redir[16955]: udp port reuse enabled
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-redir[16955]: running from root user
Sun Dec 13 15:32:31 2020 user.info : using tcp fast open
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-redir[16956]: enable TCP no-delay
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-redir[16956]: resolving hostname to IPv6 address first
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-redir[16956]: initializing ciphers... chacha20-ietf-poly1305
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-redir[16956]: listening at 0.0.0.0:1100
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-redir[16956]: tcp port reuse enabled
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-redir[16956]: UDP relay enabled
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-redir[16956]: udp port reuse enabled
Sun Dec 13 15:32:31 2020 daemon.info /usr/bin/ss-redir[16956]: running from root user
Sun Dec 13 15:32:31 2020 user.notice Shadowsocks: Rules DOWN
Sun Dec 13 15:32:32 2020 daemon.notice procd: /etc/rc.d/S99omr-bypass: iptables-restore: line 2 failed
Sun Dec 13 15:32:32 2020 kern.info kernel: [   52.567951] bcmgenet fd580000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
Sun Dec 13 15:32:32 2020 kern.info kernel: [   52.576339] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
Sun Dec 13 15:32:32 2020 daemon.notice netifd: Network device 'eth0' link is up
Sun Dec 13 15:32:32 2020 daemon.notice netifd: Interface 'lan' has link connectivity
Sun Dec 13 15:32:32 2020 daemon.err omr-tracker[10188]: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
Sun Dec 13 15:32:32 2020 user.notice Shadowsocks: Rules UP
Sun Dec 13 15:32:32 2020 daemon.err omr-tracker[10188]: Flush terminated
Sun Dec 13 15:32:32 2020 user.notice Shadowsocks: Reload omr-bypass rules
Sun Dec 13 15:32:32 2020 user.notice omr-bypass: Starting OMR-ByPass...
Sun Dec 13 15:32:33 2020 daemon.debug dnsmasq[12490]: listening on eth0(#2): fdb1:9345:668f::1 port 53
Sun Dec 13 15:32:33 2020 daemon.debug dnsmasq[12490]: listening on eth0(#2): fe80::dea6:32ff:fe43:d506%eth0 port 53
Sun Dec 13 15:32:33 2020 daemon.info dnsmasq[12490]: read /etc/hosts - 4 addresses
Sun Dec 13 15:32:33 2020 daemon.info dnsmasq[12490]: read /tmp/hosts/odhcpd - 1 addresses
Sun Dec 13 15:32:33 2020 daemon.info dnsmasq[12490]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Sun Dec 13 15:32:33 2020 daemon.info dnsmasq-dhcp[12490]: read /etc/ethers - 0 addresses
Sun Dec 13 15:32:33 2020 daemon.info dnsmasq[12490]: read /etc/hosts - 4 addresses
Sun Dec 13 15:32:33 2020 daemon.info dnsmasq[12490]: read /tmp/hosts/odhcpd - 1 addresses
[... 551 lines cut ...]
Sun Dec 13 15:49:29 2020 user.notice Shadowsocks: Reload omr-bypass rules
Sun Dec 13 15:49:29 2020 user.notice omr-bypass: Starting OMR-ByPass...
Sun Dec 13 15:49:30 2020 kern.err kernel: [ 1071.021171] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Dec 13 15:49:30 2020 user.notice omr-bypass: Reload dnsmasq...
Sun Dec 13 15:49:31 2020 daemon.info dnsmasq[19327]: read /etc/hosts - 4 addresses
Sun Dec 13 15:49:31 2020 daemon.info dnsmasq[19327]: read /tmp/hosts/odhcpd - 1 addresses
Sun Dec 13 15:49:31 2020 daemon.info dnsmasq[19327]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Sun Dec 13 15:49:31 2020 daemon.info dnsmasq-dhcp[19327]: read /etc/ethers - 0 addresses
Sun Dec 13 15:49:31 2020 user.notice omr-bypass: OMR-ByPass is running
Sun Dec 13 15:49:34 2020 kern.err kernel: [ 1075.007450] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Dec 13 15:49:36 2020 kern.err kernel: [ 1077.012912] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Dec 13 15:49:39 2020 kern.err kernel: [ 1080.012381] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Dec 13 15:49:42 2020 kern.err kernel: [ 1083.021040] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Dec 13 15:49:45 2020 kern.err kernel: [ 1086.010580] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Dec 13 15:49:48 2020 kern.err kernel: [ 1089.010613] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Dec 13 15:49:51 2020 kern.err kernel: [ 1092.017247] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Dec 13 15:49:55 2020 kern.err kernel: [ 1096.009439] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Dec 13 15:49:55 2020 kern.err kernel: [ 1096.019844] brcmfmac: brcmf_cfg80211_dump_survey: Could not get noise (-512)
Sun Dec 13 15:49:57 2020 kern.err kernel: [ 1098.012562] brcmfmac: brcmf_cfg80211_dump_survey: chanspec failed (-512)
Sun Dec 13 15:50:01 2020 kern.err kernel: [ 1102.009292] brcmfmac: brcmf_cfg80211_dump_survey: chanspec failed (-512)
Sun Dec 13 15:50:03 2020 kern.err kernel: [ 1104.011804] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Dec 13 15:50:07 2020 kern.err kernel: [ 1108.028821] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512
Sun Dec 13 15:50:13 2020 kern.err kernel: [ 1114.010468] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST unsupported, err=-512

thanks

Gthubln060920, 2020-12-13:
I had the same problem after upgrading from 56.3 to 56.5.  My 1 static wan was up, 3 dhcp wan were down.
Didn’t spend any more time troubleshooting. I just downgraded back to 56.3
This was on X86

Ysurac, 2020-12-13:
Please use template when you open an issue.
I can see this in log "time disparity of 1808 minutes detected", so there is a time issue on VPS or on router.

closed 2021-03-19 by github-actions[bot]