Ysurac/openmptcprouter #1451: bandwith agregation and cases

lyr847, 2020-12-13:
Hello,

My config is  :
Raspberry PI4 64 bits with OMR v0.56.5
OVH VPS (450 Mbits/s)
VDSL Box (12Mbits/s)
4G LTE Router (130 Mbits/s).

Multipath is on and VDSL Box is the master.
I use a 4G 100 Gb data plan. When it is full, total bandwith is good with average 140 Mbits/s. So bandwith agregation works in this case (130+12 Mbits/s).

On the other hand, when I have fully consumed my 100 Gb data plan, total bandwith falls to 1 Mbits/s. Why ? Ok My Orange Telecom provider restrains speed to 1 Mbits/s, but why bandwith agregation does not work in this case ?
I should normally have 12-13 Mbits in this case  (VDSL 12Mbits/s + Limited LTE 1Mbits/s).

Thank you by advance

Ysurac, 2020-12-14:
Please use the issue template.
Check system log if there is something different when LTE is limited.

lyr847, 2020-12-18:
Ok it's done --> [https://github.com/Ysurac/openmptcprouter/issues/1464#issue-771049075](url)
I see nothing special in system log

closed 2020-12-18 by Ysurac