Ysurac/openmptcprouter #1452: Service omr6in4@user0 failed

Heugy67, 2020-12-14:
Service omr6in4@user0 is failed even after restart or reboot VPS.

I have set the IPv6 on VPS after fresh installation of OpenMPTCProuter VPS.
I have tried to relaunch installation of OpenMPTCProuter VPS by relaunching script dedicated.

How can i resolve this issue ?

## Specifications
- VPS : OVH
- OS : Debian 10

  - OpenMPTCProuter version: N/A
  - OpenMPTCProuter VPS version: 0.1022
  - OpenMPTCProuter platform: RPI4 - not installed yet

Ysurac, 2020-12-14:
What is the error you get via ```journalctl -xe``` after launching omr6in4@user ?

Heugy67, 2020-12-14:
Here the log of "journalctl -xe" :

Dec 14 21:20:55 vps-eff9741b systemd[1]: Starting OMR6IN4 on user0...
-- Subject: A start job for unit omr6in4@user0.service has begun execution
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit omr6in4@user0.service has begun execution.
--
-- The job identifier is 12714.
Dec 14 21:20:55 vps-eff9741b omr-6in4-run[16912]: usage: omr-6in4-run FILE
Dec 14 21:20:55 vps-eff9741b systemd[1]: omr6in4@user0.service: Main process exited, code=exited, status=1/FAILURE
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- An ExecStart= process belonging to unit omr6in4@user0.service has exited.
--
-- The process' exit code is 'exited' and its exit status is 1.
Dec 14 21:20:55 vps-eff9741b systemd[1]: omr6in4@user0.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- The unit omr6in4@user0.service has entered the 'failed' state with result 'exit-code'.
Dec 14 21:20:55 vps-eff9741b systemd[1]: Failed to start OMR6IN4 on user0.
-- Subject: A start job for unit omr6in4@user0.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit omr6in4@user0.service has finished with a failure.
--
-- The job identifier is 12714 and the job result is failed.

Ysurac, 2020-12-15:
The service need a file that is created from the API when router VPN is up. This file is not (yet ?) created.

Heugy67, 2020-12-15:
I understand better the process.
Effectively, the file generated by router over API isn't yet created as i haven't build the router now.
I will do it and, depending of result, i will close this issue or continue to complete information to resolve it.

Thank you very much for your support (quick and high-performance).

closed 2020-12-19 by Heugy67