Ysurac/openmptcprouter #1454: bug after restart rasberry

tallyhome, 2020-12-14:
## Expected Behavior

Hello,

I appeal to you because I dry.

I tried everything and I can't find any solutions ..

so VPS install OK

Rasberry facilities OK

[image]
[image]

## Current Behavior

then I enter all the info everything works perfectly fine, but if I reboot the pi4 or if I turn it off and turn it on again nothing works.

If I flash the SD again and start again, I enter all the info and it works but if I reboot, nothing works.

## Specifications

  - OpenMPTCProuter version: 

I have tester
image ext4: openmptcprouter-v0.56.5-r0 + 14653-a439f1bb47-bcm27xx-bcm2709-rpi-2-ext4-factory.img.gz
and
image ext4: openmptcprouter-v0.56.5-r0 + 14653-a439f1bb47-bcm27xx-bcm2711-rpi-4-ext4-factory.img.gz

as well as this one on a PI3 B +
image ext4: openmptcprouter-v0.56.5-r0 + 14653-a439f1bb47-bcm27xx-bcm2709-rpi-2-ext4-factory.img.gz

and
image ext4: openmptcprouter-v0.56.5-r0 + 14653-a439f1bb47-bcm27xx-bcm2710-rpi-3-ext4-factory.img.gz

  - OpenMPTCProuter VPS version: 

I tested on a Debian 9
a Debian 10
Ubuntu 18
and Ubuntu 20

  - OpenMPTCProuter platform: 

testeur sur un RPI 4
et sur un RPI 3B +

Ysurac, 2020-12-14:
In first screenshot, there is no VPS defined, on second screenshot there is many problems ;)
On RPI4, you should use only openmptcprouter-v0.56.5-r0 + 14653-a439f1bb47-bcm27xx-bcm2711-rpi-4-ext4-factory.img.gz image.
What is the status page info when everything work ?
What do you set in wizard page ?
What do you have in system log when you have issue ?

tallyhome, 2020-12-14:
hi

so with a vps or not, the problem remains the same
I also did some tests with this image
  openmptcprouter-v0.56.5-r0 + 14653-a439f1bb47-bcm27xx-bcm2711-rpi-4-ext4-factory.img.gz

when everything works, it's screen 1
https://user-images.githubusercontent.com/37673782/102083343-ac3e5900-3e13-11eb-9efe-2083dd699704.jpg

for the screen wizard page:

[Log System.txt](https://github.com/Ysurac/openmptcprouter/files/5690289/Log.System.txt)
[image]
[image]

Ysurac, 2020-12-14:
In screenshot 1, it's not working as it's using direct output and not VPS.
How are connected the WANs ? it's strange to have wan1 and wan2 as real physical interface and not as MacVlan.

ebuyfr, 2020-12-14:
Hello,
J'ai eu un gros problème de ce genre lors de la mise à jour en dernière version, au la première config WAN ça allait, dès que je faisait ma deuxième en brachant le câble je perdais la main, je ne me connectait même plus sur l'interface d'admin, même en rechargeant une config sauvegardée ça ne voulait pas. J'y ai passé la nuit, 4 réinstall from scratch, au grand regret de ma femme...
La solution que j'ai trouvée :  fresh install, supprimer toutes les interfaces de type WAN, les recréer à la main dans Interfaces (ne pas utiliser Wizard) puis les administrer dans le Wizard (j'ai rebooté entre chaque admin mais je ne sais pas si c'est vraiment nécessaire).
J'ai récupérer ma config et ça fonctionne à nouveau !!

Pour info j'ai une Freebox et j'ai trouvé une petite astuce qui m'embêtait auparavant concernant le lien entre le Player et le Server (Je gardais une @IP pour le Player dans la Freebox que je démarrais en premier avant de démarrer mon RasberryPi qui sert de serveur DHCP dans OpenMptcpRouter.)
J'ai entré dans Nom d'hôtes : mafreebox.freebox.fr = @IP du Server, la télé fonctionne !!
Pas les applications (erreur réseau) mais le principal c'est la télé ;)

tallyhome, 2020-12-14:
look Movie 

New install on sdcard and new config 
for this test i just add freebox

https://drive.obione.net/drive/s/69yG6cC3gi0nE3e45YedyiHvTuBgCa

Ysurac, 2020-12-14:
Please answer to questions...
Why do you set interface type as normal ? This can't work that way.

tallyhome, 2020-12-14:
I go through the configuration wizard.
and on your site it is noted as normal.
that said, I just tested with MacVlan type and I have the same problem

Ysurac, 2020-12-14:
It is not noted as normal in the documentation.
Try again with macvlan. Only set VPS IP, key, and WANs IPs/gateway.

tallyhome, 2020-12-14:
[image]
[image]

tallyhome, 2020-12-14:
[image]

Ysurac, 2020-12-14:
wlan0 as physical interface ? you use wifi now ? Why do you don't keep default configuration ?

tallyhome, 2020-12-14:
I do nothing, it's your system which automatically puts in wlan0 the wan1 has disappeared.

tallyhome, 2020-12-14:
[video-carte-reseau.zip](https://github.com/Ysurac/openmptcprouter/files/5690805/video-carte-reseau.zip)
Look video

Ysurac, 2020-12-14:
As you can see in your own video, default physical interface is eth0.
A MacVLAN interface is a "virtual" interface based on a real physical interface. By default on RPI, as there is only one physical port, 2 MacVLANs interfaces WAN1 and WAN2 are created based on eth0. It's not possible to base a MacVLAN interface on a MacVLAN interface.

tallyhome, 2020-12-14:
okay,
so i do what?

ebuyfr, 2020-12-14:
Hey are you read my answer?
Go to Interfaces and delete all WANs.
Reboot. Go back to Interfaces and create them by hand without going through the Wizard. Use the Wan Interfaces created in Interfaces to add them with the Wizard and reboot.

tallyhome, 2020-12-14:
ebuyfr and for interface physique a choose eth0?

ebuyfr, 2020-12-14:
Yes and Type is MacVLAN

tallyhome, 2020-12-14:
ok, another problem with this config.
look screen

MacVLAN and eth0
[image]

ebuyfr, 2020-12-14:
at the top right, the changes are not applied!

tallyhome, 2020-12-14:
samme problem

[image]

tallyhome, 2020-12-14:
[image]

Ysurac, 2020-12-14:
There is 2 issues here:
* You didn't set a VPS IP at first configuration, this disable the proxy. In the wizard check "advanced settings" checkbox and set proxy to shadowsocks (it's the default)
* Check router and VPS time, there is a time problem as it's said.

tallyhome, 2020-12-14:
hi
i not proxy shadow
[image]

Ysurac, 2020-12-14:
Advanced settings checkbox in wizard.

Randdrick, 2020-12-14:
Hi tallyhome,
if your RPI 4 lost his config after restart it's beacause you have a problem on your SDCARD. You must reinstall your RPI as if the first install.
To clean your partition, enter the following command on your RPi 4: ( see documentation : https://openwrt.org/docs/guide-user/troubleshooting/failsafe_and_factory_reset )

* firstboot -y && reboot now

Now, you can restore a backup if you have.

tallyhome, 2020-12-14:
Ysurac : yes in the Advanced settings in the wizard.

I am not a proxy activated

and Randdrick 
i tested 2 sdcard and same problem

Ysurac, 2020-12-14:
So, as I said, set proxy to "Shadowsocks" in the wizard instead of none (not advanced settings tab, it's advanced settings checkbox).
Also fix time on VPS and router.

tallyhome, 2020-12-14:
ok, i fine for proxy 👍 and for 
how do we fix this problem
Large time difference between server and router

tallyhome, 2020-12-14:
it seems to work.

now i have another question.
is it possible.

I have an internal network with synology rt1900a router.
or above everything is connected, printer, computer, tablet, phone, Nas, etc ...

is it possible to add internet access via OMR (on rpi4)

because I know that OMR has its own DHCP

closed 2021-03-20 by github-actions[bot]