Ysurac/openmptcprouter #1455: OpenMPTCProuter v0.57 beta

Ysurac, 2020-12-15:
Changes:
* BBR2 alpha support for 64bits processor
* OpenVPN UDP Bonding support
* V2Ray update
* Glorytun UDP and TCP new interfaces with better support
* TCPdump web interface
* Fixes

DeeJayMX, 2020-12-16:
👍😊

Malaga82, 2020-12-16:
What's new with _OpenVPN UDP Bonding support_ ?

Ysurac, 2020-12-16:
This use Linux Ethernet Bonding Driver, can give quite good result when link are good with same latency.

shrieram, 2020-12-16:
Snapshot download link please ?

Ysurac, 2020-12-16:
It's in documentation: https://github.com/Ysurac/openmptcprouter/wiki/Snapshots

darthclide, 2020-12-20:
-OpenVPN UDP Bonding support

Am I missing something or does this mean bonding UDP connections such as SRT protocol for streaming? Does it also mean that eventually glorytun can bond UDP connections?

Ysurac, 2020-12-20:
OpenVPN UDP Bonding is Link Bonding, but instead of classical physical links it's VPN links. This can only with good connections.
In OpenMPTCProuter, all VPN can bond UDP connections. V2Ray Proxy can also bond UDP connections.

darthclide, 2020-12-22:
Ah I see. Perhaps this explains why people had a few issues with SRT when bonding. I would assume things like glorytun are only prepared for bonding things like watching a live stream or VOIP, and not something as bleeding edge such as a transport protocol like SRT?

Ysurac, 2020-12-22:
VPN or proxy doesn't care about any protocol that are in application layer. They can have some problems when we want to use them for a new protocol in protocol layer like TCP, UDP, SCTP,...
For SRT some VPN or V2Ray may help.

darthclide, 2020-12-23:
Well with the knowledge that this should theoretically work okay, I will try doing SRT tests with OMR and see what happens. Thank you for responding