Ysurac/openmptcprouter #1457: Sierra Wireless AirPrime Modem as Master Interface

rnunes77, 2020-12-15:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**

Hello, I made the configuration and everything works very well. However, I purchased a sierra wireless AirPrime MC 7340 modem (to improve the quality and stability of the connection and use it as a Master interface) and apparently it is not recognized by the system. Another point that I would like to raise, during use, in the absence of the master interface, what is the risk that I run in relation to the stability of the connection?
**Describe why you are asking for this support?**

to improve the quality and stability of the connection and use it as a Master interface

**Additional context**

OpenMPTCProuter version: 0.55.1
OpenMPTCProuter VPS version: 0.1017
OpenMPTCProuter platform: Raspberry Pi 4 Model B

Ysurac, 2020-12-16:
I can't find any info about a AirPrime MC7340, so no idea of the used chipset and needed driver for that.

rnunes77, 2020-12-16:
Hi, Ysurac! I'm sorry, I made a mistake.  follows the correct description:

MC7430 LTE 4G Module FDD-LTE TDD-LTE CAT6 interface USB Card WWAN HSPA +
GNSS 3.0 MBIM 4G

TKS!!

Att:

Rafael Nunes

Ysurac, 2020-12-16:
What do you have in State->Kernel log when you plug the modem ?

rnunes77, 2020-12-16:
[   69.295851] usb 1-1.1: new high-speed USB device number 6 using xhci_hcd
[   69.404621] usb 1-1.1: config 1 has an invalid interface number: 12 but max is 4
[   69.412965] usb 1-1.1: config 1 has an invalid interface number: 13 but max is 4
[   69.421291] usb 1-1.1: config 1 has an invalid interface number: 13 but max is 4
[   69.429635] usb 1-1.1: config 1 has no interface number 1
[   69.435930] usb 1-1.1: config 1 has no interface number 4
[   69.443024] usb 1-1.1: New USB device found, idVendor=1199, idProduct=9071, bcdDevice= 0.06
[   69.452300] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   69.460508] usb 1-1.1: Product: Sierra Wireless MC7430 Qualcomm® Snapdragon™ X7 LTE-A
[   69.469514] usb 1-1.1: Manufacturer: Sierra Wireless, Incorporated
[   69.476624] usb 1-1.1: SerialNumber: LU71747484041014
[   69.482776] usb 1-1.1: no of_node; not parsing pinctrl DT
[   69.488250] qcserial 1-1.1:1.0: no of_node; not parsing pinctrl DT
[   69.489077] qcserial 1-1.1:1.0: Qualcomm USB modem converter detected
[   69.496620] qcserial ttyUSB0: no of_node; not parsing pinctrl DT
[   69.496758] usb 1-1.1: Qualcomm USB modem converter now attached to ttyUSB0
[   69.504964] qcserial 1-1.1:1.2: no of_node; not parsing pinctrl DT
[   69.505856] qcserial 1-1.1:1.2: Qualcomm USB modem converter detected
[   69.513493] qcserial ttyUSB1: no of_node; not parsing pinctrl DT
[   69.513614] usb 1-1.1: Qualcomm USB modem converter now attached to ttyUSB1
[   69.522297] qcserial 1-1.1:1.3: no of_node; not parsing pinctrl DT
[   69.523493] qcserial 1-1.1:1.3: Qualcomm USB modem converter detected
[   69.532011] qcserial ttyUSB2: no of_node; not parsing pinctrl DT
[   69.532187] usb 1-1.1: Qualcomm USB modem converter now attached to ttyUSB2
[   69.541071] cdc_mbim 1-1.1:1.12: no of_node; not parsing pinctrl DT
[   69.561861] cdc_mbim 1-1.1:1.12: cdc-wdm0: USB WDM device
[   69.568763] cdc_mbim 1-1.1:1.12 wwan0: register 'cdc_mbim' at usb-0000:01:00.0-1.1, CDC MBIM, ba:89:02:3a:95:f8

Ysurac, 2020-12-16:
The dongle is detected and added as wwan0 interface. It need to be configured using ModemManager protocol in Network->Interface

rnunes77, 2020-12-16:
Ok.. Thanks Ysurac!!
[image: image.png]

[image: image.png]

Even after configuration, I still have an error status.

rnunes77, 2020-12-16:
[image]
[image]

rnunes77, 2020-12-17:
Hi, Ysurac! Thanks again! 

I was performing the configuration in another option. I currently have this status on the WAN interface that uses the Sierra Modem.
[image]

Ysurac, 2020-12-18:
It's not always an issue, some dongle doesn't allow to ping gateway. There is an option in System->OpenMPTCProuter, "Advanced settings" tab to "Disable gateway ping" and to remove this message.

rnunes77, 2021-02-18:
Thanks! After your help I was able to configure the modem.

closed 2021-02-18 by rnunes77