Ysurac/openmptcprouter #1458: Port forwarding from WAN to LAN

dnwk, 2020-12-16:
Version 0.56.5 on Router and v0.1022 on VPS

Is it possible to do port forwarding from WAN to LAN device when this device is running OMR-Bypass all traffic to WAN?
Currently, I could forward port from VPN to LAN when the specific traffic is not bypassed. I am trying to figure out if there is a way to do port forwarding from WAN instead of from VPS on the far end.

Ysurac, 2020-12-16:
Configuring a port forwarding from wan zone is not working ?

dnwk, 2020-12-16:
No, I configure a port forwarding from WAN to LAN. But I still can't visit the WAN public IP to get to LAN port. Forwarding from VPN zone works fine.

Ysurac, 2020-12-16:
This should work without issues but opening port may be needed on modem/box/...

dnwk, 2020-12-16:
I'm getting public IP address directly on this box. So WAN port is getting public IP directly

Ysurac, 2020-12-16:
Check with tcpdump that you get something.

dnwk, 2020-12-16:
No. | Time | Source | Destination | Protocol | Length | Info
-- | -- | -- | -- | -- | -- | --
137 | 2.580284 | External IP | WAN IP | TCP | 74 | 45034  >  443 [SYN] Seq=0 Win=65535 Len=0 MSS=1362 SACK_PERM=1 TSval=1834501263 TSecr=0 WS=512
138 | 2.580284 | External IP | WAN IP | TCP | 74 | 47566  >  443 [SYN] Seq=0 Win=65535 Len=0 MSS=1362 SACK_PERM=1 TSval=1834501266 TSecr=0 WS=512
159 | 2.880998 | External IP | WAN IP | TCP | 74 | 34688  >  443 [SYN] Seq=0 Win=65535 Len=0 MSS=1362 SACK_PERM=1 TSval=1834501516 TSecr=0 WS=512
209 | 3.622974 | External IP | WAN IP | TCP | 74 | [TCP Retransmission] 45034  >  443 [SYN] Seq=0 Win=65535 Len=0 MSS=1362 SACK_PERM=1 TSval=1834502274 TSecr=0 WS=512
210 | 3.623025 | External IP | WAN IP | TCP | 74 | [TCP Retransmission] 47566  >  443 [SYN] Seq=0 Win=65535 Len=0 MSS=1362 SACK_PERM=1 TSval=1834502273 TSecr=0 WS=512
223 | 3.924048 | External IP | WAN IP | TCP | 74 | [TCP Retransmission] 34688  >  443 [SYN] Seq=0 Win=65535 Len=0 MSS=1362 SACK_PERM=1 TSval=1834502529 TSecr=0 WS=512
407 | 5.611911 | External IP | WAN IP | TCP | 74 | [TCP Retransmission] 45034  >  443 [SYN] Seq=0 Win=65535 Len=0 MSS=1362 SACK_PERM=1 TSval=1834504289 TSecr=0 WS=512
408 | 5.611951 | External IP | WAN IP | TCP | 74 | [TCP Retransmission] 47566  >  443 [SYN] Seq=0 Win=65535 Len=0 MSS=1362 SACK_PERM=1 TSval=1834504289 TSecr=0 WS=512
433 | 5.917825 | External IP | WAN IP | TCP | 74 | [TCP Retransmission] 34688  >  443 [SYN] Seq=0 Win=65535 Len=0 MSS=1362 SACK_PERM=1 TSval=1834504545 TSecr=0 WS=512
625 | 9.500271 | External IP | WAN IP | TCP | 74 | 53684  >  443 [SYN] Seq=0 Win=65535 Len=0 MSS=1362 SACK_PERM=1 TSval=1834508289 TSecr=0 WS=512
626 | 9.500272 | External IP | WAN IP | TCP | 74 | 26450  >  443 [SYN] Seq=0 Win=65535 Len=0 MSS=1362 SACK_PERM=1 TSval=1834508289 TSecr=0 WS=512
648 | 9.860466 | External IP | WAN IP | TCP | 74 | 51846  >  443 [SYN] Seq=0 Win=65535 Len=0 MSS=1362 SACK_PERM=1 TSval=1834508545 TSecr=0 WS=512
1214 | 17.868421 | External IP | WAN IP | TCP | 74 | 32616  >  443 [SYN] Seq=0 Win=65535 Len=0 MSS=1362 SACK_PERM=1 TSval=1834516481 TSecr=0 WS=512
1215 | 17.868491 | External IP | WAN IP | TCP | 74 | 59156  >  443 [SYN] Seq=0 Win=65535 Len=0 MSS=1362 SACK_PERM=1 TSval=1834516481 TSecr=0 WS=512
1233 | 18.170132 | External IP | WAN IP | TCP | 74 | 31008  >  443 [SYN] Seq=0 Win=65535 Len=0 MSS=1362 SACK_PERM=1 TSval=1834516737 TSecr=0 WS=512
1362 | 20.465307 | External IP | WAN IP | TCP | 74 | 31134  >  443 [SYN] Seq=0 Win=65535 Len=0 MSS=1362 SACK_PERM=1 TSval=1834518996 TSecr=0 WS=512
1363 | 20.465381 | External IP | WAN IP | TCP | 74 | 48152  >  443 [SYN] Seq=0 Win=65535 Len=0 MSS=1362 SACK_PERM=1 TSval=1834518990 TSecr=0 WS=512
1396 | 20.765944 | External IP | WAN IP | TCP | 74 | 29022  >  443 [SYN] Seq=0 Win=65535 Len=0 MSS=1362 SACK_PERM=1 TSval=1834519255 TSecr=0 WS=512
1469 | 21.36867 | External IP | WAN IP | TCP | 74 | [TCP Retransmission] 48152  >  443 [SYN] Seq=0 Win=65535 Len=0 MSS=1362 SACK_PERM=1 TSval=1834520001 TSecr=0 WS=512
1470 | 21.368723 | External IP | WAN IP | TCP | 74 | [TCP Retransmission] 31134  >  443 [SYN] Seq=0 Win=65535 Len=0 MSS=1362 SACK_PERM=1 TSval=1834520001 TSecr=0 WS=512
1488 | 21.669233 | External IP | WAN IP | TCP | 74 | [TCP Retransmission] 29022  >  443 [SYN] Seq=0 Win=65535 Len=0 MSS=1362 SACK_PERM=1 TSval=1834520257 TSecr=0 WS=512
1592 | 23.052998 | External IP | WAN IP | TCP | 74 | [TCP Retransmission] 31134  >  443 [SYN] Seq=0 Win=65535 Len=0 MSS=1362 SACK_PERM=1 TSval=1834522017 TSecr=0 WS=512
1598 | 23.067725 | External IP | WAN IP | TCP | 74 | [TCP Retransmission] 48152  >  443 [SYN] Seq=0 Win=65535 Len=0 MSS=1362 SACK_PERM=1 TSval=1834522018 TSecr=0 WS=512
1640 | 23.573296 | External IP | WAN IP | TCP | 74 | [TCP Retransmission] 29022  >  443 [SYN] Seq=0 Win=65535 Len=0 MSS=1362 SACK_PERM=1 TSval=1834522273 TSecr=0 WS=512
2084 | 27.553278 | External IP | WAN IP | TCP | 74 | 28786  >  443 [SYN] Seq=0 Win=65535 Len=0 MSS=1362 SACK_PERM=1 TSval=1834526209 TSecr=0 WS=512
2085 | 27.55345 | External IP | WAN IP | TCP | 74 | 41024  >  443 [SYN] Seq=0 Win=65535 Len=0 MSS=1362 SACK_PERM=1 TSval=1834526209 TSecr=0 WS=512
2102 | 27.854398 | External IP | WAN IP | TCP | 74 | 39280  >  443 [SYN] Seq=0 Win=65535 Len=0 MSS=1362 SACK_PERM=1 TSval=1834526465 TSecr=0 WS=512
2616 | 35.720506 | External IP | WAN IP | TCP | 74 | 41712  >  443 [SYN] Seq=0 Win=65535 Len=0 MSS=1362 SACK_PERM=1 TSval=1834534402 TSecr=0 WS=512
2617 | 35.720588 | External IP | WAN IP | TCP | 74 | 19266  >  443 [SYN] Seq=0 Win=65535 Len=0 MSS=1362 SACK_PERM=1 TSval=1834534401 TSecr=0 WS=512
2632 | 36.021877 | External IP | WAN IP | TCP | 74 | 49630  >  443 [SYN] Seq=0 Win=65535 Len=0 MSS=1362 SACK_PERM=1 TSval=1834534657 TSecr=0 WS=512

Ysurac, 2020-12-16:
So data are coming.
Check with tcpdump on lan interface if traffic is still visible. If yes, check that SNAT is set, if not it's possible that traffic is redirected to shadowsocks and I will check that.

dnwk, 2020-12-16:
What do you mean by SNAT is set?  I only set port forwarding in the firewall port forwarding tab. I didn't create a separate SNAT rules.

dnwk, 2020-12-17:
Do I need to create a SNAT rules besides Port Forwarding rules?

closed 2020-12-21 by dnwk