Ysurac/openmptcprouter #1459: omr admin is crashing on VPS, NO token can't conact script on OMR side

jfh75, 2020-12-16:
## Expected Behavior
After having in this order : 
1) Updated the VPS to 1022
2) Updated OMR to 56.5
OMR should be ok like before upgrade

Current behavior : 
 user.notice OMR-VPS: Can't get token, try later
 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)

On VPS side omr admin seems down : 
faillog extract, this repeated endlessly : 
Dec 16 09:30:12 vps systemd[1]: Stopped OMR-Admin.
Dec 16 09:30:12 vps systemd[1]: Started OMR-Admin.
Dec 16 09:30:12 vps omr-admin.py[39331]: Traceback (most recent call last):
Dec 16 09:30:12 vps omr-admin.py[39331]:   File "/usr/local/bin/omr-admin.py", line 35, in <module>
Dec 16 09:30:12 vps omr-admin.py[39331]:     from netjsonconfig import OpenWrt
Dec 16 09:30:12 vps omr-admin.py[39331]:   File "/usr/local/lib/python3.8/dist-packages/netjsonconfig/__init__.py", line 5, in <module>
Dec 16 09:30:12 vps omr-admin.py[39331]:     from .backends.openvpn.openvpn import OpenVpn  # noqa
Dec 16 09:30:12 vps omr-admin.py[39331]:   File "/usr/local/lib/python3.8/dist-packages/netjsonconfig/backends/openvpn/openvpn.py", line 2, in <module>
Dec 16 09:30:12 vps omr-admin.py[39331]:     from ..base.backend import BaseBackend
Dec 16 09:30:12 vps omr-admin.py[39331]:   File "/usr/local/lib/python3.8/dist-packages/netjsonconfig/backends/base/backend.py", line 9, in <module>
Dec 16 09:30:12 vps omr-admin.py[39331]:     from jsonschema import Draft4Validator, draft4_format_checker
Dec 16 09:30:12 vps omr-admin.py[39331]: ModuleNotFoundError: No module named 'jsonschema'
Dec 16 09:30:12 vps systemd[1]: omr-admin.service: Main process exited, code=exited, status=1/FAILURE
Dec 16 09:30:12 vps systemd[1]: omr-admin.service: Failed with result 'exit-code'.
Dec 16 09:30:12 vps systemd[1]: omr-admin.service: Scheduled restart job, restart counter is at 4.
Dec 16 09:30:12 vps systemd[1]: Stopped OMR-Admin.
Dec 16 09:30:12 vps systemd[1]: Started OMR-Admin.
Dec 16 09:30:12 vps omr-admin.py[39333]: Traceback (most recent call last):
Dec 16 09:30:12 vps omr-admin.py[39333]:   File "/usr/local/bin/omr-admin.py", line 35, in <module>
Dec 16 09:30:12 vps omr-admin.py[39333]:     from netjsonconfig import OpenWrt
Dec 16 09:30:12 vps omr-admin.py[39333]:   File "/usr/local/lib/python3.8/dist-packages/netjsonconfig/__init__.py", line 5, in <module>
Dec 16 09:30:12 vps omr-admin.py[39333]:     from .backends.openvpn.openvpn import OpenVpn  # noqa
Dec 16 09:30:12 vps omr-admin.py[39333]:   File "/usr/local/lib/python3.8/dist-packages/netjsonconfig/backends/openvpn/openvpn.py", line 2, in <module>
Dec 16 09:30:12 vps omr-admin.py[39333]:     from ..base.backend import BaseBackend
Dec 16 09:30:12 vps omr-admin.py[39333]:   File "/usr/local/lib/python3.8/dist-packages/netjsonconfig/backends/base/backend.py", line 9, in <module>
Dec 16 09:30:12 vps omr-admin.py[39333]:     from jsonschema import Draft4Validator, draft4_format_checker
Dec 16 09:30:12 vps omr-admin.py[39333]: ModuleNotFoundError: No module named 'jsonschema'
Dec 16 09:30:13 vps systemd[1]: omr-admin.service: Main process exited, code=exited, status=1/FAILURE
Dec 16 09:30:13 vps systemd[1]: omr-admin.service: Failed with result 'exit-code'.
Dec 16 09:30:13 vps systemd[1]: omr-admin.service: Scheduled restart job, restart counter is at 5.
Dec 16 09:30:13 vps systemd[1]: Stopped OMR-Admin.
Dec 16 09:30:13 vps systemd[1]: omr-admin.service: Start request repeated too quickly.
Dec 16 09:30:13 vps systemd[1]: omr-admin.service: Failed with result 'exit-code'.
Dec 16 09:30:13 vps systemd[1]: Failed to start OMR-Admin.
Dec 16 09:30:17 vps ss-server[664]: failed to handshake with 109.190.201.181: authentication error

It says PAS d'ADRESSE PUBLIQUE RECUPEREE PAR SHADOWSOCKS/IMPOSSIBLE DE CONTACTER LE SCRIPT d'ADMINISTRATION DU SERVEUR. NO TOKEN NO API SCRIPT ANSWER

Doing mon-ip.com and in the dash board well appears my VPS/freebox IPV4 Wan address 82.xxx.xxx.165
Trying to configure ipv6 answers no ipv6 and indeed doens't work.
For memory purpose, I have asked a question on forum beacause of upgrade failure with wget trying to connect to openmptcp repository in IPV6, timing out, then succeeding in IPV4 but aborting before the end.
Relaunching the upgrade script taking out from 1srt freebox nexthop field the fe0::2 and let empty, the script went until the end.

I have tried to put VPS eth0 inetd6 adress in 1srt nexthop, deleting it, it is the same.

## Specifications

  - OpenMPTCProuter version: 56.5
  - OpenMPTCProuter VPS version: 1022 on ubuntu server 20.04, behind a freebox 
  - OpenMPTCProuter platform:

Ysurac, 2020-12-16:
jsonschema seems to be not installed, you can try to do ```pip3 install jsonschema``` on the VPS.

jfh75, 2020-12-16:
Oh p---n!!!! Yannick you saved my life, Am trying straight away ! :=))) 
As root always like for upgrade right ?
Merci :=)

Ysurac, 2020-12-16:
Yes as root. This should be installed as dependencies during script install, but maybe a problem sometimes...

jfh75, 2020-12-16:
root@vps:/home/jfh# pip3 install jsonschema
Collecting jsonschema
  Downloading jsonschema-3.2.0-py2.py3-none-any.whl (56 kB)
     |████████████████████████████████| 56 kB 2.0 MB/s 
Requirement already satisfied: attrs>=17.4.0 in /usr/lib/python3/dist-packages (from jsonschema) (19.3.0)
Collecting pyrsistent>=0.14.0
  Downloading pyrsistent-0.17.3.tar.gz (106 kB)
     |████████████████████████████████| 106 kB 7.1 MB/s 
Requirement already satisfied: six>=1.11.0 in /usr/lib/python3/dist-packages (from jsonschema) (1.14.0)
Requirement already satisfied: setuptools in /usr/lib/python3/dist-packages (from jsonschema) (45.2.0)
Building wheels for collected packages: pyrsistent
  Building wheel for pyrsistent (setup.py) ... done
  Created wheel for pyrsistent: filename=pyrsistent-0.17.3-cp38-cp38-linux_x86_64.whl size=106650 sha256=91d14cc635edde8b533a5f6044ee418f4a00da72208de53c387c96c47d530414
  Stored in directory: /root/.cache/pip/wheels/3d/22/08/7042eb6309c650c7b53615d5df5cc61f1ea9680e7edd3a08d2
Successfully built pyrsistent
ERROR: netjsonconfig 0.9 requires jinja2<3.0,>=2.9, which is not installed.
Installing collected packages: pyrsistent, jsonschema
Successfully installed jsonschema-3.2.0 pyrsistent-0.17.3
root@vps:/home/jfh# 
Strange it dropped during install, some necessary things were missing and indeed jsonschema wasn't installed.
I have rebooted the vps AND OMR, still the same :-(
One weird thing I noticed, is my keys from 1srt ubuntu VPS install are different from the one currently stored in /root/open...keys.txt
They are the only ones supposed to be valid right ? Anyway, I have tried both, none is working.
When rebooting OMR, it starts with shadowsocks is not launched in red, and in the system log, it seems it is relaunching 3-4 times....
Still searching....

Ysurac, 2020-12-16:
jinja2 seems to be not installed too so: ```pip3 install jinja2```
I think you have a broken VPS install.

jfh75, 2020-12-16:
And I believed it too.
The thing is that is shit happens now, this means no VPS so not high speed internet for work and holidays.
You think i'd rather relaunch the script ?
Thank you for your help !

jfh75, 2020-12-16:
No it's ok ! Went successfully and seemed to be last piece missing ! 
And V2ray after being red is now working !!
Thank you so much :=)
JFH

closed 2020-12-16 by jfh75