Ysurac/openmptcprouter #1461: VPS firewall

jfh75, 2020-12-17:
Hello,

Reading Howto's on IPV6, I see that ufw -ubuntu server default firewall- might block any incoming traffic in IPV6. And indeed it is installed on my VPS.
Can I uninstall it as I understood shorewall was OMR system default firewall ? I am not exposing my VPS doing this ?

Thank you for your help,

JFH

Ysurac, 2020-12-21:
By default shorewall block everything, only port 65000 to 65535, needed by OpenMPTCProuter services, are open.

jfh75, 2020-12-21:
Ok thank you.

My question is more that to install the VPS, we need a fresh install of 
ubuntu server or debian. It appears that ubuntu is installed by default 
with ufw (uncomplicated firewall).

As shorewall is the running part of OMR firewall, wouldn't it be more 
cautious to uninstall ufw. So that ther can't be a mess with two 
firewalls at the same time.

Thank you for your help,

Best regards

JFH

Ysurac, 2020-12-21:
For now I've not seen any conflict, so while it's work it's not a real problem.

jfh75, 2020-12-21:
Ok thank you.

jfh

closed 2021-03-26 by github-actions[bot]